Traumflug/Teacup_Firmware #111: Graphical config tool

Traumflug, 2015-01-03:
Just uploaded a showcase work by @jbernardis to the config-tool branch. To run the tool, simply type

    python config.py

in a terminal.

My impression: excellent start! Loading an existing config works flawlessly already. "Save" is missing, I take this is because it's just a mockup so far.

Some thoughts:
- Target audience should be 6-year-olds and grandmas. If these can use the tool, people in between will be able to use it, too.
- Having a GUI is an excellent opportunity to also add simple calculation tools.
- If you find inconsistencies between the existing config.h templates, feel free to fix them.
- I'll make a drawing how I think the thing could look like.

Nevertheless, great work!

Traumflug, 2015-01-03:
Here's a draft. Result of a bit brainstorming.

[image]

jbernardis, 2015-01-03:
I encourage everyone to take a look at this and give me feedback.  Being new to the effort, I don't know the firmware and its settings as well as some others might.

One question I have right from the start - the comments surrounding the ACCELERATION setting imply that it is only used with ramping acceleration.  Because of this, I placed the ACCELERATION field within the acceleration ramping branch.  

It seems to me that acceleration is more widely used and, if so, it should be promoted to a higher level of the configuration tree.

Incidentally, incorporating calculators should be relatively simple.

Also, save was missing because, as you guessed, this is a mock-up.  It will be there.

Traumflug, 2015-01-03:
> One question I have right from the start - the comments surrounding the ACCELERATION setting imply that it is only used with ramping acceleration.

Yes, because with ACCELERATION_REPRAP, acceleration is set by the incoming G-code. ACCELERATION_TEMPORAL should have acceleration like ACCELERATION_RAMPING, but it's an option which didn't take off the ground, yet (it's incomplete).

That said, using none of the accelerations is an option, too.

Whould it be possible to read these help texts from one of the config.h templates, too? Text contents should be the same in each of them. This whould ease adjusting user instructions according to code changes for developers.

jbernardis, 2015-01-08:
I uploaded a new version of the config tool.  This is quite a departure from the previous design.  I felt that the tree representation did not lend itself well to the problem, so I switched to a tabbed notebook.  I'm happy with the result, although there is still work to be done.  Some issues:

1. I need a better way to validate the PIN names.  If arduino.h is not included, you can only use integers, but with arduino.h, I allow AIOnn, DIOnn.  Are there other formats I should consider?

2. Most of the fields in the file accept integer values only.  That after all was a design goal of teacup - to minimize if not eliminate the floating point arithmetic.  Because of this, my validation routines only accept integers in these fields - no decimal points.  This causes validation errors immediately upon loading the config files, because the steps and acceleration values contain decimal points.  So I guess the basic question is:  where should I limit the input to integers and where should I accept a float.  Right now, the only float is the EWMA field on the sensor page.

3.  I have not yet been able to handle the CANNED_CYCLE field as it spans multiple lines.  At the heart of this is the way the config file is parsed.  You stated earlier that it loads the files "flawlessly".  While I appreciate the comment, it's far from true - but it might be good enough.  Things that slip through right now:
	- comments.  I don't look for comments other than ones where the line starts with optional white space and the '//'.  I don't know about anything any other types of comments, and although I look for #define at the beginning of a line, other values (such as "DEFINE_HEATER") might be picked up even if they are commented out.
	- lines that are ifdef'ed out.  I am not looking for preproseccor statements other than #define, so I have no idea of if statement nesting, etc.
	- multi-lines.  As mentioned before, statements that span multiple lines are a bit of an issue here.

Also, I made a simplification when it comes to the IFDEF or IFNDEF lines that try to determine if the correct processor type is defined.  On lines that begin with '#if', if there is a string of the form '__AVR_.*__', then I pick it up and assume it's a processor type that we need (as opposed to one we do NOT support).

So other than that, I think this program is in pretty good shape.  I still need to implement calculators for steps, and I still need to add building and possibly uploading, but I now support saving.  As far as validation goes, any time there is an invalid value in any field, 1) the field will have a pink background, 2) the tab for that notebook page will have a '?' in front of the tab name, and 3) the save button will be disabled.  Any time you change a field, the tab for its page will have a '*' in front of the tab name.  If you have errors in a field, the program does not stop you from exiting that field.  You can move freely from field to field and from page to page.  You just will not be able to save until you clear up the errors.

Give it a try and give me feedback.  Do you have any expectations that perhaps I am not thinking of?  I want this to be a useful and a user friendly utility, but I need your feedback to make it so.

triffid, 2015-01-09:
On 9 January 2015 at 10:58, Jeff Bernardis <notifications@github.com> wrote:

>
>    1.
>
>    I need a better way to validate the PIN names. If arduino.h is not
>    included, you can only use integers, but with arduino.h, I allow AIOnn,
>    DIOnn. Are there other formats I should consider?
>
> P[A..Z][0..7] should be valid too, eg PB3 or PC6

>
>    1.
>
>    Most of the fields in the file accept integer values only. That after
>    all was a design goal of teacup - to minimize if not eliminate the floating
>    point arithmetic. Because of this, my validation routines only accept
>    integers in these fields - no decimal points. This causes validation errors
>    immediately upon loading the config files, because the steps and
>    acceleration values contain decimal points. So I guess the basic question
>    is: where should I limit the input to integers and where should I accept a
>    float. Right now, the only float is the EWMA field on the sensor page.
>
> Some config fields accept floats, but they are converted to integers at
compile time. Typically you can tell these because the macros that use them
have conversion and casting stuff, Traumflug might have a handy list.

Traumflug, 2015-01-09:
Jeff, that's awesome! I'm pretty sure people will like it.

> I need a better way to validate the PIN names. If arduino.h is not included, you can only use integers, but with arduino.h, I allow AIOnn, DIOnn. Are there other formats I should consider?

Likely, Teacup won't compile without including arduino.h. This file not only defines pin names, is also maps pin names to their corresponding port and similar stuff. Accordingly, this checkbox for its inclusion would be obsolete.

Also, you can entirely stick to AIOnn and DIOnn names. Yes, some of the current templates use avr-libc names, but that's unneccessary confusion. Mapping pure numbers to AIOnn and DIOnn names is also trivial, just add these 3 characters in front. If AIOnn or DIOnn doesn't work somewhere it's a bug in the firmware code.

Regarding validity of pins. The vast majority of users have a board with fixed connections between ATmega and most of the used pins. Accordingly, changing individual pin names is more an exception. Another conclusion is, pin selection is very limited, only one of the fixed pins or one available on extension/auxiliary headers can be considered to be "valid", not all pins on the ATmega chip.

One solution I see is to add "potential" pin definitions into the template. In comments we can add as much additional information as we want. Having a config tool, there's no longer a need to keep them clean, as in the future only experts will edit them by hand.

> So I guess the basic question is: where should I limit the input to integers and where should I accept a float. Right now, the only float is the EWMA field on the sensor page.

How about adding a /* float */ comment after each number which isn't integer? ACCELERATION is another float.

> I have not yet been able to handle the CANNED_CYCLE

There's no need to parse CANNED_CYCLE. Canned cycles aren't useful for printing and the example given in these comments shall just show how this is defined.

> comments. I don't look for comments other than ones where the line starts with optional white space and the '//'.

Comments are for documentation only, so they can be dropped. Comments starting with /** or /*! are for Doxygen. You could use their \def's and similar for better parsing. And of course it's possible to clean up these comments, making them consistent.

> lines that are ifdef'ed out.

These around F_CPU can be taken as-is. Or, thinking about it, this line can probably be dropped entirely. Currently F_CPU is either set by Arduino IDE or by the Makefile.

> As far as validation goes, any time there is an invalid value in any field, 1) the field will have a pink background, 2) the tab for that notebook page will have a '?' in front of the tab name, and 3) the save button will be disabled.

Dummy-proof! Excellent!

> Do you have any expectations that perhaps I am not thinking of?

Not surprisingly I see a number of misalignements, like mentioned above. But I also think you're very well on the right track. I'm thrilled!

What do others think?

jbernardis, 2015-01-09:
Thanks for the feedback.

You mention that the firmware likely wont compile without arduino.h, so I can remove that option and always add the include statement.

As far as pin validation is concerned, rather than have pins be a free-format field, I can have a drop-down list that can be chosen from.  I guess that since some of the config.h files do not define some pins, that I should also allow for a particular pin to be not defined.

I think you misunderstood my question about float versus integer.  I can't add a float comment because I don't know to which fields it should be added.  Which ones should allow float values?

I think you also misunderstand my issue with comments and ifdef's.  I'm not carrying these forward to the output file.  The issue is that I may accidentally include things that were commented out or ifdefed out in the original file.  This same concern applies to the CANNED_CYCLE.  I can remove it from the screen, but to carry it forward to the output file, I need to parse it correctly on the way in.  I am unable to do this at present - still working on it.  I can always just write out a fixed literal value if that would be acceptable.

BTW - I haven't checked it in yet, but I have the steps calculators working now.  Next is a build button.  It should be simple to do in linux, but it's going to take some work for windows, and I have no way of testing for MAC.

Traumflug, 2015-01-09:
> As far as pin validation is concerned, rather than have pins be a free-format field, I can have a drop-down list that can be chosen from.

That'd be ideal, IMHO.

> The issue is that I may accidentally include things that were commented out or ifdefed out in the original file.

I think the point of misunderstanding here is, you consider all these config.h templates as unchangeable. That's not the case. Feel free to change them to whatever better fits your needs. They only have to to pass three tests:

- User editable (now by developers only).
- Readable by the Config Tool.
- Readable by the compiler.

Everything else is at your disposal. We can even change firmware code to better fit the config tool, e.g. add or remove #ifdefs.

> The issue is that I may accidentally include things that were commented out or ifdefed out in the original file.

Well, then remove this from the original file.

> This same concern applies to the CANNED_CYCLE. I can remove it from the screen, but to carry it forward to the output file, I need to parse it correctly on the way in.

It's plain G-code and no user will use this as-is. If you simply provide an empty by default text field you're fine.

I'll make a commit to this branch before too long which shows what I mean. Removing unused comments, standardizing comments for easier parsing, providing /* float */, providing candidate pins for thermistors and all this stuff.

jbernardis, 2015-01-09:
Thanks.  You're right - I considered these files as sacrosanct - not to be changed for a mere configuration tool.  Now that I understand the rules a little better, I have some changes that I'd like to see.  

I look forward to your examples as well.

BTW - the PIN fields are all now drop-down lists.  It made validation a lot easier and reduced the code considerably as well.

Regarding the size of the code, I hate that it's all in a single file, but I did not want to clutter the directory with a bunch of python files.  I think it's gotten the the point where I need to do so though. To mitigate the clutter, I'm thinking I'll just create a subdirectory.

Traumflug, 2015-01-09:
Fine. Jotted down my ideas, too, and pushed them to the config-tool-traumflug branch. For details, see the commit message: https://github.com/Traumflug/Teacup_Firmware/commit/0f9acc1233d52658b8c4f2ff1f6069a930a7f4cd

Yes, just a brain dump. Not mandatory, feel free to change it further or even ignore it.

jbernardis, 2015-01-11:
I like the idea of separating the printer config from the board configuration.  It's actually the printer config that most printer owners will be modifying; there should be very little change to the board settings by the average user.

There is a bit of a issue though.  I can standardize comments so that I can parse them as I read the config file, but I do NOT write comments to the output file.  I can foresee someone overwriting the input file with their changes, destroying all of the formatted comments.  Because of this I'm thinking that it might be best to have some sort of master file - actually 2: one for the printer and one for the board.  As developments are made to the firmware and new parameters are defined, the master file(s) would need to be updated.  There most likely would also need to be a corresponding change to the config tool, but I still think this is the right way to go.

I want to think this over for a few days.  I can pull together a set of master files and put them out here for you to see and comment on.

triffid, 2015-01-11:
shouldn't be too hard to read the whole file into memory as-is, find
relevant lines and modify relevant values in-place or add them to the end
of the file if they're not present, then write back out.

This would preserve all user formatting and comments and suchforth

jbernardis, 2015-01-11:
The problem with this is that there are logical groupings in the file.  If there is no definition for a particular pin, for example, and that definition is added, it should be added within its grouping.  By adding it to the end, I don't observe this grouping and the flle becomes cluttered - the lines are not with the comments that describe them.  I know that the missing pins are often in the file as comments, but I'm not parsing comments so I never see them.  This means the comments will remain behind in the original grouping, while the actual definition will appear later in the file.

I disagree that it "shouldn't be too hard" to read in and parse a file that has many authors and no formatting consistency.   At a minimum, we need to impose some rules.   For example, a comment that illustrates a pin that's not being defined needs to be made with the '//' syntax instead of a /* that might have appeared 50 lines earlier and a */ that might appear 50 lines later.  That way I can reasonably be expected to turn that comment into a #define without worrying about fixing up comments above it and below it.

I still think we need some kind of master file.  I just want to think on it a while.  My first thought was that it be XML, but that's too cumbersome.  I think the normal C header file formatting with some decorations would suffice and wouldn't be burdensome.

triffid, 2015-01-11:
On 11 January 2015 at 16:55, Jeff Bernardis <notifications@github.com>
wrote:

> I think the normal C header file formatting with some decorations would
> suffice and wouldn't be burdensome.

That's what it is right now :P

Traumflug, 2015-01-11:
> At a minimum, we need to impose some rules. For example, a comment that illustrates a pin that's not being defined needs to be made with the '//' syntax instead of a /* that might have appeared 50 lines earlier and a */

Disallow/ignore comments inside comments? Yes, why not.

Also:
- Exactly one documenting comment per group, starting with /**
- A group ends before the next documenting comment.
- If one setting allows multiple values, add all of them with a // comment and uncomment only the currently active one.
- Comments which describe the allowed range.

All this can be pretty strict, because these files are written either by the config tool or by developers. All of them start out with a valid file, so it's not too hard to identify mistakes. If a user starts to mess with these files it's also not much of a problem if the file becomes kind of invalid for the config tool. I think it's unlikely they'll read it back into the tool after manual changes.

You're pretty free as long as it's consistent and you write down the rules somewhere. :-)

jbernardis, 2015-01-14:
I just updated the config-tool branch.  I made major changes to the tool.  It now updates the output files in place, so comments, etc, are propagated.  This necessitated some syntax restrictions on the files, but these are all summarized in a txt file in the config directory.  I reworked all of Traumflug's example header files to illustrate these restrictions.

I think it's much much better, but there are still things to do.  Give it a try and let me know what you think.

Traumflug, 2015-01-14:
Very very good! I'd be entirely happy if this general layout and behaviour is kept this way.

Not surprising, I see a number of possible detail improvements:

- Printer/Mechanical -> Calculate Belt Driven: buttons at the bottom are half cut off and resizing the window isn't possible. Might be a wxPython issue.

- Printer/Acceleration: Acceleration doesn't accept float values, looks like the /* float */ in the template isn't recognized.

- Board/CPU: "Motherboard" can be removed from the GUI, just set it in the resulting file, it applies to (old) Gen3 only. All other boards require it to be always defined. In case this tool is ever used again for a Gen3, I think it'd be a good idea to have main board (MOTHERBOARD defined) and extruder board (MOTHERBOARD not defined) as two entirely separate boards.

- Board/CPU: you allow multiple CPUs here, which obviously can't work. At compilation time you have to decide which CPU is plugged in. Is it a good idea to see these #ifdef's in the template as a choice from which a user can select from? It might confuse a Sanguinololu user if he's offered with ATmega2560 as a choice. The choice can then be written as a comment to the result file, the main purpose of this selection is to choose the right compiler flags.

- Board/Pinouts: you allow virtually all ATmega pins here. In Board/Heaters you offer only available pins. The latter is IMHO much better.

- Board/Temperature Sensors: when adding a sensor you only offer types which are checkmarked in the list of available sensor types. I'd do it the other way: allow all types when creating a sensor and set the neccessary sensor types according to that. This way you can hide the list of sensor types from the user.

- Board/Temperature Sensors and Board/Heaters: when adding a sensor/heater and selecting the pin I can save the choice, but the pin then showing up in the table is a different one. E.g. selecting DIO7 results in AIO2 in the table. Nice thing is, it demonstrates how not available pins are not accepted, the table row turns red :-)

- config.h isn't written. I think it can be written every time either printer or board are written.

- Board/Communications: selecting USB serial should disable Baud rate.

All together, it looks like I can fire up my editor to start converting all the config templates. That's a good one for the traditional makefile / Arduino based configuration anyways.

jbernardis, 2015-01-14:
Thanks Traumflug

response to your items:
1) this looks fine on my windows machine.  I'll take a look at it on my ubuntu system to see if there is an issue.  Perhaps it's linked to screen resolution???

2) acceleration does accept a floating point number. It's just that my definition of a floating point number does not allow a trailing decimal point.  If you enter a digit right behind the decimal point, the "red" will go away and the field will be accepted.  I can also easily relax this constraint.  It's just a regular expression that occurs once in the program.

3) I will remove MOTHERBOARD and just carry the value through from the input file

4) regarding multiple CPU's - this section generates the #ifndef statements at the top of the file, and there are many files that allow multiple CPU types - the ramps header file allows either a 1280 or a 2560.  I added this in case you wanted to add to or remove from the list of potential CPUs.  

5) the pins in the dialog boxes are derived from the candidate list in the config file.  These exist only for heaters and sensors.  For all other pin fields, I just use a list of ALL pin names.   If there is a narrower list I should be using, just let me know.  Ideally, I'd take all of the pin names that are specified on the page and remove them from the choice lists, but doing this and adapting these lists as pins are selected might be expensive.  I can give it a try.

Incidentally, regarding the valid pin names, I'm not sure how high the numbers go.  For AIO pins, I've seen up to 63.  For DIO, I've seen in the teens, so I'm allowing 0-15.  Let me know if I need to change my ranges.

6) agreed -- I will work on that

7) you found a bug.  I found the error and know what to do to fix it.

8) I still have the config.h file on my to-do list.  I have several questions related to it.  Right now, when opening or saving a file, I allow full traversal of the file system.  I'm thinking of limiting it to the config directory.  Further, I'm thinking of asking only for the printer name or the board name, and building the file name from that.  I'd no longer be able to use the wxpython file open dialog box, but I think it might make things a bit easier.  

Regarding when I write config.h, I obviously can only write it if you have both a board and a printer file loaded.  If you are working on only 1 file, I do not have all the information to write this file.

9) agreed - will do.  BTW, regarding the communications page, it seems to me as if the xonxoff define belongs on this page instead of printer/miscellaneous.

Traumflug, 2015-01-14:
> 1) [...] Perhaps it's linked to screen resolution???

Maybe, 1920 x 1200 here. The missing height is also about the height of the window bar, so Windows and Ubuntu might have a different idea of what "window height" is.

> 2) [...] I can also easily relax this constraint.

That'd be ideal. I mostly noticed because it was set invalid right after loading the template.

> 5) the pins in the dialog boxes are derived from the candidate list in the config file. These exist only for heaters and sensors.

It's certainly possible to add such lists for other pins, too. For boards with fixed stepper drivers (most of the boards) this might lead to lists as short as one entry.

... which leads to a less trivial question. I'm writing the following not because I think the current interface needs immediate changes, more to show up what might arrive in the future.

Currently we support X, Y, Z and E axes and require all of them to be defined. However, there are at least two efforts already to extend this to more axes. In some future there might be X, Y, Z as well as A, B, C, U, V, W, and up to 9 E axes (E == E0, E1, E2, ...). Accordingly, we'd no longer have all axes neccessarily defined and maybe also a user wanting to define a set of pins as A axis where others want the same hardware as E axis. Which in turn leads me to the impression that an interface closer to that of defining heaters might be better: define a set of pins as axis in a separate box, then add this to a list. I'm not even sure wether it's a good idea to do this right now, because users are still very close to the XYZE paradigm.

> regarding the valid pin names, I'm not sure how high the numbers go.

It depends on the ATmega type. The smaller ones have AIO0..AIO7 and DIO0..DIO31, the bigger ones AIO0..15 and DIO0..63. There might be intermediate types. And ARM CPUs with different amounts are just around the corner.

> 8) I still have the config.h file on my to-do list. I have several questions related to it. Right now, when opening or saving a file, I allow full traversal of the file system. I'm thinking of [...]

I see no mandatory reason to not allow file saves anywhere. An #include "/home/somebody/somewhere/myboard.h" should work for the compiler. That said, starting in config/ is a good idea.

> Regarding when I write config.h, I obviously can only write it if you have both a board and a printer file loaded. If you are working on only 1 file, I do not have all the information to write this file.

You could use config.default.h. Only writing config.h when both are loaded is also fine, though.

> regarding the communications page, it seems to me as if the xonxoff define belongs on this page instead of printer/miscellaneous.

Yes. My mistake  :-)

jbernardis, 2015-01-18:
I just checked in a new version of the config tool, incorporating most of traumflug's comments.

To point out where I didn't incorporate his comments:
1) The issue with appearance differences between linux and windows had to do with fonts.  I just let the fonts default, and they defaulted to two quite different things on the two OS'es.  To fix it, I am now explicitly choosing a font and sizing the windows accordingly.  Appearances seem to be somewhat uniform between the toe systems (at least for MY systems).  The major difference I still see is that for the static text fields, windows is right justifying the text (as I requested) and linux is ignoring the request and left justifying.  Still the functionality is there.
2) on the pinouts page, I still present a list of ALL pins.   I tried to dynamically adjust the lists as the user makes choices, but it took about 2 to 3 seconds after choosing a pin before all controls were updated and the screen was ready for further user action.  I deemed this unacceptable and just kept it at the static list.
3) I still do not write the config.h file when saving a board and/or printer file, but I added a menu command to save config.h which writes config.h AND the printer and board files also.

As always, I await your feedback.  At some point, I want to add the ability to build the hex file, but that's going to be tricky in the windows environment where I most likely will not have make, and the avr-gcc compiler and avrdude are not going to be in any standard location.  I'll do some investigation, but in the meantime, if anybody has any suggestions for this, I'm all ears.

Traumflug, 2015-01-18:
Excellent. Dialog sizes fit now.

> 3) [...]

We have now no less than 9 menu entries just for handling load & save. Well ...

... anyways. I think it's time to get this out to users now. Let's see how they can handle this, their question often give good hints on what they expect intuitively.

Regarding build tools: I think it's fine to require an Arduino IDE installation. The user can point to this installation, then all tools are in known places. A script to replace the makefile looks like this:
````
for C in *.c; do
  <path/to/arduino/>/avr-gcc <options> -o "build/${C%.c}.o" "${C}"
done
<path/to/arduino/>/avr-gcc <options> build/*.o -lm build/teacup.elf
<path/to/arduino/>/avr-objcopy <options> build/teacup.elf teacup.hex
<path/to/arduino/>/avrdude <options> teacup.hex
```

euchcat, 2015-01-18:
Why Python tools such config.py, are not included in the "Master" branch? I find them very useful and even essential.
I find this firmware (Teacup) very clean and maintainable, I try to adapt to a Prusa i3 (Chinese Z605 printer) with Melzi Version 2.3 board (ATmega1284p + SD card). Not everything works but it still advance!.
A more and sorry for my poor english ...
Lionel

Traumflug, 2015-01-18:
> Why Python tools such config.py, are not included in the "Master" branch?

Because master is the stable branch, config.py is just two weeks old, barely completed and moving incomplete work to the stable branch doesn't magically complete or test or review this code. Writing, testing and reviewing code is work and as such, takes time :-)

That said, you're invited to give config.py a try to find out how well it works for newbies already :-)

Same for [SD card support](https://github.com/Traumflug/Teacup_Firmware/issues/104). It should work for reading, but the developer who did this stuff has no such hardware, so he has no idea wether it actually does so :-)

euchcat, 2015-01-19:
Ok no problem to go back information.
For the moment, on the Melzi 2.3 board, the cooling fan extrude does not
work, the functions of reading and writing in EEPROM are ineffective, I
look at it tomorrow what may come. (I guess it must reassign the
corresponding pins)
Good night (for me)

jbernardis, 2015-01-19:
I'd like to bring back up the subject of CPU type that was discussed earlier in this thread.  Traumflug indicated that I should only allow 1 CPU type to be specified, but I stated that this was not really specifying the CPU type - it was checking that the CPU type in the makefile was one of potentially several legitimate types.  That was fine when I was just producing the .h files, but now that I'm working on the actual build, this (and F_CPU) needs to be revisited.

The issue shows itself in two different ways, depending on if "make" is being used to build.

When make is NOT being used (windows) the F_CPU and the CPU type need to be specified on the avr-gcc command line.  However, they are also in the .h files (with potentially different values).  This seems redundant, and in the case of F_CPU, since it's on the command line I build, the value in the header file will be ignored since it is within an ifdef statement; since my code is responsible for the value in both places, it seems I should be able to simplify.  Also, since the code asks for a LIST of acceptable CPUs, it now has to ask for which one of those is the one we are actually using for the build.

when make IS being used (linux, os/x???) the issue is a little less concerning.  Both values are in the makefile.  The current checks in the header file make sure that the CPU is valid.  The F_CPU value in the header file is most likely never used since it only comes into play if the variable is not given a value in the makefile.

What I propose is the following:
1) F_CPU and CPU type are both entered on the cpu page, but only 1 cpu type is allowed.
2) The CPU type preamble in the header files is left untouched, and the CPU type entered by the user has no effect on the header file.
3) The F_CPU is entered in the board.h file, but it most likely will NOT be used because the line will be ifdef'ed out.  I'd say we omit this value from the header file entirely, but then people on windows who do not use the config tool will be negatively affected.
4) For building with makefiles, these values are not used again.  This requires that the CPU type be in the makefile; F_CPU may or may not be.  Note that in this circumstance, the CPU type specified in the config tool is not used AT ALL, and F_CPU will most likely not be used..
5) For building without make, these two values are inserted on the avr-gcc command line.

Traumflug, 2015-01-19:
@euchcat let me know when you have code to drop. As it's not allowed to send patches here, send them to mah@jump-ing.de (I'll apply them, then) or you can even get write access.

Traumflug, 2015-01-19:
> What I propose is the following:
> 1) F_CPU and CPU type are both entered on the cpu page, but only 1 cpu type is allowed.
> 2) The CPU type preamble in the header files is left untouched, and the CPU [...]

Sounds good.

Having CPU type and F_CPU in board.h also serves as a list of choices a user can choose from. Most current boards are 16 MHz only, but ARMs with many choices are just around the corner.

Yes, that means you can no longer define this list with the GUI, but I think this isn't neccessary. We have other stuff with the same property already, for example possible pins for thermistor or heater pins.

Also, there's no need to use make. It'd be perfectly fine to use sort of a hidden script with fixed flags like Arduino IDE does. Compatibility with Arduino IDE will be kept virtually forever, breaking building with it won't be accepted for experimental/master (or only by accident). This also means you never have to care about code in subdirectories or compiled with custom flags, because Arduino doesn't do such things.

I guess experts and developers will use the GUI rarely. But for newbies, which heard about RepRap just yesterday (or behave as if they did), it'll be a huuuuge benefit. And the latter are like 99 of 100 curious people stepping by.

jbernardis, 2015-01-20:
I just uploaded a new config tool with all of the changes mentioned above plus the ability to build and upload.  To support the build and upload, I have added a .ini file: configtool.ini.  There are 5 or 6 data items that need to be provided, but the file is commented so it should be pretty easy to understand.

I was able to test building the hex file under windows, and building and uploading under Linux (ubuntu 14.04).  I have no way of testing OSX, and no easy way to test upload from windows.  If someone could verify these, I'd appreciate it.

I'm not sure if I broke down the build functionality the best way or not.  Right now, there is a build and an upload menu choice.  build does compilation through linking and objcopy.  Upload just does the avrdude part.  Upload is only available if there is a teacup.hex file available.  Build requires some of the data from the config files, so it requires that they be loaded beforehand, although if they are not loaded, the program asks you if you want it to load them for you - so it's pretty easy to use.  It also makes sure that the values you have loaded match what is in the config.h file.

jbernardis, 2015-01-20:
Oh - the candidate processor types and candidate clock rates are now in the board file - look at my two examples to see the syntax.  This is also described in the config-syntax.txt file in the config directory.

jbernardis, 2015-01-21:
Unless there is something specific that needs to be done, I am finished with this branch.  I haven't added REPORT_TARGET_TEMPS to the config tool, but it would take be just a couple of minutes.  Other than that, I don't want to add to this branch if you are thinking of merging it into the experimental branch.

I guess once that merge is done, we will remove this config-tool branch?  Otherwise, it could get very confusing.

Traumflug, 2015-01-21:
> Other than that, I don't want to add to this branch if you are thinking of merging it into the experimental branch.

I've already started reviewing. Converting to Unix line endings and such stuff. Also trying to understand what's going on. No threat, just takes an evening or two.

jbernardis, 2015-01-21:
No threat taken.  I will withhold changes to this branch then.  The only one pending is the REPORT_TARGET_TEMPS in config tool.

Traumflug, 2015-01-25:
While I'm fairly good at bash scripting, I think I should confirm the following, wether it's true for python as well. I've reached now the part where multiple files appear and some of them differ in the very first line. Most files start outright with *import ...*, some files read there:

    #!/bin/env python

I take this line means to make the file a script executable. And also, that only the top level config.py should have such a line, because all others can't be executed standalone. Is this right?

Also, does this line actually work in some environment? On Ubuntu it doesn't. Easy workaround is to start with python, of course:

    $ ./config.py
    bash: ./config.py: /bin/env: bad interpreter: No such file or directory
    $ which env
    /usr/bin/env
    $ python config.py
    <...works...>

Maybe it'd be better to put something like a literal *#!/path/to/your/python* there, so all users get the same experience and don't confuse each other just because they have different OSs (aka. "You fool, it works for me out of the box!").

jbernardis, 2015-01-25:
Actually, I think eclipse puts that in there when I create a new file.  I'm so used to using the python command that I don't pay it any mind.  

You are correct that the only file that should have that is config.py.  It is the only python file that should be executed directly.  If the other files have that line, it is a mistake and I can remove it  from my source code.

I did a little digging, and I found that the line that works for me, in ubuntu, is 

#!/usr/bin/env  python

This is supposed to be the preferable/portable way to do it.  I have no idea if this works on a MAC or not, and I think on windows, the only thing that matters (at least from the GUI) is the file extension.

I've also had issues with this header line because of the fact that windows terminates lines with CR-NL instead of just NL.  For some reason, the unix shell doesn't see the CR as white space in this one context.

I have several other changes pending for the config tool.  I can make this change quickly.

Traumflug, 2015-01-26:
Thanks for the clarification. On an older Mac (OS X 10.4.11):

    $ which env
    /usr/bin/env

Seems to agree with your Ubuntu findings, let's take this :-)

I've moved from CR/NL to NL only already. Will push that soon. You can also commit and push on the (or another) topic branch, I'll rebase that, then.

jbernardis, 2015-01-26:
Just pushed to config-tool branch with all we were discussing. 

I believe I took care of the CR/NL issue.  All files now end with NL only (I think).  Between this issue and the backslash as the directory separator, moving between unix and windows/DOS drives me crazy.

Traumflug, 2015-02-05:
Just to leave a mark, review is still in progress. Git is a bit stubborn at times and sometimes I simply sit there for a while enjoying reading this excellent code. Whatever comes to mind what might be missing, you wrote it already. Messages on success, messages on failures, all there, can't get this thing to misbehave.

jbernardis, 2015-02-05:
Thanks for the kind words, but I wouldn't be surprised to see that there's a bug or two in there.  I'm eager to get it into the hands of a larger set of users just to get more testing done.

Traumflug, 2015-02-06:
Done. On the experimental branch. :-)

Jeff, I hope all the edits don't step on your toes. I'm aware this can happen, I have a tendency to overexaggerate with "reviewing". While I did quite some whitespace editing, code logic should still be 100% the same. This tool is your work and I'm very happy to have it in the repo.

Previous code is still available unchanged in the config-tool-backup branch. config-tool was rebased onto experimental and melted down to just some 20 lines in a single commit. Are these perhaps accidental changes or do they solve a problem? The commit is here: https://github.com/Traumflug/Teacup_Firmware/commit/dbb876fd9bd6c167147290967b3b2f0860f9ea1a

While doing all this, a few new questions/tasks appeared. Much less fundamental than the ones above in this thread:

1. Convert all older config.h templates one by one, don't forget to add them to the list in config.default.h. Then remove the older ones (all at once).

2. Remove the "Load default config.h" menu entry? It might be not meaningful because everybody has a different printer.

3. Somehow try to avoid to overwrite templates delivered with the Git repository (config/printer.mendel.h, config/printer.wolfstrap.h, etc.)? Users might shoot in their foot here, making it impossible to revert to the original version without re-unpacking the whole project. Possible implementation with a hardcoded list of file names. Perhaps simply dimming "save" as long as the name of the loaded file matches one of the provided files?

4. Align texts with input fields and menu choices vertically at the text baseline. Currently they align more on the top boundary. That's purely cosmetic.

5. boardpanel.py and printerpanel.py should not only enforce a .h at the end, but also a "printer"/"board" at the start of the file name, else user might not be able to open their own files.

6. Get rid of STEP_INTERRUPT_INTERRUPTIBLE. Serves no purpose any longer, the code section which was tried to be enhanced almost vanished.

7. %ALNAME% in configtool.ini and its parsing obsolete? I mean: is the build tool also convenient for developers who look at stuff like assembly code? If not, saving or handling temporary files is pointless and avoiding that whould speed up building a bit.

My current plan is to collect some of the smaller commits which appeared over the last two weeks, then continue on the configtool with 1. of the above list. Opinions or commits on the other ones are very welcome, of course.

Traumflug, 2015-02-06:
P.S.: it's just fascinating how one can edit comments in the templates, change the settings with the tool and all the stuff reappears tidily in the saved file. Entirely new #define values are kept as well. Magic!

jbernardis, 2015-02-06:
I actually find that I leave the config tool running while I am tinkering with settings.  Just make changes and do a build and an upload and the arduino is ready with its new config. Reconnect with the host and try it out.

I can start work on the 7 comments above.  I was itching for something to do anyway.  Should I continue to make my changes in the config tool branch?

Also, I have no problem with others editing the code.  This is an open source cooperative effort, and it's kind of expected.

Traumflug, 2015-02-06:
> I actually find that I leave the config tool running while I am tinkering with settings.

Nice! Looks like the --save-temps should go back into the compiler flags, I had it removed for the possible malfunction decribed in configtool.ini. BTW., it might be not a Winfows issue, but an aged gcc issue. --save-temps with arguments is relatively new. avr-gcc 4.8.1 here (which works). It should be compatible with old versions, of course.

> I can start work on the 7 comments above. I was itching for something to do anyway. Should I continue to make my changes in the config tool branch?

Sure, go ahead. config-tool branch is fine. This branch is empty anyways, except for this odd commit. It might be neccessary to re-create your local config-tool branch, because what you have is now config-tool-backup. Like

    git checkout experimental
    git branch -D config-tool
    git checkout config-tool  # will create a new branch from origin/config-tool

Generally, branch names don't matter at all as long as it's not "master" or "experimental". Gitk --all nicely lists all of them.

jbernardis, 2015-02-08:
I just pushed a new config tool.  I had some difficulty merging my changed files with all the changes traumflug made, but I believe I retained his changes as well as incorporating 6 of the 7 comments from a few days ago.  The only one I didn't do is number 7 because I think we still need to talk about what C compiler flags we need to use.  Also, although I morphed all of the config files into board files for the predefined board types, I am not familiar with all of those boards - so I might have gotten some of the pin definitions wrong.

Traumflug, 2015-02-19:
Oh shutup, Github. This was me and the commit is 12 days old, but got a superfluous space removed a few minutes ago. :-)

Anyways. Finally managed to get everything reviewed. Looks very good now.

To celebrate this wonderful tool I'd like to do a little information effort in the RepRap forum and similar places. Text might go like this:

### Teacup: firmware made simple, newbies wanted ###

What do users answer when you ask them what they want? "I also want this", "I also want that", "... additionally ..." Yet the iPad with its feature set much smaller than that of a laptop is currently one of the most wanted devices on the planet.

Meet *Teacup* firmware. Focusses on the essentials, refined for years. Does only what's actually needed and does that right. Hundreds of hours went into improving these fine details which are barely visible, yet make the distinction of a pleasurable experience. For example, step losses simply don't happen because it follows the laws of physics precisely. But, well, so far all these configuration M-codes were missing, calibration was a command line experience.

Meet *Teacup Configtool*. Suddenly, all these M-codes are obsolete! Change your configuration in a graphical interface, hit the upload button and be done. So much easier than juggling with host softwares, than juggling with M-codes, than juggling with configuration files. **Adjust, click, be done**.

We did it!

[image]

Why I'm writing all this? Well, we're developers. We know exactly how this thing works. If something goes wrong, we know how to fix it in seconds and barely recognize such events. As such, handling Teacup is always easy to us, even when compiling it from the command line.

But is it easy for you? That's what actually matters! Here we need your help. Newbies, inexperienced printer owners, those who don't know what "C" or "Python" means. Now you can tell us and help to hone and further simplify what's already easy.

To start helping, please (incomplete from here on).... download and unpack experimental branch ... double-click configtool.py ... if asked, click "execute" ... select board, select printer, hit upload. That's it, you can connect with Pronterface and start printing.

Didn't work? Tell us! Even try to abuse it and tell us how you got it killed! Whatever you don't understand or isn't intuitive for you in this GUI: tell us! Here or even on Teacup's issue tracker: https://github.com/Traumflug/Teacup_Firmware/issues (green button)

Didn't I promise one-click changes? Go ahead, try it! Change a value, select upload and you're done. No manual neccessary, all value fields show help if you hover over them.

-----

What do you think? Can we reasonably hold up with this already?

jbernardis, 2015-02-19:
I'm ready.  I'd like to get users using it to shake out the bugs.

It occurred to me too that there will be a constant effort in keeping the config tool up with the changes, and I'm not certain the best way to handle it.  

For example, I've seen recent commits dealing with delta printers and LCD support.  I assume they have configuration parameters of their own.  At some point, probably when they get pulled into the experimental branch, I should update the config tool to be in step. 

We'll just have to work out the best procedures as we go forward.

Traumflug, 2015-02-19:
> It occurred to me too that there will be a constant effort in keeping the config tool up with the changes

I fear that's unavoidable. That said, I see your code makes this easier than I could previously think of already. To get started with a new feature one can simply append #defines with a text editor to a config file and Configtool will keep them across saving and loading. Hacking is as easy as before, well done!

In boardpanel.py and printerpanel.py I found a string "New defines added to board config:", but so far I couldn't figure how to create such "new defines". No hurries, currently I'm swamped a bit with reviewing anyways.

jbernardis, 2015-02-19:
The logic behind the "New defines added to board/printer config" is as follows.  If a user runs this tool against an old config file, the config tool will add the defines it is expecting to the file.  This is to facilitate upgrading to a newer version.  You can still use your old config file - any new defines will be added for you, and then after running the tool, you can doctor the new config file to have some help text perhaps.

Adding a completely new define to the tool itself is a bit more complicated.  I tried to make it as easy as possible, but there is no way to infer what screen it should be on or where on that screen it should appear.  It's not rocket science, but there are specific things that need to be done.

jbernardis, 2015-02-19:
If I follow the 3 commands from above, namely:

>git checkout experimental
>git branch -D config-tool
>git checkout config-tool

will this base the new config-tool branch on the experimental, or do I need to specify the experimental branch as the starting point for the new branch?  If the latter, what's the syntax?

Traumflug, 2015-02-19:
These 3 commands will throw away your local copy of the branch and make a new local copy from the remote branch. You're based on the config-tool branch then, which in turn is based on experimental.

Do you have "gitk"? It usually comes with Git distributions, on Ubuntu it's an extra apt-get.

If yes, run it with --all. This will nicely show you how all the commits and branches are related. It also shows you each commit by just clicking on it. I use it all the time and Menu -> File -> Reload every few Git commands. Currently it looks like this:

[image]

Traumflug, 2015-02-25:
I've changed the name of configtool.ini to configtool.default.ini in the repository. When loading, configtool now tries to load configtool.ini, then configtool.default.ini and if this also fails, it resorts to hardcoded (mostly empty) default values. Saves always go to configtool.ini. This way distributed files are no longer overwritten with local settings, which was inconvenient when doing commits and also prohibited users from restoring factory settings.

Also changed the tool to save config.h instead of loading it in case the loaded configuration doesn't match the content of config.h. The former behaviour led to a build not matching what was visible in the GUI when selecting the build menu, likely not what a user wants. BTW., my compliments for even detecting such mismatches!

All on the experimental branch. Being not exactly experienced in Python I hope the changes are right.

----
One problem remaining (besides issue #119): when loading printer or board templates, then changing something and then trying to build, Configtool tries to save the changed board or printer. The problem is, if this is still a distributed template name, the save fails, building isn't possible (it tries anyways, which is another bug).

How should we proceed in this situation? Simply abort, asking the user to save manually? Silently save to a temporary file? Bring up a file save chooser dialog in midst of building?

What do you think?

jbernardis, 2015-02-25:
I actually have some changes ready to check into the configtool branch.   One of the changes I made was to rename the distributed in file  I chose a different name,  but the actual name is immaterial.

The bulk of what I will be checking in is code to create thermistor tables.   The only thing I have left to do is to make  the changes to the various board files.   I will do it tonight and do a commit. 

Regarding the save before a build,  my vote would be to put up the save as dialog box and let the user choose an alternate file. 

Let me get my changes in so you can merge,  and I can then deal with the build issue

jbernardis, 2015-02-26:
I hope I did the right thing right now.  I went to put all of my changes into the config-tool branch and almost every file had a conflict.  It was quite painful going through every file figuring out what I should keep and what should go, but I'm pretty certain I ended up with the code I wanted.  The uncertainly came when I did the push.  I was checked out of the config-tool branch, and my push was accepted on that branch.  However there were also messages that the master and experimental branch rejected the push because it was not fast forward.  I was not attempting to push to those branches, so if the end result was what I wanted I guess I'm ok, but I did not like seeing those reject messages.

In any event, this is the logic for thermistor tables.  I also added logic to properly handle builds when the modified file is a protected file.  There might be a hole in the logic, but it's much more robust now.

Traumflug, 2015-02-26:
Thanks a ton, Jeff. At the first glance it looks like you didn't remove your local config-tool branch but hacked away on this. No problem, I'll sort this. All the whitespace changes are gone, much easier now.

jbernardis, 2015-02-26:
I don't really understand what happened.   I did remove my local config-tool branch,  then I checked out the experimental branch,  then recreated the local config-tool branch. Should I have deleted the remote config-tool branch too.

Traumflug, 2015-02-26:
> Should I have deleted the remote config-tool branch too.

No, that would have been a (non-fatal) mistake. Good thing is, we're about done with these branch(es), so I'll simply delete all of them as soon as I've picked the stuff over.

I take that Git is a bit confusing for those with less experience. I simply use it for a number of years now for about everything, so I no longer recognize how complicated it really is.

jbernardis, 2015-02-26:
I use git at work, but I've never taken any kind of class.  Just learning it as I go.  As I said, I thought I did all the right steps, but it's as if when I did my push, it was pushing against the OLD config-tool branch instead of replacing that branch with the newly created one that I based on the experimental branch.

Traumflug, 2015-04-25:
This topic continues on Issue #119.

closed 2015-04-25 by Traumflug

el-shweikh, 2018-02-06:
HI I woundered how to connect endstop to my arduino nano ?How to configure its pins on configtool?