Traumflug/Teacup_Firmware #114: There should be a periodic temperature report while waiting for temps -  M116

jbernardis, 2015-01-21:
The location traumflug gave me of lines 92&93 in clock.c was the right place to add this logic,  but I had to make the report conditional on a few things.   I needed a new routine in dda_queue to determine if we were actually waiting.   In any event,  I have this working after a fashion.   I need to play around with it a bit more,  and get the formatting where I want it,  and then I'll create a new branch for it.

Traumflug, 2015-01-21:
Nice you got it working! Feel free to upload it as-is and reformat later. Topic branches don't have to be tidy. I'd be interested in what this new routine is and why queue_empty() isn't sufficient.

Maybe a flag raised in dda_queue.c line 85, right after the SetTimer()? Or even sending the report from there? It's interrupt time there, still you can serial_writestr_P(), up to the size of the serial buffer. HEATER_WAIT_TIMEOUT can be changed to something like one second.

jbernardis, 2015-01-21:
I will put up the branch tonight (East coast time) when I get home from work.

When I removed the comments from those two lines, the code wouldn't build because 1) temp_print needed an argument, and one  wasn't in the commented-out code, and 2) the routine temp_get_target() did not exist so there was a linking problem.

I fixed those two problems by 1) adding TEMP_SENSOR_none as the argument, and 2) writing the missing routine which I gathered from the name was determining if any target temp was set.  I also had to add sending a new line character after the temp report.  This worked, except that as soon as I set a target the report started coming out and it kept coming out after the target was reached and also while I was actively printing.  This was not what I was looking for; I wanted the reports to stop once the temperature was achieved.

So next, I rewrote the if statement in clock.c.  Instead of "if (temp_get_target())", it became "if (waiting_for_temp() and !temp_achieved())".  

I had to write the routine waiting_for_temp() in dda_queue.c which is basically a one-liner:

  return (!queue_empty() && movebuffer[mb_tail].waitfor_temp);

After this it worked just fine.

jbernardis, 2015-01-21:
branch m116-temp-report has been created.  Only 3 files changed.

Traumflug, 2015-04-25:
Thanks. Just rebased the *m116-temp-report* branch to experimental to bring it up to date.

Looking through the code I found an alternative which costs only 14 bytes binary size while your solution adds 80 bytes. Both should work equally. Could you confirm this, please? It's on the *m116-alternative* branch.

jbernardis, 2015-04-26:
I think your solution looks fine.  I was just happy to have the functionality, but obviously the code should be as efficient as possible.

Traumflug, 2015-04-26:
Fine. Thanks for bringing this up. Picked this alternative to experimental and removed these two m116-* branches. One bug less :-)

closed 2015-04-26 by Traumflug