Traumflug/Teacup_Firmware #119: Configtool: uh, totally forgot ThermistorTables.h

Traumflug, 2015-02-19:
Configtool doesn't handle Thermistors, yet. I see two possibilities:

1. Make configtool creating this themistor table(s). A Python script for calculating them is already available in createthermistortable.py. Likely needs some wrapping to create a readily usable header file.

2. Bring the effort in the thermistortable[23] branches to an end. This makes these tables obsolete and calculates the values at runtime, without interpolation. Slightly slower, though, some 2500 cycles vs. 1800 with interpolation. As with 1. it needs Configtool to accept the Thermistor values, but should write them into the board file instead.

What do you think?

jbernardis, 2015-02-19:
I think this is a great idea. I need to look into it a bit because I didn't actually create a table for my setup - I just used the one that was there.

But I think incorporation into the config tool is a good idea.

Traumflug, 2015-02-19:
> I think this is a great idea.

Which of the two? :-)

Anyways, with 1. each thermistor can use its own table or share one with another. Name of the table is the last parameter in the DEFINE_TEMP_SENSOR() macro. Fewer and smaller tables make the binary smaller, of course. These tables are based on thermistor nominal value, comparison resistor, voltage and a few others and are in principle subject for calibration.

With 2., all these values are appended to the same macro. Example:

    //                 name       type            pin        vadc
    //                 r0         t0              r2         beta
    DEFINE_TEMP_SENSOR(extruder,  TT_THERMISTOR,  AIO0,      5.0, \
                       100000,    25,             4700,      4092)

Uhm. Please use a new branch off experimental this time. Making more than one commit doesn't hurt either. ;-)

jbernardis, 2015-02-19:
I just meant the general idea of incorporating the table definitions into the config tool.  I was probably thinking closer to number 1.

jbernardis, 2015-02-19:
The more I think about this, the more I like option 2.  Except, the python script accepts more parameters than you show in your example: r1, min_adc, max_adc, etc.  Could be that they are practically unnecessary?

In any event, I could certainly change the config tool to create the macro invoke that you show above.  However, the actual definition(s) of DEFINE_TEMP_SENSOR would need to change, and the logic behind looking up a value would also need to change.  I can't see offhand how we'd generate a static table inside a macro definition like this, so I assume the table would need to be replaced by logic?  Is that what you have in mind?

Traumflug, 2015-02-19:
This is a typical command to run the tool:

    ./createTemperatureLookup.py --r0=100000 --t0=25 --r1=0 --r2=4700 --beta=4092 --max-adc=1023

- r0 ist the nominal value, 10'000 or 100'000 (ohms), exceptions exist.
- t0 is 25 (°C) for all thermistors known to me, doesn't need user editing.
- r1 is always zero with all currently available electronics, also no user-edit neccessary.
- r2 is the 4k7 resistor, some controllers use 1k and measuring this resistor exactly improves accuracy.
- beta has to be read by the user from the thermistor's data sheet.
- min_adc is zero and max_adc is 1023 for all ATmegas (actually for all 10-bit analog-digital converters).
- vadc is comparison voltage, 5.0 on 5V-controllers, 3.3 on 3.3V controllers.
- numtemps is the number of value pairs created. Against popular belief ridiculously high numbers don't raise accuracy noticeable, because values are interpolated. Temperature regulation would even work with just 3 or 4 pairs. High numbers also slow down searching the table. I think 25 values are a good compromise.

> However, the actual definition(s) of DEFINE_TEMP_SENSOR would need to change

It works now, the content of TermistorTable.{single|double}.h was created with this tool. No need to put all these value pairs into the macro. Putting the name of the table into it sufficient, just like it's currently done.

jbernardis, 2015-02-19:
what I thought you were proposing with option 2 was to change the statement:
>DEFINE_TEMP_SENSOR(BED, TT_THERMISTOR, AIO1, BED_TEMPERATURETABLE)

to

>DEFINE_TEMP_SENSOR(BED, TT_THERMISTOR, AIO1, 5, 10000, 25, 4700, 4092)

This WOULD require the macro change.

However,  I can certainly accept these parameters via the GUI and then run the table creation code to generate the header file.  That way I could enforce that the table names would match up

Traumflug, 2015-02-19:
> This WOULD require the macro change.

Yes, but only for option 2. The firmware code for this is actually in the repo already, see branches thermistortable2 and thermistortable3. Visible in commit 93611a49e79a7af4506e163a1c01c6744f9b3d7b

----
The above is about option 2., next paragraph is about option 1.

----
> However, I can certainly accept these parameters via the GUI and then run the table creation code to generate the header file. That way I could enforce that the table names would match up

Yes, this would likely give the quickest results.

Traumflug, 2015-04-25:
Fine. Picked all of the latest changes to experimental and removed the config-tool branch. Thank you very much!

I still have a ToDo list, but the items are now mostly minor. Just FYI:

- Rename createThermistorTable.py to configtool/createthermistortable.py.
- Also clean this file up, align to 80 chars/line and convert tabs to spaces.
- Enforce no whitespace at line endings.
- Uploading can't be aborted. For example, uploading with a mismatching baud rate causes avrdude to try over and over again, for minutes, and there's no way to stop that.
- Firmware baud rate is misused as avrdude baud rate. These can differ. There are already a few TODOs in the source code.
- Rename ThermistorTable.h to thermistortable.h.
- Equal thermistor tables should be joined.
- There should be a menu with presets for various known thermistors.
- Editing a non-thermistor fails, because these dummy-characters don't validate (background pink), despite they're not needed.
- All these "saved successfully" and "successfully saved" should be removed. This outcome is expected, because it only happens on a menu selection (check this!) or after a confirmation dialog (check this as well!). Confirming expected outcomes is, well, clutter.

jbernardis, 2015-04-27:
I just made an interesting observation.  If I build using one of the generated thermistor tables, I get nonsensical values in the temperature report - temperatures are reported in the 7000 - 8000 degree range and no heater will turn on.  If I replace the file with one of the delivered files (specifically the double file) the reports are normal.

I've examined the two files and other than the fact that one file has tables with only 25 entries while the other has file has tables with 102, I can see no syntactic difference between them.  More investigation is needed, but I just wanted everyone to know there is an issue here.

Traumflug, 2015-04-27:
Again, please commit and push such files somewhere :-)

jbernardis, 2015-04-27:
You didn't read the comment right - there are no files to commit.  ThermistorTables generated by the config tool do not work quite right.  I don't know why - I haven't had the opportunity to try to debug yet.  I posted here so others can see if they observe the same issue, or perhaps someone had a suggestion.

If you go thru the config tool and generate a thermistor table, load it up and send it an M115 command, you will get back a temperature report with the temperature in the thousands of degrees.  If you replace the generated table with ThermistorTable.double.h and build, the temperature report is fine.

Traumflug, 2015-04-27:
What stops you from committing this generated file? It's less for keeping this file forever in the repo or even making it part of the distribution, it's just for allowing others to see this file. A simple file exchange, similar to using pastebin.com.

jbernardis, 2015-04-27:
I figured out that the issue was the the thermistor tables that were generated had negative numbers in them.  Since the math is done with unsigned variables, the results were "not as expected".

I fixed it and pushed everything up to a new branch, but for some strange reason I named the branch issue114 - obviously it should be issue119.  Rename it if you want, but due to the nature of the change, I would advise propagating it into the experimental branch.  Only 1 file was affected.

Traumflug, 2015-04-28:
You fixed it. That's even better, of course :-)

Picked this one commit onto experimental without any edits and deleted the issue114 branch.

jbernardis, 2015-05-06:
This is addressed in branch configtoolPhaseB.

I have doubts about whether or not I work with the proper set of parameters for thermistor table generation.  For the beta algorithm, some of the parameters are specified via the ini file.  This doesn't seem consistent with the SH algorithm.  I might want to pull more parameters out of the ini file into the dialog box.  I don't think it will be too overwhelming since we have the presets now.

Traumflug, 2015-05-25:
This discussion was continued on issue #130.

closed 2015-05-25 by Traumflug