Traumflug/Teacup_Firmware #120: Configtool doesn't recognize changes for building

Traumflug, 2015-02-28:
#### Steps to reproduce:

1. Start tool, load board, load printer.
2. Change CPU type, e.g. from ATmega644 to ATmega1284P.
3. Save board.
4. Build.

#### Expected behaviour:

Binary is built for ATmega1284P, the MCU selected in the GUI.

#### Actual behaviour:

Binary is built for ATmega644, the MCU which was loaded with the board file.

#### Additional notes:

- Same problem for upload.
- Reloading the just saved board file works around the problem.

jbernardis, 2015-03-01:
I have the fix for this - two files change - boardpanel.py and  printerpanel.py.  What's the best branch to deliver them on??

Traumflug, 2015-03-01:
Any new branch, name doesn't matter. For some time we used to name branches issue-119, issue-120 ...

jbernardis, 2015-03-01:
Sorry about this, but I accidentally pushed my changes to experimental.  Just two files - printerpanel.py and boardpanel.py - changed plus I added configtool.ini to the .gitignore.

I created a local branch - issue-120 - and put my changes in place.  Then when I pushed, I didn't notice the changes were going to experimental.  I guess I should have done a checkout issue-120 first???

Traumflug, 2015-03-01:
No problem, already fixed. Previous experimental is restored, the commit is on a new branch issue-120.

> I guess I should have done a checkout issue-120 first???

This, or push like

    git push origin issue-120

Traumflug, 2015-04-25:
Just reviewed this and picked it onto experimental. Thank you very much!

closed 2015-04-25 by Traumflug