Traumflug/Teacup_Firmware #121: Reliability of the clock interrupt, especially dda_clock()

Traumflug, 2015-02-28:
@NickE37 brought it up in https://github.com/Traumflug/Teacup_Firmware/issues/117#issuecomment-76409732 , dda_clock() apparently isn't as reliable as it should be. To investigate this I've pushed some code to the interruptreliability branch. The idea is to make a copy of the movement duration, then to count this backwards in dda_clock(). This counter should be zero at movement end, else dda_clock() calls got lost.

First tests with ACCELERATION_TEMPORAL, without Delta, aren't that promising. The following is a transscript from real hardware, ATmega1284P @ 16 MHz:

```
start
ok
2 milliseconds = 32000 CPU clocks.

X10 F200
ok
interval 3750
exact 48000000  duration 48000000
 = 3000 ms / 3000 ms
exact 128000  duration 128000
 = 8 ms / 8 ms

X20 F400
ok
interval 1875
exact 24000000  duration 24000000
 = 1500 ms / 1500 ms
exact 96000  duration 96000
 = 6 ms / 6 ms

X30 F600
ok
interval 1250
exact 16000000  duration 16000000
 = 1000 ms / 1000 ms
exact 96000  duration 96000
 = 6 ms / 6 ms

X50 F800
ok
interval 937
exact 24000000  duration 23987200
 = 1500 ms / 1499 ms
exact 32000  duration 19200
 = 2 ms / 1 ms

X100 F1000
ok
interval 750
exact 48000000  duration 48000000
 = 3000 ms / 3000 ms
exact 224000  duration 224000
 = 14 ms / 14 ms

X0
ok
interval 750
exact 96000000  duration 96000000
 = 6000 ms / 6000 ms
exact 448000  duration 448000
 = 28 ms / 28 ms

X90
ok
interval 750
exact 86400000  duration 86400000
 = 5400 ms / 5400 ms
exact 384000  duration 384000
 = 24 ms / 24 ms

X0
ok
interval 750
exact 86400000  duration 86400000
 = 5400 ms / 5400 ms
exact 512000  duration 512000
 = 32 ms / 32 ms

X100 F1100
ok 
interval 681
exact 87200000  duration 87168000
 = 5450 ms / 5448 ms
exact -4032000  duration -4064000
 = -252 ms / -254 ms
```
As one can see, we have non-neglibile errors at a boring 200 mm/min already. Having 1280 steps/mm, that's 4266 steps/s or 3750 CPU clocks per step. Things don't get worse up to 1000 CPU clocks/step. At 750 clocks/step ( = 21'000 steps/s) noticeable errors appear, 680 CPU clocks/step are not good.

Traumflug, 2015-02-28:
I get no such errors when doing the same without any acceleration type. Reliable 0, -2 or -4 ms difference, as expected.

Traumflug, 2015-02-28:
No problem either, down to 575 clocks/step,  when commenting out the code wrapped by ACCELERATION_TEMPORAL in setTimer() in timer.c. Going even faster, and it degrades quickly.

I think this narrows down the problem a bit. It's this juggling with timestamps, the effort to allow impossible short delays, like setTimer(0).

dc 42 had a better idea anyways: instead of trying to fire an interrupt as quickly as possible in case an impossible short period was requested, he simply repeats the step interrupt routine directly. Much simpler maths, less overhead.

Traumflug, 2015-03-01:
I've implemented dc42's idea with repeating the interupt routine directly instead of trying to re-scedule an interrupt as fast as possible. The result is nice, I've measured up to 40'000 steps/s ( = down to 400 CPU clocks) on the 16 MHz controller. That's 50'000 steps/s on 20 MHz, even a bit faster than ACCELERATION_RAMPING.

First tests went good, it looks like the algorithm works right.

Scheduling of dda_clock() is apparently also much more reliable. I no longer get 0, -2 or -4 ms (without the ACCELERATION_TEMPORAL extra code), but now -2 (rarely), 0, 2 or 4 ms over all the range. No more mismatches in the 2-figure milliseconds range.

Before putting this onto the experimental branch I'll have to measure this for the ramping code, because timer_set() got a return value and also a second argument, which might waste time.

NickE37, 2015-03-14:
I tried the updated ACCELERATION_TEMPORAL dda_step() on real hardware.

It does seem to fix the interrupt reliability issue, but introduced another issue:  Unstep() is called immediately after _step().  While the driver chip only needs 1us to register, internet research indicates that a typical stepper motor need around 350us to step.  It is unclear what effect microstepping has on the time required.  Testing on a real printer has the majority of the steps not happening.

It is unclear how to handle this issue.  Have separate unstep() for each axis with a built in (interruptable) delay?

triffid, 2015-03-14:
Some drivers may need pulse widths of 2-3us (our allegro drivers are happy
with 1us), but 350us would be insane, that's the time for 10 or more steps
at teacup's max speed

you could call unstep from another timer if you want, Smoothie uses this
technique for configurable pulse widths. Note that it may reduce Teacup's
top speed as it takes several cpu cycles to enter and leave an interrupt

NickE37, 2015-03-15:
I apologize if I was being unclear.  I understand the driver only needs 1us or so pulse *widths* to register, but the stepper motor itself needs considerably longer than 1us to actually *perform* the step.  What is the result if the stepper driver chip is being told to move the motor much faster than the motor can perform?  Are the steps lost?  Does the motor catch up?

Traumflug, 2015-03-15:
> What is the result if the stepper driver chip is being told to move the motor much faster than the motor can perform? Are the steps lost? Does the motor catch up?

Steps within microstepping are catched up, that's what allows strategies like quadstepping. Larger amounts are lost or not lost, depending on current stepper speed. Too high step frequency over a longer time and the stepper simply stops moving while continueing to beep.

Also noteworthy, unevenly distributed steps always cause a rougher, louder movement of the stepper.

That said, the timing of unstep() has no business in all this. The step is registered at the raising signal flange, falling flange doesn't matter. See page 11 top and 14/15/16 in the A4988 datasheet http://www.allegromicro.com/~/media/Files/Datasheets/A4988-Datasheet.ashx

Traumflug, 2015-04-25:
This appears to be solved. If not, please re-open.

closed 2015-04-25 by Traumflug