Traumflug/Teacup_Firmware #122: Compile Issues for Teensy 3.1 - Ubuntu

michaeljball, 2015-03-06:
Similar to Compile Issues for Teensy 3.1 #93  @nikkiverre 
Following the Developer's instructions from http://reprap.org/wiki/Teacup_Firmware#Developer_Installation

Using the newest teensy3 folder from https://github.com/PaulStoffregen/cores

I can't sort out why it can't find pin declarations DIOx  in dda.c  
I'm sure it's just a stupid oversight on my part, but...

Satellite-L305:~/sketchbook/Teacup_Firmware$ make && make program
make: Nothing to be done for 'all'.
  CC        build/usb_serial.o
  CC        build/teensy_analog.o
  CC        build/usb_dev.o
  CC        build/home.o
  CC        build/dda_queue.o
  CC        build/crc.o
  CC        build/usb_mem.o
  CC        build/usb_desc.o
  CC        build/sermsg.o
  CC        build/dda.o
dda.c: In function 'dda_create':
dda.c:251:3: warning: implicit declaration of function 'WRITE' [-Wimplicit-function-declaration]
dda.c: In function 'dda_start':
dda.c:473:9: error: 'DIO6' undeclared (first use in this function)
dda.c:473:9: note: each undeclared identifier is reported only once for each function it appears in
dda.c:474:9: error: 'DIO8' undeclared (first use in this function)
dda.c:475:9: error: 'DIO10' undeclared (first use in this function)
dda.c:476:9: error: 'DIO12' undeclared (first use in this function)
dda.c: In function 'dda_step':
dda.c:533:10: error: 'DIO5' undeclared (first use in this function)
dda.c:551:10: error: 'DIO7' undeclared (first use in this function)
dda.c:569:10: error: 'DIO9' undeclared (first use in this function)
dda.c:587:10: error: 'DIO11' undeclared (first use in this function)

Traumflug, 2015-03-06:
> Using the newest teensy3 folder from https://github.com/PaulStoffregen/cores

It might be a better idea to use the package version nikkiverre used. Should work, though.

>  dda.c:473:9: error: 'DIO6' undeclared (first use in this function)

These are defined in arduino_teensy.h, which is included in arduino.h. Obviously your Makefile doesn't meet this criteria in arduino.h:

    #if defined (_mk20dx256_h_)

I'd try with a -D_mk20dx256_h_ as additional gcc flag. I guess Arduino IDE (which is was nickiverre used, AFAIK) adds such a flag on it's own. Also AFAIK, gcc doesn't support such flags as internal definitions, because -mmcu exists for AVR, but not for ARM.

michaeljball, 2015-03-07:
I was originally using the Teensyduino plugin in the Arduino IDE (v1.0.6) and had this same error.  I thought I would get more verbose messages by following the Developer Intructions and compiling from the command line. 

It turns out I wasn't paying attention to naming convention and took the config file at face value.

In config.teensy3.h, where the hardware pin definistions are kept, the one in the repository has DIOx as opposed to DIOx_PIN defined.   These are user replaceable dependent on hardware build.  I missed that.  I've corrected the file and am past that issue now.  
Thank you.

example:
    /* starting down the left side for digital, and later down the right for analog */
    /* General layout: ccw from upper left: stops,  X,Y,Z,E,  temp sensors, PWMs */

    #define PS_ON_PIN     DIO0                            <---  Should be DIO0_PIN
    //#define PS_INVERT_ON
    #define STEPPER_ENABLE_PIN    DIO4         <---  Should be DIO4_PIN
   #define STEPPER_INVERT_ENABLE

drf5n, 2015-03-07:
Hi,

Its been a long while since I've looked at this and my .   I've got Arduino 1.0.6, a teensyduino plugin of uncertain age, and I compile from the command line.  

    $ git clone https://github.com/Traumflug/Teacup_Firmware.git && cd Teacup_Firmware
    Cloning into 'Teacup_Firmware'...
    remote: Counting objects: 6901, done.
    remote: Compressing objects: 100% (51/51), done.
    remote: Total 6901 (delta 24), reused 0 (delta 0), pack-reused 6850
    Receiving objects: 100% (6901/6901), 4.62 MiB | 5.11 MiB/s, done.
    Resolving deltas: 100% (4099/4099), done.
    Checking connectivity... done.
    $ git co teensy3
    Branch teensy3 set up to track remote branch teensy3 from origin.
    Switched to a new branch 'teensy3'
    $ cp config.teensy3.h config.h
    $ cp ThermistorTable.single.h ThermistorTable.h
    $ cp Makefile-teensy3 Makefile
    $ make
    mkdir -p build
      CC        build/analog.o
      CC        build/analog_ARM.o
    In file included from config.h:320:0,
                     from config_wrapper.h:12,
                     from analog.h:16,
                     from analog_ARM.c:1:
    arduino.h:109:2: error: #error pins for this chip not defined in arduino.h! If you write an appropriate pin definition and have this firmware work on your chip, please tell us via the forum thread
    make: *** [build/analog_ARM.o] Error 1
    $

I found I needed to add an additional condition to the arduino.h file:

    $ git diff HEAD^ |sed -e 's/^/    /'
    diff --git a/arduino.h b/arduino.h
    index 176b0bc..f18a3e2 100644
    --- a/arduino.h
    +++ b/arduino.h
    @@ -101,7 +101,7 @@
     #endif
     #endif /* SIMULATOR */
     
    -#if defined (_mk20dx128_h_) // Teensy 3.0 uses a MK20DX128 32 bit ARM Cortex-M4 48 MHz Test on this for config.h
    +#if defined (_mk20dx128_h_) || defined ( __MK20DX128__ ) // Teensy 3.0 uses a MK20DX128 32 bit ARM Cortex-M4 48 MHz Test on this for config.h
     	#include    "arduino_teensy3.h"
     #endif

...and it seems to compile from the command line just fine with the config.teensy3.h
     
    $ make  | sed -e 's/^/    /' > junk
    $ cat junk
    mkdir -p build
      CC        build/analog.o
      CC        build/analog_ARM.o
      CC        build/clock.o
      CC        build/copier.o
      CC        build/crc.o
      CC        build/dda.o
      CC        build/dda_lookahead.o
      CC        build/dda_maths.o
      CC        build/dda_queue.o
      CC        build/debug.o
      CC        build/delay.o
      CC        build/gcode_parse.o
      CC        build/gcode_process.o
      CC        build/graycode.o
      CC        build/heater.o
      CC        build/home.o
      CC        build/intercom.o
      CC        build/mendel.o
      CC        build/mk20dx128.o
      CC        build/pinio.o
      CC        build/pins_teensy.o
      CC        build/sd.o
      CC        build/serial.o
      CC        build/sermsg.o
      CC        build/sersendf.o
      CC        build/teensy_analog.o
      CC        build/temp.o
      CC        build/timer.o
      CC        build/usb_desc.o
      CC        build/usb_dev.o
      CC        build/usb_mem.o
      CC        build/usb_serial.o
      CC        build/usb_serial_t3.o
      CC        build/watchdog.o
      CC        build/yield.o
      LINK      build/teacup.elf
      OBJCOPY   teacup.hex

There were a few warnings related to the special 'q' format for the fixed point, but I've ignored them.

    dda.c: In function 'dda_create':
    dda.c:121:16: warning: unknown conversion type character 'q' in format [-Wformat]
    dda.c:121:16: warning: unknown conversion type character 'q' in format [-Wformat]
    dda.c:121:16: warning: unknown conversion type character 'q' in format [-Wformat]
    dda.c:121:16: warning: too many arguments for format [-Wformat-extra-args]
    dda.c: In function 'dda_start':
    dda.c:448:16: warning: unknown conversion type character 'q' in format [-Wformat]
    dda.c:448:16: warning: unknown conversion type character 'q' in format [-Wformat]
    dda.c:448:16: warning: unknown conversion type character 'q' in format [-Wformat]
    dda.c:448:16: warning: too many arguments for format [-Wformat-extra-args]
    dda_lookahead.c: In function 'dda_join_moves':
    dda_lookahead.c:327:17: warning: 'crossF' may be used uninitialized in this function [-Wuninitialized]
    gcode_process.c: In function 'process_gcode_command':
    gcode_process.c:584:21: warning: too many arguments for format [-Wformat-extra-args]

I've push the modified arduino.h back to github.

The DIOx definitions seem to work fine.

Dave

triffid, 2015-03-07:
@michaeljball

> #define PS_ON_PIN     DIO0                            <---  Should be DIO0_PIN
> //#define PS_INVERT_ON
> #define STEPPER_ENABLE_PIN    DIO4         <---  Should be DIO4_PIN

This is absolutely the wrong way to fix the issue, and *will* break things, I have no idea how you managed a successful compilation with them set that way.

The READ/WRITE/etc macros in arduino.h use string concatenation to extract _PIN, _PORT, etc from the pin you specify, so DIO0_PIN would become DIO0_PIN_PIN and DIO0_PIN_PORT which should not exist.

@drf5n seems to have fixed things by adding another condition to the arduino.h file to include arduino_teensy3.h

michaeljball, 2015-03-07:
Thank you so much Dave and Michael.  Dave, I followed your direction to a tee and am now compiling.
and thank you for the "make  | sed -e 's/^/    /' > junk"   yeah... Anyway, appreciated.

I did notice one thing.  in the make file, previous instructions had discussed modifying TOOLCHAIN = and TOOLSPATH = ...  but I had missed the "CFLAGS += -idirafter " path to link in the  "arduino-1.0.6/hardware/teensy/cores/teensy3"  core files from Paul.  <facepalm>

It all compiles appropriately now.   

I am modifying config.teensy3.h now (Yes, Michael, properly) to represent my particular hardware implementation, but truly appreciated the help.

michaeljball, 2015-03-19:
(Apologies, I innappropriately posted this to Issue #42 and have since removed it)

Are there any tricks I need to be aware of to get USB serial showing up on the Teensy 3.1?
It compiles now error free (thank you again!) but I know Paul had mentioned elsewhere that the serial interface doesn't register until first use...
I'm at a stand still at the moment. Everything "appears" proper. If I compile and upload a different app to the Teensy, I have USB Serial, but not so much with Teacup_Firmware..

I'm using the Teensy3.1 branch with Paul's latest Teensyduino as per instructions here. The Makefile does have: "CFLAGS += -DUSB_SERIAL" as enabled from the repo... As I said... not sure where to look or how to debug.

Thoughts?
Thank you in advance.

michaeljball, 2015-03-21:
Success!!! 
So... stupidity on my part.  I folowed Dave's instructions VERBATIM... yeah.  So I've been working off of the teensy3 branch instead of Nikki's teensy3.1 branch...
started over with a "git co teensy3.1"  
Copied the appropriate files into place updated the config.h file as per the missing definition Dave pointed out... aaaaand...
Well config.h ALSO needed a similar fix for the teensy3.1's MK20dx256 
#if defined (_mk20dx256_h_) || defined ( __MK20DX256__ ) 
// Teensy 3.0 uses a MK20DX128 32 bit ARM Cortex-M4 48 MHz Test on this for config.h
    #include    "arduino_teensy3.h"
#endif

**Then I had issues with:**
*dda_lookahead.h:19:2: error: #error Your config.h does not specify one of MAX_JERK_X,*
*dda_lookahead.h:19:2: error: #error MAX_JERK_Y, MAX_JERK_Z or MAX_JERK_E while LOOKAHEAD is enabled!*

Once I defined them (someone is going to tell me that they are autodefined somewhere else, right?) then it compiled fine.  My serial port /dev/ACM0 showed up, and (drum roll please!!!)
**repetier host saw my teensy3.1 as Teacup Firmware!**

Thank you all for your patience with me. 
I've documented it here at : http://arduino-pi.blogspot.ca/2015/03/success-teacupfirmware-on-teensy31-and.html

closed 2015-03-21 by michaeljball