Traumflug/Teacup_Firmware #125: Simulator compile issues on Ubuntu 14.04

RaD, 2015-03-20:
Hello, 
I have sucessfully built the firmware itself but still can not build simulator for it:

    rad@n150:~/reprap/PCB/Teacup_Firmware$ LANG=C make -f Makefile-SIM
      CC        build/sim/dda.o
    In file included from dda.c:14:0:
    dda_maths.h: In function 'um_to_steps':
    dda_maths.h:32:3: warning: implicit declaration of function 'pgm_read_dword' [-Wimplicit-function-declaration]
      CC        build/sim/dda_kinematics.o
    In file included from dda_kinematics.c:8:0:
    dda_maths.h: In function 'um_to_steps':
    dda_maths.h:32:3: warning: implicit declaration of function 'pgm_read_dword' [-Wimplicit-function-declaration]
      CC        build/sim/dda_maths.o
    In file included from dda_maths.c:6:0:
    dda_maths.h: In function 'um_to_steps':
    dda_maths.h:32:3: warning: implicit declaration of function 'pgm_read_dword' [-Wimplicit-function-declaration]
      CC        build/sim/delay_sim.o
    In file included from ./delay.h:4:0,
                     from simulator/delay_sim.c:3:
    /usr/lib/avr/include/stdint.h:142:17: error: conflicting types for 'intptr_t'
    /usr/include/unistd.h:268:20: note: previous declaration of 'intptr_t' was here
    make: *** [build/sim/delay_sim.o] Error 1

What's wrong there?

Traumflug, 2015-04-12:
> dda_maths.h:32:3: warning: implicit declaration of function 'pgm_read_dword' [-Wimplicit-function-declaration]

pgm_read_dword() reads from flash, which isn't available on a PC. It shoud read from RAM (normal memory) instead. Something like a

    #define pgm_read_dword(x) *(x)

should solve this. Now, the simulator worked before, so there's likely already a solution for this. Just can't find it right now. @phord, do you have an idea?

> /usr/lib/avr/include/stdint.h:142:17: error: conflicting types for 'intptr_t'
> /usr/include/unistd.h:268:20: note: previous declaration of 'intptr_t' was here

It's unclear to me why the AVR version gets included at all. Shouldn't happen.

phord, 2015-04-13:
Fixed in a80433d4188559b7fe517de37ae37a0f445b2e3a

Traumflug, 2015-04-14:
Thanks. Picked to this repo.

Traumflug, 2015-04-26:
Uhm. I've seen the issue-125 branch appeared again. The one commit on this branch is on experimental already. OK to delete this branch?

phord, 2015-04-26:
Yep. Not sure why it showed up again.

commit 437cc996 (2015-04-13): simulator: Add pgm_read_dword macro

pgm_read_word was already defined, but pgm_read_dword did not get in our
way during earlier testing. These functions force the arduino to read
these "defined in program" constant arrays from "program memory" instead
of RAM as a means to save on RAM usage. The PC-based simulator doesn't
need such tricks and only needs to read the value directly from the in-RAM
array.  Therefore it is safe to convert read directly from the pointer
being passed.

pgm_read_word already does this. Define pgm_read_dword similarly to
pgm_read_word. Fixes #125
commit 653d3e15 (2015-04-14): Simulator: add pgm_read_dword() macro.

pgm_read_word() was already defined, but pgm_read_dword() did not
get in our way during earlier testing. These functions force the
arduino to read these "defined in program" constant arrays from
"program memory" instead of RAM as a means to save on RAM usage.
The PC-based simulator doesn't need such tricks and only needs to
read the value directly from the in-RAM array. Therefore it is safe
to convert read directly from the pointer being passed.

pgm_read_word() already does this. Define pgm_read_dword()
similarly to it. Fixes #125.
closed 2015-04-26 by phord