Traumflug/Teacup_Firmware #126: If I want to run X/Y/Z as DC motor, do I just define as "heaters"?

michaeljball, 2015-03-24:
I'm slowly going through the Teacup_firmware trying to understand how to integrate my closed loop DC servomotor axes, and it looks very quickly like you have done most of the work already.  

I saw your comments re: DC motor for Extruder.  Would I be safe to assume that I could use same for my X/Y/Z axes PIDs?  In the "heater" code, you have already created robust PID routines, so I do not need to drag mine along.   Do you know of any "gotcha's" with regard to the number of concurrent PID loops? 

What are your thoughts on a data structure for ACTUAL position/velocity/accel as provided by the AXIS linear encoders?

triffid, 2015-03-24:
Teacup does not support DC servos on any axis. The extruder can run as an
open-loop DC motor but there is no feedback. XYZ would require encoder
feedback, and Teacup doesn't have enough spare CPU cycles to keep up with
encoders.

If you're determined to use DC servos for XYZ, they will need a separate
controller that accepts step/direction pulses.

michaeljball, 2015-03-25:
I respect your position, however I am running Teacup on the Teensy 3.1 with a 32bit ARM core.  Teacup plus the encoder code takes less than 12% of available memory, and at 72Mhz is barely tickling the processor.  There is plenty of overhead available. 
I've already successfully completed the separate controller route, with STEP/DIR support for DC servomotor.  Now it is time to integrate them onto the same processor. 

http://arduino-pi.blogspot.ca/2015/03/success-teacupfirmware-on-teensy31-and.html

I am a stubborn goat.  Not very well educated in coding, I understand, but determined to make this combination work.  

I love how you have set up the HEATER PIDs.  There are a few things I will need to work around, like your minimum/maximum  temperature failsafes, that have no context in DC motor control, but I already have the X/Y axis decoders running to tell me where the extruder ACTUALLY is in relation to 0:0.  I just need to really understand how to inferface with your PID to control the PWM out (target location) in relation to the encoder current location. 
  
I'm just looking for some patient guidance, and understanding when my questions seem stupid. 
Thank you Michael.

Michael Ball  ( unix_guru at hotmail dot com )

Traumflug, 2015-04-12:
> There is plenty of overhead available.

Yupp. This "Firmware XYZ doesn't have enough spare cycles" is a myth implemented be the 32-bit fans. Actually, a firmware doesn't provide any CPU cycle, the CPU does. And regarding usage of these CPU cycles Teacup is certainly among the most sparse ones, giving you the most room for expansion. 

> I will need to work around, like your minimum/maximum temperature failsafes

Teacup has no such failsafes I'm aware of.

Regarding using the heater PID algorithms: they're executed every 10 ms only and also put more emphasis on accuracy. dda_clock() is executed every 2 ms, so it should be better suited. These 2 ms can be reduced further with TICK_TIME_MS in timer.h.

Regarding the algorithm its self: there's likely a different implementation advised. Having an encoder your signal is precise, so you don't need averaging/smoothing of the signal and accordingly no history record. For an implementation I'd search literature. Servo control is as old as CNC machining and early CNC machines had just 8 bit @ 1 MHz (6502, 6509, Z80 CPUs) available, so it's undoubtly doable on 8 bit @ 20 MHz or 32 bit @ 72 MHz. LinuxCNC (Open Source!) also has such algorithms implemented.

polskiziom, 2015-12-11:
Hey!

 

Open message <http://kabelxusa.com/face.php?tj3>

 

mrkey@op.pl

polskiziom, 2016-02-01:
Hey!

 

Open message <http://ks.mpt.ru/expression.php?fi>

 

mrkey@op.pl