Traumflug/Teacup_Firmware #127: Create a channel for this repository on Gitter

RaD, 2015-04-14:
That is so useful.

Traumflug, 2015-04-14:
What would be the point of this?

RaD, 2015-04-14:
Hmm, for instance, to ask about strange formatting of code. When I have opened files in Emacs I see strange formatting. It seems because of TAB&SPACE mix. Maybe it would be good to bring the source code to solid formatting. I want to make some pull request but this thing make me uncomfortable.

Also, it there a prepared tables for 100K Epcos like chinese thermistors? Can I connect thermistors on A6/A7 pins on Arduino Nano3 board?

I think that there will be enough questions that are related on Teacup but not issues.

Traumflug, 2015-04-14:
> I think that there will be enough questions that are related on Teacup

Please discuss them here. I certainly don't have the time to attend such a channel 24/7 and/or to solve issues the minute they come up.

> It seems because of TAB&SPACE mix.

Yes, currently it's a mix. Looks fine if you set your editor to 2 spaces/tab.

The underlying problem is, Git doesn't deal too well with whitespace changes. Changing just whitespace would make abut all topic branches stale, introducing conflicts everywhere. Accordingly the strategy is to change only modified lines to indentation with two spaces.

> I want to make some pull request but this thing make me uncomfortable.

Pull requests make indeed things a lot more complex than neccessary. They'd work well if such requests came with perfectly fine edits, but in practice this rarely happens. Wrong indentation, small misunderstandings, unneccessary code bloat, whatever. People fixing things usually don't care about code quality or compatibility with others than their own setup, they want the "works for me" experience. Accordingly about every patch needs review.

Opening an issue containing a simple link to the commit is entirely fine.

Traumflug, 2015-04-14:
P.S.: if there's a question you can't solve by reading instructions and/or the tooltips in the configtool, that's undoubtly an issue. Setup has to be self-explaining.

RaD, 2015-04-14:
Ok, I can't get right temperature values from EPCOS 100k like chinese thermistors. I use Arduino Nano3 with Atmega328p. Can I connect thermistors on A6/A7 pins? Is there a ready-to-use thermistor table?

Traumflug, 2015-04-14:
> Can I connect thermistors on A6/A7 pins?

Depends on wether you manage to let the pin slide into the connector hole. :-)

No, seriously, all analog pins are equally fine. You have to change the pin definition accordingly, of course.

> Is there a ready-to-use thermistor table?

There are some 20 to 50 types which are all "EPCOS 100k". The default table used one of these.

Then, what is "right"? What temperature do you have and what does Teacup display?

Traumflug, 2015-04-25:
The Configtool on the experimental branch supports creating thermistor files on the fly now. Start it with *./configtool.py* or *python configtool.py*, then load a board and a printer, then step through the tabs to adjust your settings. One of these tabs allows editing thermistor properties. Hover over the editing fields to see helpful tooltips. There are still a few rough edges, but it certainly works to build and upload a working firmware.

Traumflug, 2015-05-22:
Looks like all questions are solved. In case additional ones come up, please open a new issue.

closed 2015-05-22 by Traumflug