Traumflug/Teacup_Firmware #130: Fix createThermistorTable.py

Traumflug, 2015-04-28:
Rename createThermistorTable.py to configtool/createthermistortable.py. Nope.
There's an even better algorithm:
http://hydraraptor.blogspot.co.uk/2012/11/more-accurate-thermistor-tables.html

Also clean this file up, align to 80 chars/line and convert tabs to spaces.

jbernardis, 2015-04-29:
I just took a quick look at Nophead's algorithm here.  This algorithm does not use the beta rating at all, but bases its results at three measured temperatures and the resistances at those temps.  This is significantly different from the current algorithm we have implemented.  I'm not saying we can't do it; rather I'm saying that it would be difficult to do both; if we decided to go this route, then I would remove the current logic and instead of asking for R0/Beta/R2/Vadc, I would ask for T0/R0/T1/R1/T2/R2.

Do we want to do this?  I know that the EPCOS 100K datasheet has a table giving the necessary information, but I don't know if this will be the case for other thermistors.  I guess if we could build up a good set of presets, then this would mitigate my concern that we're making it too difficult for users.

Traumflug, 2015-04-29:
Hmm. So the traditional algorithm uses two temp/value pairs, Steinhart-Hart uses three. Beta is a value which can be calculated from the two pairs: http://reprap.org/wiki/MeasuringThermistorBeta#Calculate_Beta

Looking at such data sheets it's not exactly idiot proof to calculate these three temperature pairs: http://sensing.honeywell.com/index.php?ci_id=3108&la_id=1&pr_id=53807

Looking at product pages like http://mendel-parts.com/en/parts/extruder-heater-set-v2.html or http://3d-industries.myshopify.com/collections/hotends/products/aluhotendv6 vendors couldn't care less about telling anything about the used thermistor or neccessary firmware settings.

My gut feeling says we have to at least keep this Beta thing, because it's so commonly used.

----
> instead of asking for R0/Beta/R2/Vadc, I would ask for T0/R0/T1/R1/T2/R2

These two *R2* are not to be confused here. The left R2 is this comparison resistor on the controller board, the right R2 is the thermistor resistance at elevated temperature. Accordingly, for the three pair algorithm you'd need T0/R0/T1/R1/T2/R2/R2/Vadc. In [Nophead's script](https://github.com/nophead/Marlin/blob/Marlin_v1/Marlin/MakeTempTable.py) the second R2 is called *Rp*, "pullup up resistor".

jbernardis, 2015-04-29:
Presets are working - I just have to build up a list of them.  Also, I unify tables that are constructed with equivalent parameters.

Let me spend a bit of time seeing if I can figure out a good way to support both models so the user can enter data either way.  I'm thinking I could have 2 sets of fields which are mutually exclusive based on a pair of radio buttons.  One set of fields requires the 4 pieces of data; the other requires the 6 pieces of data.  The user can select a radio button and enter the corresponding data.  I can even see how this might work in conjunction with the presets.

BTW - I realize that the r2's are different.  Like you said, in Nophead's model, it's just another data point on the resistance/temperature curve.  In the beta model, it is the pull-up resistor (which nophead calls rp)

jgrjgr, 2015-04-30:
for some of these thermistors it's actually quite difficult to get a datasheet or much data at all, i'm using the rs thermistor which is #10 in marlin ,  rs don't even list a beta value for it , or a datasheet

this is actually all they give you:

Dimensions	2 Dia. x 4.2mm
Length	4.2mm
Maximum Operating Temperature	+250°C
Minimum Operating Temperature	-40°C
Package/Case	DO-35
Resistance @ 25 °C	100kΩ
Temperature Coefficient Type	NTC
Thermal Time Constant	7s
Tolerance	±10%

jbernardis, 2015-05-06:
I totally reworked this in branch configtoolPhaseB.  I no longer use this file.  I've extracted the logic I need for my purposes, and I leave it to you to keep this or not for those who want a command line interface.

Traumflug, 2015-05-22:
Hmm. Just tried to verify the new tool against the old one and results are ... well, different.
```
$ ./createTemperatureLookup.py --r0=100000 --t0=25 --r1=0 --r2=4555 --beta=4092 --max-adc=1023 --num-temps=25 --min-adc=1 --max-adc=1023
[...]
const uint16_t temptable[NUMTEMPS][2] PROGMEM = {
// {ADC, temp*4 }, // temp         Rtherm     Vtherm      resolution   power
   {   1,   3324}, //  831.25 C,        4 Ohm, 0.005 V, 174.25 C/count, 0.01mW
   {  43,   1087}, //  271.85 C,      200 Ohm, 0.210 V, 1.79 C/count, 0.22mW
   {  85,    895}, //  223.84 C,      412 Ohm, 0.415 V, 0.78 C/count, 0.42mW
   { 127,    792}, //  198.23 C,      645 Ohm, 0.620 V, 0.49 C/count, 0.60mW
   { 169,    723}, //  180.79 C,      900 Ohm, 0.825 V, 0.36 C/count, 0.76mW
   { 211,    669}, //  167.47 C,     1182 Ohm, 1.030 V, 0.28 C/count, 0.90mW
```
... and with Configtool:
```
const uint16_t PROGMEM temptable[NUMTABLES][NUMTEMPS][2] = {
  // EXTRUDER temp table using Beta algorithm with parameters:
  // R0 = 100000, T0 = 25, R1 = 0, R2 = 4555, beta = 4092, maxadc = 1023
  {
    {  26,  1248}, //  312 C,    119 ohms, 0.127 V, 0.14 mW
    {  30,  1196}, //  299 C,    140 ohms, 0.146 V, 0.15 mW
    {  36,  1144}, //  286 C,    165 ohms, 0.176 V, 0.19 mW
    {  42,  1092}, //  273 C,    197 ohms, 0.205 V, 0.21 mW
    {  50,  1040}, //  260 C,    236 ohms, 0.244 V, 0.25 mW
    {  60,   988}, //  247 C,    286 ohms, 0.293 V, 0.30 mW
    {  73,   936}, //  234 C,    350 ohms, 0.356 V, 0.36 mW
```

How could we find out which one is closer to the truth?

Also have to check wether interpolation still works with ADC values out of range, because the new ones start at 26 and stop at 1011.

jbernardis, 2015-05-22:
I'm fairly certain that the tables are accurate - look at the rows for adc 42/43 and you can see that they are very close in temperature.   The way to really tell is to increase the number of lines in the table with the numtemps setting and see how well they correlate,  In my testing, I created large tables so that I could do this check and it all seemed to agree.  It's the exact same code used in the actual adc to temperature calculation.

As far as the min and max ADC values.  The value of 26 corresponds to 312 degrees - there is no point in going further because this is already higher than we're interested in (you can see in the other table that 11 corresponds to 800+ degrees - no point in going there).   Similar with the larger ADC values - you'll eventually get to a temperature that is below what we care about - also, if you go too far you get negative temperatures which I've already noted throws off the calculations because of the use of unsigned variables.

What I can do is add an entry to the beginning of the table for adc=0 (or should it be 1) and add an entry to the end of the table with the ADC value that results in 0 degrees.  I'm not sure it's really necessary though.

Traumflug, 2015-05-22:
> I'm fairly certain that the tables are accurate

I agree now. I made a temptable with all 1023 values with the older tool and this is what I get (excerpt):
```
   {  26,   1249}, //  312.42 C,      119 Ohm, 0.127 V, 3.39 C/count, 0.14mW
   {  30,   1201}, //  300.35 C,      137 Ohm, 0.146 V, 2.82 C/count, 0.16mW
   {  36,   1142}, //  285.60 C,      166 Ohm, 0.176 V, 2.24 C/count, 0.19mW
   {  42,   1094}, //  273.63 C,      195 Ohm, 0.205 V, 1.84 C/count, 0.22mW
   {  50,   1042}, //  260.61 C,      234 Ohm, 0.244 V, 1.48 C/count, 0.25mW
   {  60,    990}, //  247.53 C,      284 Ohm, 0.293 V, 1.18 C/count, 0.30mW
```
Values aren't equal, but fairly close.

> As far as the min and max ADC values.

It's true that there's not much interest in temperatures below 20 °C and above the thermistors specified range, but we should have a defined behaviour there. Will look into this now.

Traumflug, 2015-05-22:
> As far as the min and max ADC values.

Here we go. I tweaked analog_read() to send consecutive numbers starting at 0 instead of real values, enabled some of the debugging sersendf_P()s in temp.c and ran this thing in the simulator.

Good news:
- The algorithm manages to extrapolate before the first entry (using the first pair of entries).
- Everything at or beyond the last entry is clamped to the value of the last entry.
- We're safe. Excellent.

For those interested, table was made with this, 25 entries:
// R0 = 100000, T0 = 25, R1 = 0, R2 = 4555, beta = 4092, maxadc = 1023

Excerpt from the result:
```
Raw ADC:0 table entry: 1 temp:396.50
Raw ADC:1 table entry: 1 temp:393.25
Raw ADC:2 table entry: 1 temp:390.0
Raw ADC:3 table entry: 1 temp:386.75
Raw ADC:4 table entry: 1 temp:383.50
...
Raw ADC:28 table entry: 1 temp:305.50
Raw ADC:29 table entry: 1 temp:302.25
Raw ADC:30 table entry: 2 temp:299.0
Raw ADC:31 table entry: 2 temp:296.75
Raw ADC:32 table entry: 2 temp:294.50
Raw ADC:33 table entry: 2 temp:292.50
Raw ADC:34 table entry: 2 temp:290.25
Raw ADC:35 table entry: 2 temp:288.0
Raw ADC:36 table entry: 3 temp:286.0
...
Raw ADC:1007 table entry: 24 temp:4.0
Raw ADC:1008 table entry: 24 temp:3.0
Raw ADC:1009 table entry: 24 temp:2.0
Raw ADC:1010 table entry: 24 temp:1.0
 (clamp) temp:0.0
 (clamp) temp:0.0
 (clamp) temp:0.0
```

jbernardis, 2015-05-25:
Can this be closed?   Or are there still unanswered questions?

Traumflug, 2015-05-25:
I think it can be closed.

closed 2015-05-25 by Traumflug