Traumflug/Teacup_Firmware #131: Configtool: uploading can't be aborted

Traumflug, 2015-04-28:
Uploading can't be aborted. For example, uploading with a mismatching baud rate
causes avrdude to try over and over again, for minutes, and there's no way to
stop that. The only way to re-gain control is to kill avrdude from the command line.

jbernardis, 2015-05-19:
I added branch issue131 to address this issue.  I'm fairly confident on Linux, but I'm not certain about the degree of process control that Windows gives you, and I don't really have the setup to test under windows.  The python API is basically the same, but I'm not sure if the underlying implementation is.  I'd be glad if someone with a windows setup could put this through the paces.

There is no new button for cancellation.  Just push the "close window" button.  If the upload process is active, it will warn you and give you a change to change your mind.

Traumflug, 2015-05-20:
Excellent! Really nice user experience. Picked to experimental and removed the issue131 branch.

closed 2015-05-22 by Traumflug