Traumflug/Teacup_Firmware #132: Configtool: absolute paths for _all_ files.

Traumflug, 2015-04-28:
I had a few occasions where a firmware wouldn't upload with "teacup.hex: no such file or directory".

Also a few cases where gcc complains about missing .al files. It still builds, then, but that's not nice.

Unfortunately, I found no clear case to reproduce this, so far. I think the best solution would be to pass all file names as absolute paths to the tools. That's done for .c and .o files already, at least .al and the name of the binary are missing.

Watching build and upload logs (this window appearing automatically) should show which files are missing.

jbernardis, 2015-04-29:
This issue is addressed in branch configtoolPhaseA

Traumflug, 2015-05-04:
For the records, during review I catched an instance while trying to upload, right after building:
```
avrdude -c stk500v2 -b 115200 -p atmega1284p -P /dev/ttyUSB0 -U flash:w:teacup.hex
avrdude: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.01s
avrdude: Device signature = 0x1e9705
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "teacup.hex"
avrdude: error opening teacup.hex: No such file or directory
avrdude: input file teacup.hex auto detected as invalid format
avrdude: can't open input file teacup.hex: No such file or directory
avrdude: read from file 'teacup.hex' failed
avrdude: safemode: Fuses OK (E:00, H:00, L:00)
avrdude done.  Thank you.
RC = 1 - Build terminated
Upload terminated abnormally.
```

jbernardis, 2015-05-04:
Yes - I realized I forgot that change when I separated my fixes last week.  I have a working version.  My plan is to test what I have for another day or two and then push up another branch that has the changes for temperature sensors.  The dialog box is much more dynamic and it supports defining thermistors with either the Beta algorithm or the Steinhart-Hart algorithm.  I think it works pretty well, but I'm anxious for others to see it.  It also supports presets for thermistors using either algorithm.

Traumflug, 2015-05-04:
Sorry, I forgot to mention your big commit was only half applied.

Traumflug, 2015-05-04:
P.S.: my suspicion is, Configtool somehow looses or changes its working directory. No idea where or why this happens.

Traumflug, 2015-05-04:
> I'm anxious for others to see it.

Huh? Is it so brilliant that one risks to get blinded? I can take my sunglasses. :-)

> It also supports presets for thermistors using either algorithm.

Sounds very good!

jbernardis, 2015-05-04:
I agree with you.  Although I can think of where I'm changing directories, I'm pretty sure that's what's happening.  I have to check, but I think some of the file open dialog boxes might change into the directory where a file is chosen as default behavior.  Using full path names is the safe path.

If the big commit was only half applied, then I'd say wait until I push up my next changes.  It is based back on the experimental branch, so everything that was in configtoolPhaseA will also be in this one.  I will commit to pushing it up by EOD wednesday (East Coast US time).  I want to get it up there because I am going to be away next week and I don't want you to have to wait for another week and a half.

Traumflug, 2015-05-05:
> It is based back on the experimental branch, so everything that was in configtoolPhaseA will also be in this one.

That's no problem. Rebasing will detect and eliminate what's already there.

Actually, that's how I review these commits:

1. Cherry-pick the commit.
2. Find a topic to review.
3. Step through the changes, evaluate them, revert everything off-topic.
4. Test the topic against the issue here.
5. Rebase the topic branch, which reduces this branch by the picked stuff and keeps the stuff which was reverted.
6. Start over at 1. (Yes, some commits get picked multiple times, until they're empty).

I'm comfortable with it. Since these whitespace differences are gone It works effectively.

jbernardis, 2015-05-06:
Branch configtoolPhaseB addresses one overlooked scenario for this issue

Traumflug, 2015-05-08:
I see you also moved to absolute paths inside config.h. This doesn't hurt (except for some readability), but doesn't help either. If this were needed, none of the local includes would work. I'll make a patch from this commit and move it to attic/ in case it's ever needed.

The change to .al files looks right.

closed 2015-05-19 by Traumflug