Traumflug/Teacup_Firmware #134: Configtool: there should be a menu with presets for various known thermistors

Traumflug, 2015-04-28:
There should be a menu with presets for various known thermistors. If that's too complicated, at least a list of popular types in the tooltip.

Most people simply know "I have a 100k EPCOS" and never heard what "beta" is. Help them :-)

Traumflug, 2015-04-28:
Continued from issue #76:

> I'm one of those users who only know about the EPCOS 100K

You see, searching on Mouser for "EPCOS 100k" results in 64 hits ... http://www.mouser.de/Passive-Components/Thermistors-NTC/_/N-6g7mw?P=1yzotm4&Keyword=epcos+100k&FS=True

That said, the apparenly only ones capable of doing 300 °C are the B57560G1104+000 and B57560G1104+002 , which both have a beta (more precisely: B25/100) of 4092.

This page lists more types together with beta values: http://reprap.org/wiki/Thermistor And of course there's our collect-everything Marlin, which doesn't list all beta values, but some part numbers for lookup (starting at line 100): https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Configuration.h

Looking at the latter it even looks like the thermistor code works for PT100 and PT1000 type sensors, too, but that's another story :-)

jbernardis, 2015-05-06:
This issue is addressed by branch configtoolPhaseB

closed 2015-05-22 by Traumflug