Traumflug/Teacup_Firmware #139: Configtool: config.h (or config.default.h) should load automatically on startup

Traumflug, 2015-04-28:
Pure convenience. When starting the tool it's very likely one wants to edit the current configuration, so load this without asking. Even if one wants to start over it's not more work to do so whith this happening.

jbernardis, 2015-04-29:
This issue is addressed in branch configtoolPhaseA

Traumflug, 2015-04-29:
Jeff, you're incredible. :-)

jbernardis, 2015-04-29:
I missed that config.default.h should load if there is no config.h.  I  will include that in the next delivery..

I'm working on thermistor presets at the moment.   I have it working.   I just need to figure how to best unite identical tables. 

One question.   If the sensor type is not thermistor,  what is the macro expecting as the "additional"  parameter.   I tried nothing both with and without a comma,  and an empty string.   All failed to compile.   0 seemed to satisfy the compiler,  but I don't know if it produced the desired result.

Traumflug, 2015-05-08:
> I missed that config.default.h should load if there is no config.h.

I don't think it's missing. config.default.h is close to useless, because we can't predict which kind of printer/board a user has. With this in mind, enabling only "Load board" and "Load printer" with no user-generated config.h present is the right thing to do, IMHO. That's what currently happens.

The only reason I didn't remove config.default.h entirely is, it's a template for those configuring/compiling without Configtool.

jbernardis, 2015-05-08:
I actually added the read of config.default.h in my last commit.  I can remove it easily enough if you think it's the right thing to do.

Traumflug, 2015-05-08:
> I can remove it easily enough if you think it's the right thing to do.

Thanks, I already did. I'm mentioning such stuff here to explain why some snippets of your fabulous work don't enter the experimental branch and also to listen to others. Perhaps there's somebody out there who finds config.default.h important. In such a case it's a good idea to speak up now :-)

closed 2015-05-10 by Traumflug