Traumflug/Teacup_Firmware #140: Emergency Stop Button Addition

youssefaly97, 2015-05-04:
Hello, I suggest to add a hardware emergency stop feature which can easily done just by shutting down the power supply through the PS-on wire but I want to add the feature that it sends a pause command back to host software or pause the G-Code from the SD Card and save the line number it paused at to resume at this line back when the error is neutralized.

Traumflug, 2015-05-04:
The reset button is so far considered to be the emergency stop button.

What you describe sounds more like a pause button. If you want to implement one, the implementation of endstops in dda_clock() should give a good example. Denoising the button is important, else you might get an accidental pause as result of some random EMF radiation. Other that that, yes, go ahead.

youssefaly97, 2015-05-20:
Here's how I'm going to try and implement it:
1-debounce button in the same way as the endstops
2-delay command acknowledgement to host software if not already delayed
3-decelerate or freeze all axes (freeze or deceleration selection will be in config.h)
4-shutdown power supply
5-locally pause the g-code received while maintaining buffer data
6-resume when the emergency button has been twist-to-release
7-when the buffer is partially empty resume acknowledgements

Tell me if I missed or if I should change anything.

Traumflug, 2015-05-20:
Sounds reasonably. Exceptions:

- 3: freezing, like stopping immediately, is not only harmful for the mechanics, it's also a warranty for step losses. Proper deceleration is a must.
- 4: Shutting down the PSU is probably not a good idea, because this way you easily loose position.
- 7: I see no need to wait for the buffer to partially drain.

A more simple implementation could be to simply stop accepting new G-code. Movements in the buffer get executed then, of course, still this might be more helpful than having nothing.

youssefaly97, 2015-05-20:
I agree:
3-Totally agree
4-store current position then shut down power supply. When resuming, home all axes and re-position them at the stored position. Can't risk failing stepper driver for example ruining rest of mechanics.
7-for that, I think to add some kind of acknowledgements-delay over ride, turn it on in step 2 and turn it back off in step 7.

I want to make a full implementation that is optional by defining the button or not in config.h if some user don't want to use it.

Traumflug, 2015-05-21:
> in config.h

We no longer put configuration stuff into config.h, there's now an entire folder (config/) with such descriptions and a GUI tool to edit them.