Traumflug/Teacup_Firmware #142: Configtool: move protected files list and thermistor presets into configtool.default.ini

Traumflug, 2015-05-14:
While reviewing this big change to support the Steinhart-Hart algorithm (BTW, works beautifully :-) ), I found that to list of protected files was moved from configtool/protectedfiles.py to configtool.default.ini. This obviously works fine for newbies, but if a configtool.ini exists already, changes in the distribution, e.g. an new template for a printer, go unnoticed. Similar for Thermistor presets.

As this unlikely happened by accident, I'm wondering now what the purpose of this move was.

Side note: I've separated the protected files list part into its own commit already, so no coding neccessary in either case (keeping/dropping this change).

jbernardis, 2015-05-15:
When adding the presets,  I simply put them in the ini file.   Once I did this,  the protected files list seemed out of place.   I actually moved it for consistency :)  

I do see your point though,  and propose moving them both into a second ini file that can only be altered through external editing.   If you agree with this,  I can do it as soon as I get home from vacation.

BTW,  I like the new way protected files are identified by just listing the boards and printers.

jbernardis, 2015-05-19:
I posted branch issue142 which splits the settings into two files.  I'm not sure I have the right file names - these are debatable - but the current file - configtool.default.ini contains settings that the user can modify via the GUI.  If they so modify, the changes will be written to configtool.ini.  The default file will never get written to by the configtool.

The second file - configsettings.ini - contains the protected files and the thermistor presets.  This file is never presented to the user through the GUI and there is no way to over-write it from within the GUI.

The addiitonal change is that I added ALL of the fields in configtool.default.ini/configtool.ini to the edit-settings dialog box, and I added help text for all of them.

Traumflug, 2015-05-20:
Quick status update:

The commit on the issue142 branch didn't do what it's expected to do, because you moved the list of protected files into configsettings.ini, but forgot to actually use this list and kept config/protectedfiles.py instead. As I had a solution using less code in my head already I implemented it, using a file named thermistorpresets.py. This is currently on top of the experimental branch in case you want to see it.

Now I feel how you're more comfortable with a .ini file instead of a pair of .py files, Jeff. With this in mind I think I'll put together yet another commit doing what you originally intended. The neccessary bits are lying around in various commits.

Also not yet put onto experimental are the additions to the user configurable settings. Just a matter of "not yet", it'll follow tomorrow.

Traumflug, 2015-05-20:
Hmm. Looking at this commit menioned before my last post https://github.com/Traumflug/Teacup_Firmware/commit/e1277e4f4a250e245345d2471c6171c85079ec65 , I edited this so much I should have made it under my own name. Sorry for this.

Anyways, thanks for the good work. I'm sure things will align nicely before too long.

iosdev1024, 2015-05-20:
Hello,

This is probably not the right group for this questions, but for learning purposes, I was trying to make a simple command line ui for simulavr using c/c++. This command line program would display the AVR pin states, and allow the user to set or change any pin from outside the running avr simulation.

I have seen several TCL examples, but I dont know TCL/Qt language, so I cant understand it (been trying for a week already). 

I was wondering if anybody has a simple c/c++ version of the tclavr ui that can display pins and allow the user to change them while the avr simulation is running.

Thanks
iosdev

Traumflug, 2015-05-20:
@iosdev1024, you hijacked a bug discussion. At very very least open a new issue. I'll remove your and this post in a minute.

Traumflug, 2015-05-22:
The more I look into these things, the more I see room for improvement. For example, this *maxadc* value isn't really a choice of the user, but a property of the MCU.

But ... but ... I think it's a good idea to stop on improvements and new features now. Let's first see what users say. Accordingly I simply moved all remainings of this thermistor table stuff into a branch *configtool-leftover* and removed all others. Things work, that's the most important aspect.

jbernardis, 2015-05-22:
I agree with you 100%.  I recognized that the use of the adc values is inconsistent between the two algorithms and - yes - it is an attribute of the MCU and so should probably go on one of the "board" pages.

Traumflug, 2015-05-25:
Removing this from the milestone because the milestone-relevant parts are solved. Some smaller bits maybe not.