Traumflug/Teacup_Firmware #143: Configtool(?): can't add solo temperature sensor

Traumflug, 2015-05-14:
##### Steps to reproduce:
1. Take a board file, e.g. the one for Gen7 v1.3.
2. Add just a temperature sensor (AD595 or Thermistor, doesn't matter) with name "bla".
3. Save unter a different name, add a printer, etc.
4. Build.

##### Results:
```
avr-gcc -c -DF_CPU [...] "/home/mah/RepRap/Teacup_Firmware/temp.c"
In file included from /home/mah/RepRap/Teacup_Firmware/config.h:3:0,
                 from /home/mah/RepRap/Teacup_Firmware/config_wrapper.h:12,
                 from /home/mah/RepRap/Teacup_Firmware/temp.c:63:
/home/mah/RepRap/Teacup_Firmware/config/board.mydevelopment.h:152:26: error: ‘HEATER_bla’ undeclared here (not in a function)
 DEFINE_TEMP_SENSOR(bla,      TT_THERMISTOR, AIO1,  THERMISTOR_BLA)
                          ^
RC = 1 - Build terminated
Compile terminated abnormally.
```
----
##### Discussion
I'm not sure wether this is a fault of Configtool or the code in temp.c. It should be possible to add a sensor without corresponding heater. Just forgot how and can't dive into this issue right now, so I'm jotting down this issue here to make sure it doesn't get forgotten.

Traumflug, 2015-05-22:
Investigated this a bit:

- To define a sensor without a heater, it currently has to be named "noheater".
- Currently there's only one such solo sensor possible, two sensors with this name clash.
- Accordingly it makes no sense to offer arbitrary names for temp sensors, only "noheater" and the names of already defined heaters make sense.

Will continue on this issue tomorrow, it's late here.

Traumflug, 2015-05-24:
Reflecting the "solution" to issue #96 I've changed Configtool to allow only temperature sensor names which match a heater name (or "noheater"). The text input field was replaced with a menu choice.

Code is on the [tempsensordefinition branch](https://github.com/Traumflug/Teacup_Firmware/commits/tempsensordefinition) (last 3 commits). Please review.

jbernardis, 2015-05-26:
I think this is a reasonable way to do this.  Your code looks fine to me.

Traumflug, 2015-05-26:
Thanks for the review. Picked to experimental and removed the tempsensordefinition branch.

closed 2015-05-26 by Traumflug