Traumflug/Teacup_Firmware #146: simulavr UI

iosdev1024, 2015-05-20:
Hello simulavr experts

For learning purposes I was trying to learn how to create a simple UI for simulavr using c/c++ (since I dont know Python/TCL/QT/Perl or any of such languages I only know c/c++ etc).

The UI I was going to make is just a simple command line UI that can show the state of any pin on the AVR (both analog and digital), and also allow the user to change any pin (also both analog and digital), during the AVR simulation.

I looked at the ui.cpp code in simulavr, but I cant figure out how to interface to it, is it using c/c++ sockets?

Does anybody have any simple UI code written in C/C++ for interfacing to SIMULAVR? or maybe you can write a short pseudo code for me in c/c++ to start me off, since I have no clue how to do it.

I would be so grateful.

Thanks
iosdev.

Traumflug, 2015-05-20:
> Hello simulavr experts

SimulAVR experts are on the SimulAVR mailing list ... and you did ask the same there already, right?

That said, we have an example here which reads some pins, all put together in one Bash script:

https://github.com/Traumflug/Teacup_Firmware/blob/experimental/testcases/run-in-simulavr.sh

iosdev1024, 2015-05-21:
Hello Traumflug,

Thanks for the bash script, I took a look at it even though I dont know shell language.

Yes I did ask over in the simulavr group, but no good response so far, so I decided to ask over here.

Seems a C/C++ UI to simulavr is hard to come by, even for just reading or writing the AVR portds.

The bash script you gave doesnt seem to change any pins (i cant really be sure since I dont know shell).

Also I thought the "-u" had to be specified on the simulavr command line, but that switch seems its not present in the bash script.

So sorry to bother you, but do you know how to do it using c/c++? I just need to write a command line ui to display any pin, or to change any pin, during the simulation, thats all.

thanks

Traumflug, 2015-05-21:
> I dont know shell language

Do you know anything?

Closing, because off-topic for this project.

closed 2015-05-21 by Traumflug