Traumflug/Teacup_Firmware #147: Board and Printer configuration files should not be C header files

jbernardis, 2015-05-22:
This is not really an issue but an improvement I've been thinking about,  We can discuss here.

Parsing the C header files is somewhat ambiguous and artificial logic is in place to try to be tolerant of something a user might edit.   Also, we are coming up with different parameters that do not end up in the actual header file but need to be carried as comments so that the configtool can remember them from run to run (for example temperature table parameters).

I am proposing that the board and printer C header files be replaced with XML files,  The configtool would load the files at the users direction, and would produce 1) a single config.h file with all of the settings needed by the compiler, and 2) the temperature table file. 

As an example, I have encoded a board file below.  I am not suggesting this as the final syntax - just a starting point:

```

<board>
	<name>jjb</name>
	<cpu>
		<description>
			CPU types a user should be able to choose from in configtool.
		</description>
		<options>atmega1280,atmega2560</options>
		<value>atmega2560</value>
	</cpu>
	<cpuclock>
		<description>
			CPU clock frequencies a user should be able to choose from in configtool.
		</description>
		<options>16000000UL</options>
		<value>16000000UL</value>
	</cpuclock>

	<baud>
		<description>
			Baud rate for the serial RS232 protocol connection to the host.
			Usually 115200, other common values are 19200, 38400 or 57600.
			Ignored when USB_SERIAL is set to true.
		</description>
		<value>115200</value>
	</baud>

	<xonoff>
		<description>
			Redundant when using RepRap Host for sending G-code, but mandatory when
			sending G-code files with a plain terminal emulator, like GtkTerm (Linux),
			CoolTerm (Mac) or HyperTerminal (Windows).
		</description>
		<value>false</value>
	</xonoff>

	<usbserial>
		<description>
			Define this for using USB instead of the serial RS232 protocol. Works on
			USB-equipped ATmegas, like the ATmega32U4, only.
		</description>
		<value>false</value>
	</usbserial>

	<pins>
		<x_step_pin>DIO54</x_step_pin>
		<x_dir_pin>DIO55</x_dir_pin>
		<x_min_pin>DIO3</x_min_pin>
		<x_max_pin></x_man_pin>
		<x_enable_pin>DIO38</x_enable_pin>
		<x_invert_dir>false</x_invert_dir>
		<x_invert_min></x_invert_min>
		<x_invert_max></x_invert_max>
		<x_invert_enable>true</x_invert_enable>

		<y_step_pin>DIO60</y_step_pin>
		<y_dir_pin>DIO61</y_dir_pin>
		<y_min_pin>DIO14</y_min_pin>
		<y_max_pin></y_man_pin>
		<y_enable_pin>DIO56</y_enable_pin>
		<y_invert_dir>false</y_invert_dir>
		<y_invert_min></y_invert_min>
		<y_invert_max></y_invert_max>
		<y_invert_enable>true</y_invert_enable>

		<z_step_pin>DIO46</z_step_pin>
		<z_dir_pin>DIO48</z_dir_pin>
		<z_min_pin>DIO18</z_min_pin>
		<z_max_pin></z_man_pin>
		<z_enable_pin>DIO62</z_enable_pin>
		<z_invert_dir>false</z_invert_dir>
		<z_invert_min></z_invert_min>
		<z_invert_max></z_invert_max>
		<z_invert_enable>true</z_invert_enable>

		<e_step_pin>DIO26</e_step_pin>
		<e_dir_pin>DIO28</e_dir_pin>
		<e_enable_pin>DIO24</e_enable_pin>
		<e_invert_dir>false</e_invert_dir>
		<e_invert_enable>true</e_invert_enable>

		<ps_on_pin></ps_on_pin>
		<ps_mosfet_pin></ps_mosfet_pin>
		<stepper_enable_pin></stepper_enable_pin>
		<stepper_invert_enable>false</stepper_invert_enable>

		<debug_led_pin></debug_led_pin>
	</pins>

	<sensors>
		<!-- MAX6675, AD595, PT100, INTERCOM -->
		<options>THERMISTOR</options>
		<pinoptions>AIO13,AIO14,AIO15</pinoptions>
		<sensor>
			<name>extruder</name>
			<type>THERMISTOR</type>
			<pin>AIO13</pin>
			<table>
				<type>beta</type>
				<r0>100000</r0>
				<beta>4092</beta>
				<r2>4700</r2>
				<vadc>5.0</vadc>
			</table>
		</sensor>
		<sensor>
			<name>bed</name>
			<type>THERMISTOR</type>
			<pin>AIO14</pin>
			<table>
				<type>beta</type>
				<r0>100000</r0>
				<beta>4092</beta>
				<r2>4700</r2>
				<vadc>5.0</vadc>
			</table>
		</sensor>
	</sensors>

	<heaters>
		<pinoptions>DIO4,DIO5,DIO8,DIO9,DIO10</pinoptions>

		<heater>
			<name>extruder</name>
			<pin>DIO10</pin>
			<pwm>true</pwm>
		</heater>

		<heater>
			<name>bed</name>
			<pin>DIO9</pin>
			<pwm>true</pwm>
		</heater>

		<heater>
			<name>fan</name>
			<pin>DIO8</pin>
			<pwm>true</pwm>
		</heater>
	</heaters>
</board>
```

Traumflug, 2015-05-22:
Hmm. This makes it more difficult for those building without Configtool. Actually, I'm excited how well parsing the header file works.

Traumflug, 2015-05-22:
Thinking of it, you undoubtly had something in mind with this. What would be the advantages?

About the future of Configtool: besides further simplification of the UI I could see a third tab next to printer and board with tools to do proper calibration. At the right of this tab some shell to send G-code manually and to see the responses. To the left a few buttons to test endstops, stepper movements, etc., together with instructions on how to test/calibrate them.

jbernardis, 2015-05-22:
I don't want to rock the boat too much.  I'm happy to leave it as it is.  I just thought divorcing the configuration from the C syntax and imposing a little structure to it in the form of XML syntax would make it easier to adapt to future changes.

It wouldn't be a trivial change.  It just illustrates what I've long believed - software is never finished - you can just keep tinkering with it.  

But it ain't broke, so we won't "fix" it.

RaD, 2015-07-09:
I guess, the python scripts are much better to make all preparation work to configure and test firmware.