Traumflug/Teacup_Firmware #148: Configtool: some mild formatting of the help texts collected from header files

Traumflug, 2015-05-25:
At least a strip() and removing not all, but most newlines. Tooltips don't size after line length, but have their own idea on how wide to make the box.

Traumflug, 2015-05-25:
Thanks to the nice design of Configtool, doing this was no rocket science. Code is on the new issue148 branch, please review.

jbernardis, 2015-05-26:
I did a quick look at this commit.  Looks like it was more work in the header files than in the actual configtool code. 

I think this looks fine.

Traumflug, 2015-05-26:
> Looks like it was more work in the header files than in the actual configtool code.

Yep, even a bit more with radio button tooltips now working. :-)

Thanks for the review, picked to experimental and removed the issue148 branch.

closed 2015-05-26 by Traumflug