Traumflug/Teacup_Firmware #149: Configtool: radio buttons show no tooltips

Traumflug, 2015-05-25:
Stepped a few hours through the code now and to all what I can see, radio buttons get a tooltip attached properly. But for some unknown reason they don't show up. It works to set a tooltip onto the surrounding StaticBox. Then the title of this box as well as all radio buttons get the same tooltip.

Now, the general code design of wxPython and Configtool apparently doesn't provide a connection between a StaticBox and its radio buttons. Parent of these buttons is the page panel its self. Accordingly I'd have to reach parameters around to set a tooltip to the surrounding box instead of the button it's meant for.

What might go wrong here? Both radio button boxes are on the printer panel, one on the Mechanical, one on the Acceleration tab.

jbernardis, 2015-05-25:
I've  noticed this behavior.   I'm away from home for the next few hours,  but when I get home later I can look for a good way to accommodate this.

jbernardis, 2015-05-26:
Interestingly enough, this behavior is only evident on Linux.  Windows shows the help text properly.

In any case, I have put up branch radiobuttontooltip which has changes that will attach the help text to the enclosing static text box.  The static box tool tip is set repeatedly for every button that is in that box, so only the help text for the last button will show, but thus far, all radio buttons have the same text - so this will not be an issue.

When making this change, I noticed that the config file for the mendel and wolfstrap printers (the only 2 printers in the repository) had a problem where they defined the values for kinematics.  Those two changed files are also in this commit.  Otherwise, only accelerationpage.py, mechanicalpage.py and page.py changed.

Traumflug, 2015-05-26:
The commit works just fine! Thank you very much. Picked to experimental and removed the radiotooltip branch.

closed 2015-05-26 by Traumflug