Traumflug/Teacup_Firmware #150: Configtool: background picture as dessert topping

Traumflug, 2015-05-25:
As about all known bugs are solved now, my idea was to put a nice big (mostly white + light colors) Teacup logo into the background, so all the controls appear on top of this. So I started playing and getting these controls white isn't difficult. Even shadows, borders and whatnot still display nicely. Drawing the background picture was not much of a problem either.

However I pulled my hair out on getting all these panel backgrounds transparent to let the background shine through. I managed to custom draw them, but transparency is apparently impossible. Accordingly, I see the background picture only on the small margin between the panel and the main window its self.

Is the conclusion right that this is a task wxPython isn't made for? The only way to achieve such an effect I can currently think of is to draw the very same background on all the panels, appropriately offsetted to appear as an undisturbed picture.

It's not much of a problem if such a background picture is too troublesome, it's just meant as some eye candy.

jbernardis, 2015-05-26:
What file type are you using for the graphic?  PNG files have an inherent transparency feature.  I'm not sure if this will render the way you want, but it might be worth a try.

Traumflug, 2015-05-26:
> What file type are you using for the graphic?

It's currently a PNG without transparency. The problem isn't that the PNG doesn't display, the problem is that it's 95% covered by the background of the panels in front of it.

This screenshot shows the current state, the PNG is some random Excel graph. 
[image]

jbernardis, 2015-05-26:
Perhaps you need to make the background on each page.   This only needs to be done in one place in the page class since all of the pages are subclasses of it.

jbernardis, 2015-05-26:
I have it working where it will display a png image in the background of each page.  It looks pretty good - except for the fact that the image shows through the static boxes I use to group controls.  I would like very much for the static boxes to be opaque, but I haven't found the way to do this yet.  This could be mitigated by using an image that is VERY light, but I'd sure like to solve the background color issue.

I'll commit something later this evening.

Traumflug, 2015-05-26:
Tadaaa!

Got it. Lots of wrestling with GetPosition, GetClientPosition and whatever shit, nothing worked as expected. Even wrote a loop climbing the window hierarchy, offsets for the inner panels were still wrong. wx.NoteBooks claim an offset, but don't paint it ?!?

Until I discovered ScreenToClient() and ClientToScreen(). Now there is one picture which is displayed in each panel/frame/whatever, with offsets matching to form a smooth picture. Code is on the *prettification* branch.

This picture looks a bit messy, due to the graphics contens of the PNG. The important thing is, all panel background pictures fit together.

[image]

Traumflug, 2015-05-26:
Tests went mostly well so far. Resizing the window basically works, just some refresh trigger is missing.

Traumflug, 2015-05-27:
Even transparency of the PNG works here on Linux, so a logo can be placed without changing background colors.

jbernardis, 2015-05-27:
For what it's worth, I just pushed up a new branch - backgroundimage - to attempt to do the same thing.

Traumflug, 2015-05-27:
Thanks. I see you bind to wx.EVT_ERASE_BACKGROUND to do the drawing. I tried the same, it has zero effect here on Panels. Works for Frames only. This makes me wonder wether my version works on your side.

jbernardis, 2015-05-27:
I can give yours a try when I get home tonight - it's all up on github I assume.

From the look of the graphic, it looks like you're putting the image onto the printer and board panels because I see that it spills over the page boundary.  I just added mine to each individual page.  Although they too are panels rather than frames, my approach seems to work there.  I just got lucky I guess - I simply copied the code from a web page that I googled.

Traumflug, 2015-05-28:
> I see that it spills over the page boundary.

It's put on all panels, not only the frontmost one. Offsetted in each to have the origin always at the same pixel relative to the whole window. I think this technique is called "pseudo transparency".

If a mechanical engineer tries to be artisic, this is the result for the theme "Teacup Firmware for RepRap":

[image]

jbernardis, 2015-05-29:
I think that's a cool logo and I think it will work well as a background.

We're almost there.

Traumflug, 2015-05-30:
Moved all this onto the experimental branch:

[image]

"Accidentially" also added another three commits, adding a modify button to the heaters list, and removing already used pins from the new/modify heater dialog. My local notepad is almost empty.

This issue done.

closed 2015-05-30 by Traumflug