Traumflug/Teacup_Firmware #151: Configtool doesn't work on Windows

Traumflug, 2015-05-26:
This is the second report I got about Configtool "not working" on Windows. Let me open this issue to track this, I consider it a serious one. See http://forums.reprap.org/read.php?13,512632,513519#msg-513519

Unfortunately neither of the two wrote a word about which kind of error happens. Apparently the application doesn't even start up to the GUI.

Traumflug, 2015-05-26:
... and yes, this unwillingness of people to just copy & paste the error message or to write a quick sentence about what's happening is driving me nuts. Both reporters write code, so they should know how important this simple task is. No idea on how to cure it.

jbernardis, 2015-05-26:
It was almost completely written and debugged on a Windows system,  and I just had it running there yesterday to verify the help text for radio buttons.  

I suspect the issue is that people haven't installed wxpython or something similar.   I should try to come up with a set of dependencies.

jbernardis, 2015-05-26:
The only dependencies for the configtool are 1: python 2.x (not 3.x), and 2: wxPython.  I tested with version 3.0 of wxpython, but there is nothing exotic going on so it might work with earlier versions as well.  Just make sure wxpython matches both your platform and you version of python. (i.e. do not install 64 bit wxpython if you only have 32 bit python - even on a 64 bit OS)

The only other thing I can think of is that on Linux, in order to upload to the board, one probably has to make sure their account is in the serial or dialout group.

Traumflug, 2015-05-26:
Thanks for the investigation, Jeff. Such situations make me wish this tool sends out a silent email with the error message ...

I'll try on an old Mac OS X tomorrow.

Traumflug, 2015-05-26:
> and 2: wxPython

That's it: http://forums.reprap.org/read.php?13,512632,513973#msg-513973

Traumflug, 2015-05-26:
Pronterface uses this to battle the same problem:
```python
try:
    import wx
except:
    print("wxPython is not installed. This program requires wxPython to run.")
    if sys.version_info.major >= 3:
        print("""\
As you are currently running python3, this is most likely because wxPython is
not yet available for python3. You should try running with python2 instead.""")
        sys.exit(-1)
    else:
        raise
```
Sounds good?

jbernardis, 2015-05-27:
The idea here is good, but do we want to preclude python3 independently of testing for wxpython? (actually I wasn't even aware that wxpython was not available for python 3)

```
if sys.version_info.major >= 3:
    print "python 3 not supported"
    sys.exit(-1)

try:
    import wx
except:
    print "wxpython needs to be installed..."
    raise
```

Given the fact that wxpython is not available for python3, this is probably a moot point.  I just present it the way I would normally think of it.

I think it's a good idea to have this test.

Traumflug, 2015-05-27:
Ive put two independent tests in two commits onto the new issue151 branch. Github mentions 4 commits above, because I slightly reworded it a minute after the first push.

Replaced the "raise" with a faked message to reduce visual clutter. I guess some people get scared solely from seeing a bunch of undecipherable text.

Looks good?

jbernardis, 2015-05-27:
Looks fine to me.  I agree with the simple message rather than the raise - too intimidating

Traumflug, 2015-05-30:
I think this is solved as far as possible.

closed 2015-05-30 by Traumflug