Traumflug/Teacup_Firmware #152: Configtool: no board/printer load on Mac OS X

Traumflug, 2015-05-27:
Sorry for nagging yet again :-} I just tried Configtool on my old iBook running Max OS X, with more recent (2.7.3) Python installed. 

The good news: it starts up fine! Takes a while, like 30 seconds, but it works. Very nice.

The bad news: when trying to load a board, things get stuck.
- File chooser doesn't open in config/, but in the folder configtool.py sits.
- It's not possible to change the folder. One can look into config/, though.
- All files are greyed out. Like not selectable.

[image]

jbernardis, 2015-05-27:
I have no idea what to do here and even if I had an idea, I have no way to test. The code clearly opens the config subdirectory.  The fact that yours is greyed out makes me wonder about file permissions?

Hopefully somebody who knows apple can give us something to work on here.

Traumflug, 2015-05-27:
After a round of printf-type debugging I found that wildcard support is apparently a bit limited. "*.h" works fine, "board.*.h" does not. Put a commit directly onto experimental to take care of this.

No solution for the wrong directory issue. If I remove self.dir entirely, I end up totally elsewhere, probably the folder last used by Python.

Now things appear to work at least for configuring. Building ... uh, this "Arduino Directory" setting is apparently ignored, so entering the path to Arduino there is of no use, the avr-gcc in there isn't found. And the error message is unfortunately only "An exception occured when trying to run avr-gcc ...". No hint on what actually went wrong.

Traumflug, 2015-05-28:
Created a new *toolpaths* branch to tackle this tool finding issue. Instead of looking at the platform, it now looks at wether arduinodir is set. This way one can build on Linux as well as on Mac OS X and choose on both, wether to use the system-wide tools or those inside Arduino.

Does it still work on Windows? I tried to take care of this.

One thing still left is that on Mac OS X the working directory is apparently lost. It says "arduino.h not found" or something. A "cd" at the start of the script should help. Tomorrow. It's already 2:20 AM here again, these small, quickly fixable issues keep me awake too long ... Good night :-)

Traumflug, 2015-05-28:
It turned out arduino.h wasn't found because the board config file was stored alongside config.h, not in config/. As this #include is apparently the only reason to not store board configurations outside of config/, I simply moved this include to config_wrapper.h, which has a fixed location. Voilá, one user-recognizable restriction less. :-)

Traumflug, 2015-06-05:
This one is solved as much as reasonably possible.

closed 2015-06-05 by Traumflug