Traumflug/Teacup_Firmware #158: Configtool: another config.h not found error

Traumflug, 2015-06-04:
See http://forums.reprap.org/read.php?245,516838,517132#msg-517132 , a first success of the announcements. German, but messages are copy&pasted.

With Configtool:
> F:\Program Files (x86)\Arduino -c -DF_CPU=20000000UL -mmcu=atmega1284p -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns="G:\3D\Firmware\Teacup neu\Teacup_Firmware\build\analog.al" -save-temps=obj -o "G:\3D\Firmware\Teacup neu\Teacup_Firmware\build\analog.o" "G:\3D\Firmware\Teacup neu\Teacup_Firmware\analog.c"
    Exception occurred trying to run

> F:\Program Files (x86)\Arduino -c -DF_CPU=20000000UL -mmcu=atmega1284p -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns="G:\3D\Firmware\Teacup neu\Teacup_Firmware\build\analog.al" -save-temps=obj -o "G:\3D\Firmware\Teacup neu\Teacup_Firmware\build\analog.o" "G:\3D\Firmware\Teacup neu\Teacup_Firmware\analog.c"
    Build terminated abnormally.

Using Arduino:
> In file included from /config_wrapper.h:20,
> from /temp.h:4,
> from analog.c:7:
> /config.h:3:32: error: config\board.taalu.h: No such file or directory
> /config.h:6:34: error: config\printer.taalu.h: No such file or directory
> In file included from /temp.h:4,
> from analog.c:7:
> /config_wrapper.h:26:4: error: #error config.h missing. Please follow instructions at reprap.org/wiki/Teacup_Firmware#Simple_Installation
> In file included from /temp.h:18,
> from analog.c:7:
> /config_wrapper.h:26:4: error: #error config.h missing. Please follow instructions at reprap.org/wiki/Teacup_Firmware#Simple_Installation
> In file included from analog.c:22:
> /config_wrapper.h:26:4: error: #error config.h missing. Please follow instructions at reprap.org/wiki/Teacup_Firmware#Simple_Installation
> In file included from analog.c:33:
> /config_wrapper.h:26:4: error: #error config.h missing. Please follow instructions at reprap.org/wiki/Teacup_Firmware#Simple_Installation

Specifying full paths to printer and board file in config.h solved the problem.

In the answer to the above post I wrote about a lost working directory. Is this really the case? Or is it a problem that config\printer.taalu.h is specified with a backlash instead of a forward slash?

Traumflug, 2015-06-04:
From [gcc documentation](https://gcc.gnu.org/onlinedocs/cpp/Search-Path.html):

> GCC looks for headers requested with #include "file" first in the directory containing the current file, then in the directories as specified by -iquote options, then in the same places it would have looked for a header requested with angle brackets. For example, if /usr/include/sys/stat.h contains #include "types.h", GCC looks for types.h first in /usr/include/sys, then in its usual search path.

jbernardis, 2015-06-04:
I don't use a literal backslash in my code.   I use os.path.sep which is the official python way of doing it in an os independent way. 

I'll look at the code,  but didn't you get rid of config wrapper.h?

Traumflug, 2015-06-04:
> didn't you get rid of config wrapper.h?

Nope, I just moved #include "arduino.h" into config_wrapper.h, away from config/board.*.h. config_wrapper.h is apparently found; config.h, too.

jbernardis, 2015-06-04:
The first message above isn't complaining about config.h missing.  It's complaining that it can't run F:\Program Files (x86)\Arduino.  Obviously this is an incomplete path because there is no compiler at the end of that string.  My initial comment would be that it's due, perhaps, to spaces in the file name, but I see spaces in the directory at the root level - so it can't be that.

Let me look at it from that point of view.

jbernardis, 2015-06-04:
In looking at build.py I see a class named ScriptTools which I don't immediately recognize.  But there is a method there named figureCommandPath.  In this method, if arduinodir is set, it looks for the compiler inside a series of subdirectories.  If it cannot find anything there, it just returns the value of arduinodir, which of course does not include a compiler,   I think this is what's happening, and if it can't find the compiler, then I wonder if the arduino environment is not set up properly.

I am at work right now and am unable to make changes, but maybe a few print statements would be helpful here

Traumflug, 2015-06-04:
> The first message above isn't complaining about config.h missing. It's complaining that it can't run F:\Program Files (x86)\Arduino.

Well spotted, didn't recognize this. This also explains why there are not gcc error messages :-)

So we have two points of failure.

> In looking at build.py I see a class named ScriptTools which I don't immediately recognize.

My work, it should allow to use Arduino as well as the native gcc on all platforms. Might be possible that Arduino uses a different directory structure inside the package on Windows. Didn't check this.

Traumflug, 2015-06-04:
I took the plunge and assembled a virtual machine with XP. Uhm. Now I know why some people are a bit reluctant to use Git or the command line in general.

Anyways, first thing I noticed is that our fine error message about missing wxPython doesn't appear when configtoo.py is double-clicked, because the window closes immediately. Added 10 seconds delay, see branch *issue158*

The good thing is, double-clicking works after a plain Python installation. Nice!

Traumflug, 2015-06-04:
> It's complaining that it can't run F:\Program Files (x86)\Arduino

The solution to this one was simple: Windows uses .exe suffixes :-) Added a possible solution to the *issue158* branch, which is now made of the three commits Github strays into our conversation.

Traumflug, 2015-06-04:
Next problem: it compiles, but at link time I receive this:

    collect2.exe: error: ld returned 5 exit status

Will try to hunt this next.

Traumflug, 2015-06-04:
Found an ugly hack for the ld.exe problem, adding "-Wl,-V" (for more verbose output) lets the linker do its work ... hmm.

----
The reason why compilation fails with Arduino IDE is pretty obvious, the command Arduino uses is: 

C:\Programme\Arduino\hardware\tools\avr/bin/avr-gcc [...] C:\DOKUME~1\MARKUS~1\LOKALE~1\Temp\build329253637943700400.tmp\analog.c -o C:\DOKUME~1\MARKUS~1\LOKALE~1\Temp\build329253637943700400.tmp\analog.c.o 

In other words: all sources and headers are copied elsewhere, but not recursively. Right now I have no answer to this one. It'll get worse when arch-avr/ and arch-arm/ get introduced to allow multiple architectures. Any idea besides dropping Arduino support?

jbernardis, 2015-06-04:
That's really ugly, but under the covers, there's still just avr-gcc.  We're not really using the IDE; Arduino might copy the files, but that doesn't mean we have to.  I should be able to invoke that compiler, tell it where my source file is and tell it where to put the object file.

I'd hate to drop arduino support because other firmwares work with it.  Let me see if I can make any progress on my windows machine.  I thought I was building it there earlier, but my memory might be off.

jbernardis, 2015-06-05:
It builds just fine for me on windows:

```
"c:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-gcc.exe" -c -DF_CPU=16000000UL -mmcu=atmega2560 -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns="C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\sermsg.al" -save-temps -o "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\sermsg.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\sermsg.c"

"c:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-gcc.exe" -c -DF_CPU=16000000UL -mmcu=atmega2560 -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns="C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\sersendf.al" -save-temps -o "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\sersendf.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\sersendf.c"

"c:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-gcc.exe" -c -DF_CPU=16000000UL -mmcu=atmega2560 -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns="C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\temp.al" -save-temps -o "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\temp.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\temp.c"

"c:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-gcc.exe" -c -DF_CPU=16000000UL -mmcu=atmega2560 -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns="C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\timer.al" -save-temps -o "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\timer.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\timer.c"

"c:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-gcc.exe" -c -DF_CPU=16000000UL -mmcu=atmega2560 -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns="C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\usb_serial.al" -save-temps -o "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\usb_serial.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\usb_serial.c"

"c:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-gcc.exe" -c -DF_CPU=16000000UL -mmcu=atmega2560 -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns="C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\watchdog.al" -save-temps -o "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\watchdog.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\watchdog.c"

Compile completed normally.

"c:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-gcc.exe" -Wl,--as-needed -Wl,--gc-sections -DF_CPU=16000000UL -mmcu=atmega2560 -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns=teacup.elf -save-temps -o "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\teacup.elf" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\analog.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\clock.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\crc.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\dda.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\dda_kinematics.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\dda_lookahead.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\dda_maths.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\dda_queue.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\debug.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\delay.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\gcode_parse.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\gcode_process.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\graycode.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\heater.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\home.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\intercom.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\mendel.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\pinio.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\sd.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\serial.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\sermsg.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\sersendf.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\temp.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\timer.o" "C:\Users\Jeff\Downloads\Teacup_Firmware-master\build\usb_serial.o" "C:\ [... cut ...]
         RAM:    1304 bytes   127.34%    63.67%    31.84%    15.92%
      EEPROM:      48 bytes     4.69%     2.34%     2.34%     1.17%

Build completed normally.
```

in my settings, I set my arduino dir to this:
```
c:\Program Files (x86)\Arduino
```

no double backslashes or anything.  I also had to remove the =obj from the compiler -save-temps option.

I also had to make the following change in build.py:
```
       cmdpathTry = os.path.join(cmdpathTry, baseCommand+".exe")
```
a temporary hack just to see if it would work.

Traumflug, 2015-06-05:
Is it possible you store your own printer and board files not in config/, but directly in Teacup_Firmware/ ? In this case it works as well, because these files are then copied, too.

Regarding the future: it looks much like ARM support will bring in a lot of additional files, about 4 of them for every MCU variant, plus some 10 files to access hardware the ARM way. To deal with this I outlined a file placing strategy here: https://github.com/Traumflug/Teacup_Firmware/issues/42#issuecomment-45040091

Traumflug, 2015-06-05:
> I also had to make the following change in build.py

This one should be fixed on the issue158 branch (about the same solution).

Traumflug, 2015-06-06:
Moved all the commits on the issue158 branch directly to master; either because they're harmless or because they're cruicial.