Traumflug/Teacup_Firmware #159: Add a Help menu, add "Report problem"

Traumflug, 2015-06-07:
A quick brain dump between 11 PM and 1 AM resulted in a Help menu, see the issue159 branch. With a nice about box, mentioning the top 10 contributors.

Perhaps more interesting is a "Report problem" item. The idea is to open a mailto: URL with board and printer configuration attached. And perhaps a few questions: "What did you do?", "What did you expect to happen?", "What happened instead?". This hopefully encourages people to not stay silent.

Where should these emails be sent to? Ideally here in this issue tracker, IMHO. How to achieve this? What do you generally think of this idea?

jbernardis, 2015-06-07:
I like it.   I'm not sure where the email will go,  but I think this is a good idea.

Also,  I don't mind if you use my real name like you do for everyone else: Jeff Bernardis

Traumflug, 2015-06-07:
Answering this by email to see how it works. The To: address is apparently customized:

reply@reply.github.com

Let's cross fingers these addresses stay valid for a while.

board.mydevelopment.h:
```c

/***************************************************************************\
*                                                                           *
* 1. CPU                                                                    *
*                                                                           *
\***************************************************************************/

/** \def CPU_TYPE
  CPU types a user should be able to choose from in configtool. All
  commented out.
*/
//#define CPU_TYPE atmega644
//#define CPU_TYPE atmega644p
//#define CPU_TYPE atmega1284
//#define CPU_TYPE atmega1284p

/** \def CPU
  CPU actually present on the board.
*/
#define CPU                      atmega1284p

/** \def F_CPU_OPT
  CPU clock frequencies a user should be able to choose from in configtool.
  All commented out.
*/
//#define F_CPU_OPT 16000000UL
//#define F_CPU_OPT 20000000UL

/** \def F_CPU
  Actual CPU clock rate. #ifndef required for Arduino compatibility.
*/
#ifndef F_CPU
#define F_CPU                    20000000UL
#endif

/** \def MOTHERBOARD
  This is the motherboard, as opposed to the extruder. See extruder/ directory
  for GEN3 extruder firmware.
*/
#define MOTHERBOARD

/***************************************************************************\
*                                                                           *
* 2. PINOUTS                                                                *
*                                                                           *
\***************************************************************************/

#define X_STEP_PIN               DIO19
#define X_DIR_PIN                DIO18
#define X_MIN_PIN                DIO7
//#define X_MAX_PIN                DIO6
//#define X_ENABLE_PIN             xxxx
//#define X_INVERT_DIR
//#define X_INVERT_MIN
//#define X_INVERT_MAX
//#define X_INVERT_ENABLE

#define Y_STEP_PIN               DIO23
#define Y_DIR_PIN                DIO22
#define Y_MIN_PIN                DIO5
//#define Y_MAX_PIN                DIO2
//#define Y_ENABLE_PIN             xxxx
[... 162 lines cut ...]
  115200, other common values are 19200, 38400 or 57600. Ignored when USB_SERIAL
  is defined.
*/
#define BAUD                     115200

/** \def XONXOFF
  Xon/Xoff flow control.

  Redundant when using RepRap Host for sending G-code, but mandatory when
  sending G-code files with a plain terminal emulator, like GtkTerm (Linux),
  CoolTerm (Mac) or HyperTerminal (Windows).
*/
//#define XONXOFF

/** \def USB_SERIAL
  Define this for using USB instead of the serial RS232 protocol. Works on
  USB-equipped ATmegas, like the ATmega32U4, only.
*/
//#define USB_SERIAL
```

Am 08.06.2015 um 01:28 schrieb Jeff Bernardis:
> I don't mind if you use my real name

I just took what Git reported :-) Will change that.

Traumflug, 2015-06-07:
OK, formatting for C works in preview, but not after hitting "update comment". Also, the full file is quite lengthy.

Traumflug, 2015-06-08:
Picked the menu its self and the about box to *experimental*. Also added yet a third item, a simple message hinting to the tooltips.

The Report Problem thing is still on the *issue159* branch, because I'm not too sure I did things right. This regexp ... eeek :-) It's this commit: https://github.com/Traumflug/Teacup_Firmware/commit/97516c16dedadbef13cf317f58c10e0bebfc1c4e

jbernardis, 2015-06-08:
I'm not sure what you're trying to do with that code.  It looks like you're trying to extract a portion of all the board and printer files, although I don't know why.

Your regular expression matches a string that begins with an asterisk, contains an arbitrary number (including 0) of anything that is not a carriage return, followed, optionally by an asterisk.  Is that what you intended?

Traumflug, 2015-06-08:
> It looks like you're trying to extract a portion of all the board and printer files

I hope it's only the actually used printer and board file. ... which reminds me that thins might be None, in which case this code fails, right?

> Is that what you intended?

It should match anything starting with / * * and ending with * /, including newlines in between. The \r is a dummy, because I can't find something matching everything. A dot doesn't match a newline.

jbernardis, 2015-06-08:
I've never tried to do an re that matched multilines, although I know it's possible.

I found this site:
http://stackoverflow.com/questions/587345/python-regular-expression-matching-a-multiline-block-of-text

I don't know if it will help

Traumflug, 2015-06-09:
Please answer these questions before hitting "send":

What did you try to do?

What did you expect to happen?

What happened instead?

To allow developers to help, configuration files are attached, with  
help comments stripped:

(no file loaded)

printer.wolfstrap.h:
----------------------------------------------

#define KINEMATICS               KINEMATICS_STRAIGHT

#define STEPS_PER_M_X            1280000
#define STEPS_PER_M_Y            1280000
#define STEPS_PER_M_Z            1280000
#define STEPS_PER_M_E            96271

#define MAXIMUM_FEEDRATE_X       600
#define MAXIMUM_FEEDRATE_Y       600
#define MAXIMUM_FEEDRATE_Z       600
#define MAXIMUM_FEEDRATE_E       2000

#define SEARCH_FEEDRATE_X        50
#define SEARCH_FEEDRATE_Y        50
#define SEARCH_FEEDRATE_Z        50

#define ENDSTOP_CLEARANCE_X      1000
#define ENDSTOP_CLEARANCE_Y      1000
#define ENDSTOP_CLEARANCE_Z      100

//#define X_MIN                    0.0
//#define X_MAX                    200.0

//#define Y_MIN                    0.0
//#define Y_MAX                    200.0

//#define Z_MIN                    0.0
//#define Z_MAX                    140.0

#define E_ABSOLUTE

//#define ACCELERATION_REPRAP
#define ACCELERATION_RAMPING
//#define ACCELERATION_TEMPORAL

#define ACCELERATION             100

#define LOOKAHEAD

#define MAX_JERK_X               20
#define MAX_JERK_Y               20
#define MAX_JERK_Z               0
#define MAX_JERK_E               200

//#define USE_INTERNAL_PULLUPS

#define TEMP_HYSTERESIS          10

#define TEMP_RESIDENCY_TIME      60

#define TEMP_EWMA                1.0

#define REPORT_TARGET_TEMPS

//#define HEATER_SANITY_CHECK

#define EECONFIG

//#define BANG_BANG

//#define BANG_BANG_ON             200

//#define BANG_BANG_OFF            45

#define MOVEBUFFER_SIZE          8

//#define DC_EXTRUDER              HEATER_motor
//#define DC_EXTRUDER_PWM          180

//#define USE_WATCHDOG

#define TH_COUNT                 8

//#define FAST_PWM

#define PID_SCALE                1024L

#define ENDSTOP_STEPS            4

/*
#define CANNED_CYCLE "G1 X100 F3000\n" \
"G4 P500\n" \
"G1 X0\n" \
"G4 P500\n"
*/
----------------------------------------------

- - - - - - - - - - - - - - - - - - -
Dipl. Ing. (FH) Markus Hitter
http://www.jump-ing.de/

Traumflug, 2015-06-09:
Heh :-) The previous comment was sent with the "Report problem" thing, from Mac OS X :-)

Tried a lot with this regex, eventually even read the manual. Something like [.\r\n] would simply not match newlines, no matter what. But there are additional modifiers, in particular re.DOTALL. The regex ended up like this:
```python
reHelpText = re.compile("/\*\*.*?\*/\r?\n", re.DOTALL)
```
Thanks for helping.

Branch *issue159* is updated. If you don't mind I'd consider this thing solved.

jbernardis, 2015-06-09:
This is nice.  I didn't realize what you were after, but this gives us their configuration minus comments.  Great idea and well done.

Traumflug, 2015-07-01:
I just see this wasn't picked to experimental, yet. Catched up on this a minute ago and removed the issue159 branch. Thanks for the review.

closed 2015-07-01 by Traumflug

Traumflug, 2015-07-24:
Please answer these questions before hitting "send":

What did you try to do? 

Load a previously saved config file.

What did you expect to happen?

File to be loaded and all values to be populated with saved values

What happened instead?

Some values are not populated by loading the config file, Min/Max pins for X,Y, and Z, and the PS_ON pin remain listed as “-“. This issue is only present on my OS X 10.10.4 machine. Windows 7 and linux both operate as expected. Files being sent at your request.

To allow developers to help, configuration files are attached, with help comments stripped:

board.frank.h:
----------------------------------------------

//#define CPU_TYPE atmega1280
//#define CPU_TYPE atmega2560

#define CPU                      atmega2560

//#define F_CPU_OPT 16000000UL

#ifndef F_CPU
#define F_CPU                    16000000UL
#endif

#define MOTHERBOARD

#define X_STEP_PIN               DIO54
#define X_DIR_PIN                DIO55
#define X_MIN_PIN                DIO3
#define X_MAX_PIN                DIO2
#define X_ENABLE_PIN             DIO38
//#define X_INVERT_DIR
//#define X_INVERT_MIN
//#define X_INVERT_MAX
#define X_INVERT_ENABLE

#define Y_STEP_PIN               DIO60
#define Y_DIR_PIN                DIO61
#define Y_MIN_PIN                DIO14
#define Y_MAX_PIN                DIO13
#define Y_ENABLE_PIN             DIO56
//#define Y_INVERT_DIR
//#define Y_INVERT_MIN
//#define Y_INVERT_MAX
#define Y_INVERT_ENABLE

#define Z_STEP_PIN               DIO46
#define Z_DIR_PIN                DIO48
#define Z_MIN_PIN                DIO18
#define Z_MAX_PIN                DIO17
#define Z_ENABLE_PIN             DIO62
//#define Z_INVERT_DIR
//#define Z_INVERT_MIN
//#define Z_INVERT_MAX
#define Z_INVERT_ENABLE

#define E_STEP_PIN               DIO26
#define E_DIR_PIN                DIO28
#define E_ENABLE_PIN             DIO24
#define E_INVERT_DIR
#define E_INVERT_ENABLE

#define PS_ON_PIN                DIO12
//#define PS_MOSFET_PIN            xxxx
//#define STEPPER_ENABLE_PIN       xxxx
//#define STEPPER_INVERT_ENABLE

//#define DEBUG_LED_PIN            DIO13

#ifndef DEFINE_TEMP_SENSOR
  #define DEFINE_TEMP_SENSOR(...)
#endif

//#define TEMP_MAX6675
#define TEMP_THERMISTOR
//#define TEMP_AD595
//#define TEMP_PT100
//#define TEMP_INTERCOM

//#define TEMP_SENSOR_PIN AIO13
//#define TEMP_SENSOR_PIN AIO14
//#define TEMP_SENSOR_PIN AIO15

//DEFINE_TEMP_SENSORS_START
//                 name      type           pin    additional
DEFINE_TEMP_SENSOR(extruder, TT_THERMISTOR, AIO13, THERMISTOR_EXTRUDER)
DEFINE_TEMP_SENSOR(bed,      TT_THERMISTOR, AIO14, THERMISTOR_BED)

// Beta algorithm      r0      beta  r2    vadc
// Steinhart-Hart      rp      t0    r0      t1    r1      t2    r2
//TEMP_TABLE EXTRUDER (100000, 4092, 4700, 5.0)
//TEMP_TABLE BED      (100000, 4092, 4700, 5.0)
//DEFINE_TEMP_SENSORS_END

#ifndef DEFINE_HEATER
  #define DEFINE_HEATER(...)
#endif

//#define HEATER_PIN DIO4
//#define HEATER_PIN DIO5
//#define HEATER_PIN DIO8
//#define HEATER_PIN DIO9
//#define HEATER_PIN DIO10

//DEFINE_HEATERS_START
//            name      port   pwm
DEFINE_HEATER(extruder, DIO10, 1)
DEFINE_HEATER(bed,      DIO8,  1)
DEFINE_HEATER(fan,      DIO9,  1)

#define HEATER_EXTRUDER HEATER_extruder
#define HEATER_BED HEATER_bed
#define HEATER_FAN HEATER_fan
//DEFINE_HEATERS_END

#define BAUD                     115200

//#define XONXOFF

//#define USB_SERIAL

//#define SD_CARD_SELECT_PIN
----------------------------------------------

printer.frank.h:
----------------------------------------------

#define KINEMATICS               KINEMATICS_STRAIGHT

#define STEPS_PER_M_X            160000
#define STEPS_PER_M_Y            160000
#define STEPS_PER_M_Z            2560000
#define STEPS_PER_M_E            625000

#define MAXIMUM_FEEDRATE_X       6000
#define MAXIMUM_FEEDRATE_Y       6000
#define MAXIMUM_FEEDRATE_Z       200
#define MAXIMUM_FEEDRATE_E       6000

#define SEARCH_FEEDRATE_X        200
#define SEARCH_FEEDRATE_Y        200
#define SEARCH_FEEDRATE_Z        50

#define ENDSTOP_CLEARANCE_X      1000
#define ENDSTOP_CLEARANCE_Y      1000
#define ENDSTOP_CLEARANCE_Z      100

//#define X_MIN                    0.0
//#define X_MAX                    200.0

//#define Y_MIN                    0.0
//#define Y_MAX                    200.0

//#define Z_MIN                    0.0
//#define Z_MAX                    140.0

#define E_ABSOLUTE

//#define ACCELERATION_REPRAP
#define ACCELERATION_RAMPING
//#define ACCELERATION_TEMPORAL

#define ACCELERATION             1000

#define LOOKAHEAD

#define MAX_JERK_X               200
#define MAX_JERK_Y               200
#define MAX_JERK_Z               0
#define MAX_JERK_E               200

#define USE_INTERNAL_PULLUPS

#define TEMP_HYSTERESIS          10

#define TEMP_RESIDENCY_TIME      60

#define TEMP_EWMA                1.0

#define REPORT_TARGET_TEMPS

//#define HEATER_SANITY_CHECK

#define EECONFIG

#define BANG_BANG

#define BANG_BANG_ON             255

#define BANG_BANG_OFF            0

#define MOVEBUFFER_SIZE          8

//#define DC_EXTRUDER              HEATER_motor
//#define DC_EXTRUDER_PWM          180

//#define USE_WATCHDOG

#define TH_COUNT                 8

//#define FAST_PWM

#define PID_SCALE                1024L

#define ENDSTOP_STEPS            4

/*
#define CANNED_CYCLE "G1 X100 F3000\n" \
"G4 P500\n" \
"G1 X0\n" \
"G4 P500\n"
*/
----------------------------------------------

Traumflug, 2017-03-20:
Please answer these questions before hitting "send":

What did you try to do?

I tried to build my config setting for arduino uno 3d printer.
What did you expect to happen?
normal build completion
What happened instead?
”build terminated abnormally”
with error “avr-gcc.exe error : no such file or directory

To allow developers to help, configuration files are attached, with help comments stripped:

board.UNO.h:
----------------------------------------------

//#define CPU_TYPE atmega168
//#define CPU_TYPE atmega168p
//#define CPU_TYPE atmega328
//#define CPU_TYPE atmega328p

#define CPU                      atmega328p

//#define F_CPU_OPT 16000000UL

#ifndef F_CPU
#define F_CPU                    16000000UL
#endif

#define MOTHERBOARD

#define TX_ENABLE_PIN            DIO12
#define RX_ENABLE_PIN            DIO13

#define X_STEP_PIN               DIO2
#define X_DIR_PIN                DIO5
#define X_MIN_PIN                DIO9
//#define X_MAX_PIN                DIO21
#define X_ENABLE_PIN             DIO8
//#define X_INVERT_DIR
//#define X_INVERT_MIN
//#define X_INVERT_MAX
#define X_INVERT_ENABLE

#define Y_STEP_PIN               DIO3
#define Y_DIR_PIN                DIO6
#define Y_MIN_PIN                DIO10
//#define Y_MAX_PIN                DIO26
#define Y_ENABLE_PIN             DIO8
//#define Y_INVERT_DIR
//%