Traumflug/Teacup_Firmware #162: Extruder advances during homing movement (G28)

jbernardis, 2015-06-11:
I notice that when the printer is homing via G28, as it is moving on the X, Y, and then Z axes, the extruder gear is turning.  Not very fast, but turning none the less.

neildarlow, 2015-09-12:
I have configured Teacup for the Mendel90 and observed this also.

The extruder movement was coincident with the X homing movement. Towards the endstop the extruder moved forwards grinding the hobbed-bolt against the filament. On the X endstop backoff the extruder reversed direction and filament was expelled from the hotend.

The movement was not insignificant and the G-Code macro executed in Pronterface that brought this about was:
G28
G1X0Y0F9000

Regards,
Neil Darlow

Traumflug, 2015-09-12:
Hmm. Two people, now I have to investigate this.

Can you send me your board and printer config as email attachment to mah@jump-ing.de? This would allow me to reproduce the issue as close as possible.

triffid, 2015-09-14:
Perhaps the extruder position is non-zero when homing is performed, and the
homing procedure feeds a zero position to DDA which then generates
appropriate steps?

That would explain why this issue is rare, most people only home after a
fresh start when E is zero, or after a print finishes, again with E likely
to be zero

neildarlow, 2015-09-14:
My observation was made as the first operation of the following sequence:
1) Power on printer
2) Open Pronterface and connect to printer
3) Execute homing macro as described in my first post
I would imagine that most Teacup state and variables would have been at power on conditions at this time.

Regards,
Neil Darlow

Traumflug, 2015-09-14:
I found a first issue: STEPS_PER_M_E comes with a lengthy formula, which, according to my pocket calculator, results in 53.8 = 53. Just 53 steps per *meter* is unlikely. More likely is 53844, which would be 53.844 steps per *mm*.

Traumflug, 2015-09-14:
Just writing what I see ... you set ENDSTOP_CLEARANCE_{XYZ} to zero, which causes slow and/or inprecise homing. This isn't Marlin, which can't do better.

Googling a bit it looks like the Mendel90 uses microswitches without lever, so I think setting this clearance to just 500 (0.5mm) should be OK. For comparison: when touch-probing PCBs I use about 30 um. The endstop clearance feature still works in such cases.

Traumflug, 2015-09-14:
P.S.: I just see in the email you wrote me that you actually tried with endstop clearance. So I accused you wrongly, sorry.

Your config files are now on a new branch *issue-162*, so everybody can help finding the cause. While @triffid's idea is a thing to investigate I see in the sources that homing movements don't start off 0, 0, 0, 0, but from the current position.

neildarlow, 2015-09-14:
Erm... Try that equation again.

For me and ignoring the last term, which is a 100mm extrusion calibration factor, yields:

124800000.0 / 233.2935 = 534948.466202445 steps/metre or 534.948 steps/mm.

Regards,
Neil Darlow

phord, 2015-09-14:
I don't see any errant steps in the simulator, but trying to find them forced me to fix up some pieces of the simulator which are lacking.  

I did notice this warning when building with your config:

    home.c:219:4: warning: #warning Z_MAX_PIN defined, but not Z_MAX. home_z_positive() disabled. [-Wcpp]
       #warning Z_MAX_PIN defined, but not Z_MAX. home_z_positive() disabled.

In the simulator this config causes the Z to home anyway, but that's probably because it defines settings your config does not.  Seems way convenient in any case. 

I'm curious: do you simply operate without any Z-homing?

neildarlow, 2015-09-15:
No, I use Z homing and my compile doesn't omit that warning.

Do you have the correct files? I see Traumflug committed two versions of my configuration files. It appears that he thought my settings were deviations from a Mendel90 standard (which they are not) and one set of files has a Z minimum endstop defined (which the Mendel90 does not).

You need: board.melzi-neil.h and printer.mendel90-neil.h for the settings I use in reality.

Traumflug, 2015-09-15:
> one set of files has a Z minimum endstop defined (the endstop is pysically at the top of Z on a Mendel90)

The location of endstops is a board setting. The Melzi can be used with other printers than the Mendel90, so I reset this to the usual "all min endstops". The config files with *-neil* in the name are unchanged. Well, now changed to adapt to the recently introduced new heater macro.

phord, 2015-09-16:
Sorry.  I didn't see both files in there; I had a bad combination.  It compiles for me without warning now.