Traumflug/Teacup_Firmware #163: Ultra LCD support

Traumflug, 2015-06-19:
No idea which kind of display this is, but [here](http://forums.reprap.org/read.php?147,33082,523623#msg-523623) is a nice guy who wants to implement it.

jgrjgr, 2015-06-20:
i would be happy to do any testing as needed,  i have a number of rumba boards here at my disposal plus other random lcd stuff too

jgrjgr, 2015-07-11:
i have created a new branch with some semi working lcd init code for the reprap discount smart lcd, current it doesn't compile because i've shifted the code into it's own file (previously i had it in gcodeprocess.c)  i've also added all the required pin defines to the rumba board file in /config/

Traumflug, 2015-07-11:
For reference, that's the *lcd2* branch.

Also rebased the original *lcd* branch to master.

Third, just extracted all the display stuff from the delta branch into a branch *lcd3*. @jgrjgr, you said you had this working already. It's on top of *experimental*, so you get SD card stuff, too.

Now we have two LCD rather hackish libraries in *lcd2* and *lcd3* and a more proper one (support for various sizes) in *lcd*. I think it would be a good idea to get one of them working and doing changes with many different displays in mind. That is, put configuration #defines into a separate config file, display.<name>.h.

Stuff on the *lcd3* branch also has lots of useful stuff for actually writing meaningful information to the display, it just duplicates existing code for writing to serial a bit too much. Deduplication can be done later, though.

Now feel free to hack on top of one of these branches or to refine (git rebase -i) them. Or to create yet another one, pulling the goodies of the other three together.

Traumflug, 2015-07-16:
I thin @jgrjgr makes good progress on this one. The *lcd3* branch now compiles properly, display works. Today two new commits came in, it looks like rotary encoder now shows signs of life. Excellent!

jgrjgr, 2015-08-21:
update ***   the lcd branch i have here locally is now working pretty well displaying filenames from the sdcard , there's still a lot to do to make it usable by the end user:

todo:

- fix encoder code so it recognises both rising and falling clicks
- add in code that initialiases the sdcard on powerup and upon card insert
- add code for various basic menu based functions eg: heaters on and off , Zeroing  XYZE , spindle on an off, gcode file selection/execution

the bad news is i'm not going to be able to actually commit it until monday night when i have a stable internet connection, currently i'm in a beach paradise  on the way to sydney and i'm using my mobile phones dodgy internet connection to comment here, it annoyingly not stable enough for github though. by the time i get to sydney though i will likely have most of the above todo list finished 

tmorrow i'll stick all the files on dropbox and paste a link to it here in another comment so anyone who wants to have a play with it is able to untill i can commit them to a branch properly

Traumflug, 2015-08-21:
@youssefaly97 wrote in #183:

Hello. So I opened this branch some time ago, I have completed over the LCD3 branch's code to get to:

What's done:

-made a customizable splash screen at start up

-made main menu screen with sub menus (up to 3 levels)

-used the beeper to indicate successful selections and when printer is ready after startup (can be turned off in config.wrapper)

-added debounce code for encoder press down click

-added cursor for selection

What's yet to be done:

-vertical LCD shifting code still needs some work

-might add debouncing code for the encoder wheel itself (it works with about 90% reliability)

-add code for the "back" entry (go up a level in the menu)

-store strings in PROGMEM to go with the Teacup objectives (optimal ram usage)

-connect the actual functions to the menu structures

Please give it a try and tell me about your findings if I missed any while I was testing.

Traumflug, 2015-08-22:
@jgrjgr wrote:

Hi all,

i have created another branch which contains my semi working version containing the start of a basic lcd menu and file listing on the lcd, currently it's being developed on the rumba board with a reprepdiscount smart lcd controller.

i used bits and pieces from the other branches so far it is functional to the point:

- the rotary encoder works (2 clicks for every movement)
- the button click works
- there is a series of placeholder menu options( i am putting code behind these to make the properly functional atm)
- files will display on the bottom line of the 4 line lcd one at a time, this is perfectly fine and usable however it's probably desirable to utlise all 3 - 4 lines at some point
- lcd inits and sdcard inits have been swapped around so sdcard mounting information can me displayed on the lcd
- it will attempt once (will increase this) upon powerup to init the sdcard (it only works sporadically at the moment)(i'm going to tie this in with the sdcard detect pin and have it put the letters "SD" on the top right part of the screen when it detects and successfully mounts the sdcard)

in summary we are not far off being able to totally run teacup in a standalone way.

Traumflug, 2015-08-22:
Sounds very good, @youssefaly97 and @jgrjgr !

Once it works I'll chime in :-) For example, much of the display code duplicates code for the serial line. To me it looks like we need an output selector, similar to the recently introduced input selector (gcode_active and gcode_sources, defined in gcode_parse.h, mostly used in mendel.c). Think of an SPI based display, which whould require yet another duplication of all this code.

Also getting all this display stuff into Configtool in a dummy-proof fashion.

But there's no hurry, once you two have what you see need for these LCD branches can be rebased to master/experimental, a few "tidy" bits implemented there, rebased again, and so on. Until the LCD branches become empty and everything is in mainstream.

I'm currently a bit off of Teacup, wrestling with a STEP exporter for gEDA/pcb.

> in summary we are not far off being able to totally run teacup in a standalone way.

Excellent!

RaD, 2015-09-02:
Just finished my work on OLED 128x32 I2C display (similar to Adafruit one).

[image]

At what point of the firmware source tree can I start to implement its support?

Traumflug, 2015-09-02:
Looks nice!

You can start any point you want. Usually this would be *experimental*. It's display code, so taking a look into the various existing display branches is probably a good idea. At some point all display code needs to be joined, maintaining three flavours of code doing the same makes no sense.

As I'm aware you're a talented coder, can I ask you to separate the I2C stuff from display functions, similar to how it's done for SPI? Sooner or later, I guess, you'll want to connect these displays to various controllers, AVR and ARM ones.

RaD, 2015-09-02:
I think from point of Arduino Nano. It has low RAM amount. So the most problem would be to build a graphical representation of display view. The OLED displays like this works the same manner. There would be a difference in send_i2c or send_spi function. Anyway, I will implement first version of support code, then we can make it better.

Traumflug, 2015-09-02:
Good plan!

RaD, 2015-09-22:
I am on my way of implementing I2C support into TeaCup.

RaD, 2015-11-03:
My branch with experimental SSD1306 I2C OLED display support.
https://github.com/Traumflug/Teacup_Firmware/tree/ssd1306

I need to solve the low RAM/queue problem. At this time it shows simple text.
[image]