Traumflug/Teacup_Firmware #165: Issues with configtool on OSX

ntoff, 2015-07-04:
OSX Yosemite

1st issue is with the latest Arduino IDE 1.6.5 configtool can't execute any of the build commands.

Line https://github.com/Traumflug/Teacup_Firmware/blob/master/configtool/build.py#L39
reads: cmdpath = os.path.join(cmdpath, "Contents/Resources/Java")
changed to: cmdpath = os.path.join(cmdpath, "Contents/Java")

and it works

2nd issue is that loading the board config doesn't populate the values in the config tool, yet they exist in the config file itself and will build just fine. This behaviour doesn't occur on my windows machine, loading the board config populates all the values in the configtool under windows 7.

Traumflug, 2015-07-04:
Thanks for reporting, @ntoff.

For the second part, not sure what to say. It works on my old PPC iBook running OS X 10.4. Could you please try to populate some fields manually after loading a board file? It'd be interesting wether the entered value or the original one ends up in the saved board file. This might help to narrow down on what's going wrong.

Traumflug, 2015-07-10:
> Line https://github.com/Traumflug/Teacup_Firmware/blob/master/configtool/build.py#L39
> reads: cmdpath = os.path.join(cmdpath, "Contents/Resources/Java")
> changed to: cmdpath = os.path.join(cmdpath, "Contents/Java")

I've just pushed a new branch *issue165* to address this part of the issue and made sure it still works on old Mac OS X. See the commit above this comment.

Looks good?

ntoff, 2015-07-10:
I haven't had any time to play with it lately but I can't use teacup on my printer(s) anyway. I can check out whether configtool works but, while teacup will move the printer, I use 2 stepper drivers for 2 Z motors so I can't actually print with it. Nor does it seem to support M80/M81 (that I can see?) so I can't actually turn the printer on! I had to hot wire it just to test.

This isn't a feature request, just letting you know I can't provide any feedback other than whether or not the configtool works.

Traumflug, 2015-07-10:
> I use 2 stepper drivers for 2 Z motors so I can't actually print with it.

Why not? Simply connect them to the same signal pins, like you'd do it with any other firmware. Doing this duplication in the firmware would be trivial, but also make configuration more complicated for no gain.

> I can't actually turn the printer on!

Did you try to move the printer without sending M80 before? Did you try to turn on a heater? Teacup will take care of turning the PSU on and off as needed. M80/81 is simply a no-op in Teacup.

I mean, many others are printing with Teacup, so one can pretty sure it works at least basically.

ntoff, 2015-07-12:
I use an xbox 360 PSU which needs the power pin to go HIGH to turn it on, where an ATX needs it pulled LOW.

>Why not? Simply connect them to the same signal pins, like you'd do it with any other firmware. 

Marlin has the option to use the 5th driver on the RAMPS board which is how my printer is set up. I don't have the technical know how to figure out how to make teacup send the step signal to both drivers so only one side of the gantry moves up & down since only one stepper is activated.

I can't just plug both motors into one driver because then there's not enough torque and I've had trouble with skipped steps.

triffid, 2015-07-12:
Have you tried hooking the motors in series?

Traumflug, 2015-07-12:
> I use an xbox 360 PSU which needs the power pin to go HIGH to turn it on

I've put a commit onto the *issue165* branch to deal with this. Put a line like this into your board configuration:

    #define PS_INVERT_ON

Note for the bug resolution: not in Configtool, yet, not distributed to all the board files, yet.

ntoff, 2015-07-12:
>Have you tried hooking the motors in series?

Didn't even know that was possible, I'd just plugged them both into the little ports on the RAMPS. I made up an adapter and tested it, seemed to work ok.

The only other issue is going to be the home offsets. My physical home positions are negative, the nozzle is 2mm below the bed and -10mm x and y. I do *not* fancy ever doing manual z0 leveling ever again. I've been there and after many frustrating hours, never again.

Traumflug, 2015-07-12:
> My physical home positions are negative, the nozzle is 2mm below the bed and -10mm x and y.

No problem, after homing send a G92 with the position where the nozzle actually is. There is some discussion ongoing in https://github.com/Traumflug/Teacup_Firmware/pull/112 on making this more convenient, but so far sending a G92 is the way to go and works fine.

Traumflug, 2015-07-13:
I don't think we'll see a confirmation about the commits here anytime soon, so I picked the fix for OS X onto experimental. The other commit for the inverted PS_ON pin may be delayed until its also available in Configtool.

ntoff, 2015-07-14:
You said the fix wasn't in yet
>Note for the bug resolution: not in Configtool, yet, not distributed to all the board files, yet.

is it in somewhere? I don't know how github works. Which branch should I download to test?

Traumflug, 2015-07-14:
> is it in somewhere?

Nope. The code didn't magically write its self. You have to use the *issue165* branch and add this #define manually.

ntoff, 2015-07-17:
Well if all I'm testing is the new search path for the arduino app then yes, that part works. However it's still very much broken in loading (and saving) of board config files, at least for me it is. I can save (overwrite) but not save-as because it tries to tell me board.boardname.h isn't valid and that I must save it as board.*.h.

I don't know if it's something that's changed in yosemite because I've not tried teacup on anything prior.

Traumflug, 2015-07-17:
> Well if all I'm testing is the new search path for the arduino app then yes, that part works.

Excellent, thank you for reporting.

> I can save (overwrite) but not save-as because it tries to tell me board.boardname.h isn't valid and that I must save it as board.*.h.

That's odd. After loading one of the configurations coming with the distribution, a plain save should actually be disabled. Overwriting distributed files might be handy short term, but is a pain when one wants to return to "factory" settings.

> it tries to tell me board.boardname.h isn't valid and that I must save it as board.*.h.

As an experiment, I've put another two commits onto the *issue165* branch, which applies the relaxation on wildcards of the file open dialog to the save as dialog. On how to update the branch, see http://reprap.org/wiki/Teacup_Firmware#Updating_the_repository

Tests for allowable file names are here, in case you want to play with it:

https://github.com/Traumflug/Teacup_Firmware/blob/issue165/configtool/boardpanel.py#L435 and https://github.com/Traumflug/Teacup_Firmware/blob/issue165/configtool/boardpanel.py#L462 and https://github.com/Traumflug/Teacup_Firmware/blob/issue165/configtool/boardpanel.py#L469

Pendants are in printerpanel.py

ntoff, 2015-07-17:
If you took a look at my fork in the configtool_osx branch I've already relaxed the restrictions to just *.h and that works for both the board.* and printer.* files.

I decided to update python to see if a different version would help with the loading of values which turned into a fiasco and broke my ability to run pronterface to control the printer and I ended up falling asleep while waiting for xcode to download (don't ask).

ntoff, 2015-07-18:
I can't figure out why it's not loading the values. I tried making it print the contents of cfgValues and it looks like it's reading it, the min and max pins are in there but for some reason it's just not setting them in the GUI. It seems to be *just* the min and max pins affected.

I wonder if it's just a problem with my machine, since no one else seems to be having any issues. I guess we'll call it quits until someone else comes along with the same issues, especially since it works for you.

Traumflug, 2015-07-18:
Is it possible they're simply commented out? Commented out means disabled, so the pin is ignored and the code handling it removed. Sending a M119 to the controller will tell you which ones are defined and what's their status.

To re-enable them you have to select the pin in question. So far there's no visible hint in the GUI to see which one it is, but since a number of recent commits on the experimental branch it's at least remembered internally which one this is, so it's "only" a matter of adding some visual notice on what the disabled pin should be when enabled. Perhaps by adding a " (default)" behind the pin name in the choice menu.

ntoff, 2015-07-19:
Nope, not commented out. Editing the config files by hand and using config tool to build / upload works. It just doesn't populate the min & max pins when loading the config. Sometimes it'll populate one of the x end stop options, sometimes it won't.

Might as well close the issue.

Traumflug, 2015-07-19:
Can you send me you config files? I'd like to investigate the issue. There's a "Report problem.." in the help menu which collects the necessary files and put them into an email.

ntoff, 2015-07-24:
Probably not a good idea, I've been messing about with all the files trying to print out various variables to see where it's failing. I could wipe it and send you the fresh files but I honestly don't know what good it would do since the same config file works on other OS's.

I did have an issue once with a manually edited file where OSX did this weird thing with the " marks when I edited it with the GUI text editor (no issues when using nano), but the config file was saved with configtool.

I've tried it on my pcduino which is running linux and python 2.7.6 where it works perfectly (well, it loads anyway I haven't bothered trying to build because we already know that part works) so unless someone else comes along with the same trouble I'd say close it. 

The config file is the same one across osx and windows where the windows machine loads it fine while OSX doesn't so there's obviously nothing wrong with the config file and since I now have it working under linux (listed as being linaro 14.04)

I'm going to assume it's something broken on my machine.

Traumflug, 2016-06-11:
Reviewed the commits made during this discussion, added PS_INVERT_ON handling to Configtool and moved all this to *experimental*. Thanks for the help, @ntoff!

One issue branch less. If there's anything left, please open a new issue.

closed 2016-06-11 by Traumflug