Traumflug/Teacup_Firmware #166: Add SD_CARD_SELECT_PIN support into configtool

RaD, 2015-07-09:
Self described

Traumflug, 2015-07-09:
D'oh. You also opened an issue for it. Sorry, missed this. Will take the stuff here.

Traumflug, 2015-07-09:
Sorry for not adding you earlier as collaborator, @RaD. Just catched up on this, you have write access now. The rules here are simple: create as many branches as you feel need for, but change other branches only in agreement with those who created these branches. Usually I review contributions by cherry-picking from such topic branches over to experimental.

Your patch works indeed fine, I just shortened the label a bit to make it fit into the space available. Then put it to *experimental*.

Traumflug, 2015-07-09:
Now to the missing bits :-)

- All the distributed board config files need such a line, else it won't work as expected.

        #define SD_CARD_SELECT_PIN <pin>

- All these #defines need a describing comment. These comments may help users compiling from the command line and are also used to create a helping tooltip on the GUI. Yes, we're aware this is duplicate work.

- See issue #169.

Traumflug, 2015-07-09:
> Usually I review contributions by cherry-picking from such topic branches over to experimental.

This also means, there are no merges. May be unusual, but works very well: #81

Traumflug, 2015-07-30:
I've just added a commit to *experimental* which distributes SD_CARD_SELECT_PIN to all the board files, along with a describing comment. This should complete this issue. Anything forgotten?

RaD, 2015-07-30:
OK. I am in Crimea at this time. Will check the change when come back to home and report about.

RaD, 2015-09-02:
Is this issue actual yet?

Traumflug, 2015-09-02:
If it works for you we can close it.

commit 247659e6 (2015-07-09): Merge pull request #167 from halfakop/experimental

Fix #166: Add SD card select pin support.
commit 72f4d4a7 (2015-07-09): Fix #166: Add SD card select pin support.
commit 46a76ab0 (2015-11-03): Fix #166: Add SD card select pin support.

Conflicts:

	configtool/pinoutspage.py
closed 2015-09-04 by RaD