Traumflug/Teacup_Firmware #169: Configtool: distribute new entries to older user-saved files

Traumflug, 2015-07-09:
A new #define just appeared in the board config file, see issue #166. The problem now is, users which saved their configuration before will see this pin and can choose it, but it won't be written out at save time. Very misleading.

Ths underlying problem is, when saving, Configtool takes the previously saved file and just changes the values already existing in there. It does *not* write a file from scratch.

One possible solution is to change this behaviour to write this file solely from information existing in memory. With all the formatting, so the result becomes the same as what we have now.

This would give another advantage: when loading a printer or a board, Configtool can load a generic file first, then add the configuration from the file actually requested. This way, new #defines and other changes automatically get into previously saved, not distributed files. Overriding one configuration with another one at load time works already.

If you have other ideas, they're welcome.

jbernardis, 2015-07-10:
I like this idea, and it's somewhat along the lines I was talking about earlier where I wanted to "normalize" the files by separating the content from the formatting.

So what I would do is 
1) read the standard file thereby getting all of the variables and their default values
2) read the user's file overriding the default values with their values
3) write the user's file using the formatting from the standard file and the in memory values

I'm actually away from home until Sunday.  When I get back, I will assess the effort.

jbernardis, 2015-07-12:
I just pushed up branch configtoolupdate to address this.  This update adds two files: board.metadata and printer.metadata that I read BEFORE I read the configuration files themselves.

This metadata file conatins 1) help text for the define variables, 2) Placeholders for where the heaters and sensors will be inserted, 3) and candidate values it has for pins, etc that need to be set at a global level, and 4) the define variable names - no values.

The only other bit of "data" in the metadata files is the ordering of the lines.  The user files will be written with this same ordering.

The normal board and printer files are exactly as they are today, although there are several considerations:
    1) we can remove defines that we do not assign values to because they are also in the metafile.
    2) we can, if we want to, remove the help text from here because it is no longer parsed.
    3) if there are candidate values that are board/printer specific, they belong in the corresponding file and not in the metafile

If a new define is being added, all that need be done is to add the new name to the metadata file.  Of course it also helps to have help text, but it's optional.  The user then needs to enter the configtool, load their file, possibly define a new value for the new define, and then write their file.

Likewise, if a define is being removed, just remove it from the metafile.  The user needs to cycle through their files and the define will be gone.

One other feature I was looking into adding - the help text and candidate pins do not need to be written to user files.  It should be pretty simply to not do so.

Traumflug, 2015-07-13:
> This update adds two files: board.metadata and printer.metadata

I wish it would, but I see only changed to board-/printerpanel.py. Perhaps you simply forgot to git-add them?

jbernardis, 2015-07-13:
you're exactly right.  I pushed up the two new files this AM.

Traumflug, 2015-07-13:
Honestly, when I read "metadata" I thought ... But then I had a look and found that this is basically a copy of one of the board files. Excellent!

My famous nitpickings :-)

- Values of disabled items get lost when just loading and saving a file. For example ramps-v1.3, one of the diff lines is ( - = original, + = saved copy)

        -//#define Z_MAX_PIN                DIO19
        +//#define Z_MAX_PIN

    Is this intentional? A user might want to enable it with a text editor later or Configtool might learn one day to prefer the default pin when enabling.
- If the metadata file has no values, how can one give default values for new items?
- It might be nicer to either name the metadata file board.generic.h, making it a user loadable file, or to stuff it into configtool/, next to all the .py files, to hide it from the user.
- The two choice lists on the board -> CPU page shows values of both, the metadata file and the actually loaded configuration. Loading a NanoHearth board leads to a choice of atmega1280, atmega2560, atmega168, atmega168p, ...
- Loading and saving said NanoHeart configuration (or a Gen7) leads to a configuration with only the CPU choices of the metadata file. Closing Configtool and loading it again gives only atmega1280 and atmega2560 as choice (and a question mark before "Board" in the tab title without changing anything, because the actually used CPU is still right).

Not sure on how to deal with these choice lists, right now I can't imagine a way to apply one metadata file to many different lists. Maybe it'd be wise to simply ignore changes to them, CPU options don't change that often, especially not by changes in Teacup.
 
Other than that it appears to work very well. Tried to remove an option and it vanished. Tried to add an option and it appeared. Very good, I like it!

jbernardis, 2015-07-13:
I am contemplating some additional changes to this structure.  I will upload later this evening after I implement them.

Hopefully this will address some of your issues above.  If not, then we can continue the discussion.

jbernardis, 2015-07-14:
I just now pushed up some new changes.  The metadata files are down to the bare minimum (the printer metadata file already was).  I also have examples of the printer and board files stripped to their minimum - for ramps1.3 and for mendel.  The original files are still there and are still valid to use; I just thought that it would be easier to see where this could go if I included some examples.  The new ramps1.3 and mendel files are still valid C and should compile.

I also addressed some of your issues:
   - metadata files can have values.  These values could be overlaid by a printer or board file, and then they can be changed again by the user for inclusion into their file.  Bear in mind that there are 3 files here.  The metadata files are global in context, the protected printer and board then go to the next level of specificity.  Finally the user file has values specific to the user's environment.  Even though I now allow values at the metadata level, they are, in my mind, more appropriate at the protected-file level.
   - the metadata files can be named anything we want, but they are different from the directly loadable files and so need to be kept apart from them.  We can do this either through the directory or the file name or both.  There should never be a need to load this file.  The user chooses a protected file they want to start with, and the metadata is loaded automatically.
   - I solved an issue with loading choices from both the metadata and from the board/printer file.  There will be no more duplicates.  Having said that, it probably makes the most sense to not define ANY candidate processors in the metadata file;  I think this specific data item is best defined at the board level.  The only processors that should be pre-defined at the meta level is a processor that is appropriate on every single board we have.

Traumflug, 2015-07-16:
To be honest, I tried this latest fix and the duplicate entries were still there.

So I took the first attempt and created a bunch of fixes onto it. It's in the *configtoolupdate2* branch. Duplicates are gone, because choices in the metadata file are now entirely ignored. Additionally a few minor changes which I think increase code robustness (let the parsing method decide wether it's a valid entry or not).

Biggest change is, all values are now remembered between load and save. For now this is only to restore disabled values instead of replacing them with the ones in the metadata file. In the future, this could be used to give the user some hint on what the defaults for disabled values are.

As far as my imagination for tests reaches, everything works fine. It even detects now keys present in the metadata file, but missing in the GUI, and vice versa. In such cases, a simple hint is written to the command line (no GUI message to not confuse the dummies :-) ).

Looks good? Did I miss to pick something important from the latest fix?

jbernardis, 2015-07-16:
I can't think of anything you might have missed.  The main thing is you got it working.  

As I was working on it, it was becoming apparent to me that the metafile is the wrong place for choices anyway.

Traumflug, 2015-07-30:
Looks like this is solved. Nice :-)

closed 2015-09-23 by Traumflug