Traumflug/Teacup_Firmware #170: My Build is failing on Windows 8 using Config Tool

wegunterjr, 2015-07-09:
I am trying to run the build on my windows machine. I made sure I Had python, wxpython, and arduino 1.7.6.   But the build is failing.

"C:/Program Files (x86)/Arduino\hardware\tools\avr\bin\avr-gcc.exe" -c -DF_CPU=16000000UL -mmcu=atmega2560 -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns="C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\build\usb_serial.al" -save-temps=obj -o "C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\build\usb_serial.o" "C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c"
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c: In function 'usb_init':
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:328:2: warning: implicit declaration of function 'HW_CONFIG' [-Wimplicit-function-declaration]
  HW_CONFIG();
  ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:329:9: warning: implicit declaration of function 'USB_FREEZE' [-Wimplicit-function-declaration]
         USB_FREEZE();    // enable USB
         ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:330:9: warning: implicit declaration of function 'PLL_CONFIG' [-Wimplicit-function-declaration]
         PLL_CONFIG();    // config PLL, 16 MHz xtal
         ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:331:18: error: 'PLLCSR' undeclared (first use in this function)
         while (!(PLLCSR & (1<<PLOCK))) ; // wait for PLL lock
                  ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:331:18: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:331:31: error: 'PLOCK' undeclared (first use in this function)
         while (!(PLLCSR & (1<<PLOCK))) ; // wait for PLL lock
                               ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:332:9: warning: implicit declaration of function 'USB_CONFIG' [-Wimplicit-function-declaration]
         USB_CONFIG();    // start USB clock
         ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:333:9: error: 'UDCON' undeclared (first use in this function)
         UDCON = 0;    // enable attach resistor
         ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:336:9: error: 'UDIEN' undeclared (first use in this function)
         UDIEN = (1<<EORSTE)|(1<<SOFE);
         ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:336:21: error: 'EORSTE' undeclared (first use in this function)
         UDIEN = (1<<EORSTE)|(1<<SOFE);
                     ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:336:33: error: 'SOFE' undeclared (first use in this function)
         UDIEN = (1<<EORSTE)|(1<<SOFE);
                                 ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c: In function 'usb_serial_getchar':
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:361:2: error: 'UENUM' undeclared (first use in this function)
  UENUM = CDC_RX_ENDPOINT;
  ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:363:6: error: 'UEINTX' undeclared (first use in this function)
  c = UEINTX;
      ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:364:16: error: 'RWAL' undeclared (first use in this function)
  if (!(c & (1<<RWAL))) {
                ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:366:15: error: 'RXOUTI' undeclared (first use in this function)
   if (c & (1<<RXOUTI)) {
               ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:374:6: error: 'UEDATX' undeclared (first use in this function)
  c = UEDATX;
      ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c: In function 'usb_serial_available':
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:389:3: error: 'UENUM' undeclared (first use in this function)
   UENUM = CDC_RX_ENDPOINT;
   ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:390:7: error: 'UEBCLX' undeclared (first use in this function)
   n = UEBCLX;
       ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:392:8: error: 'UEINTX' undeclared (first use in this function)
    i = UEINTX;
        ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:393:16: error: 'RXOUTI' undeclared (first use in this function)
    if (i & (1<<RXOUTI) && !(i & (1<<RWAL))) UEINTX = 0x6B;
                ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:393:37: error: 'RWAL' undeclared (first use in this function)
    if (i & (1<<RXOUTI) && !(i & (1<<RWAL))) UEINTX = 0x6B;
                                     ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c: In function 'usb_serial_flush_input':
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:408:3: error: 'UENUM' undeclared (first use in this function)
   UENUM = CDC_RX_ENDPOINT;
[... 166 lines cut ...]
      UECONX = (1<<STALLRQ)|(1<<EPEN);  //stall
                                ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:855:4: error: 'UDADDR' undeclared (first use in this function)
    UDADDR = wValue | (1<<ADDEN);
    ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:855:26: error: 'ADDEN' undeclared (first use in this function)
    UDADDR = wValue | (1<<ADDEN);
                          ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:869:6: error: 'UECFG0X' undeclared (first use in this function)
      UECFG0X = pgm_read_byte(cfg++);
      ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:870:6: error: 'UECFG1X' undeclared (first use in this function)
      UECFG1X = pgm_read_byte(cfg++);
      ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:873:11: error: 'UERST' undeclared (first use in this function)
           UERST = 0x1E;
           ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:933:19: error: 'STALLRQC' undeclared (first use in this function)
      UECONX = (1<<STALLRQC)|(1<<RSTDT)|(1<<EPEN);
                   ^
C:\Users\Walter.Gunter\Downloads\Teacup_Firmware-master\Teacup_Firmware-master\usb_serial.c:933:33: error: 'RSTDT' undeclared (first use in this function)
      UECONX = (1<<STALLRQC)|(1<<RSTDT)|(1<<EPEN);
                                 ^
RC = 1 - Build terminated
Build terminated abnormally.

Traumflug, 2015-07-09:
First, thanks for reporting.

> I made sure I Had python, wxpython, and arduino 1.7.6.

This part is apparently working fine, Configtool launches, the compiler is found.

> [...]-mmcu=atmega2560 [...]
and
> USB_FREEZE(); // enable USB

It looks like you use an ATmega2560 and also configured internal USB, which the ATmega2560 doesn't feature (ATmegas like 32U2 or 32U4 do). Make sure "USB Serial" is unchecked on the Board -> Communications tab. In doubt it's always a good idea to build with an unmodified configuration first, then doing changes one by one.

Now I'm wondering wether we could improve the wording of the label of this checkbox. Only few boards use this ATmega-internal USB, but almost all boards feature an ATmega-external, still on-board USB adapter. The latter isn't meant with this option.

wegunterjr, 2015-07-09:
ah...got ya. I went ahead and started over, and selected the mendel and ramps1.3 (there isn't a 1.4) without any changes, saved the config, and then proceeded to build. It dies along the way and I get a pop up saying ld.exe 

[image]

jbernardis, 2015-07-10:
This is odd.  ld.exe (or just ld) is the underlying linker that avr-gcc invokes.   It's not even a command line I build inside configtool.

There were no other errors?  The information on the screen is really not useful.  Would it be possible for you to copy/paste that command line on the screen into a batch file and then run it from a windows command line?  I'd really like to see if there are any messages to standard output that are being lost here because of the windowing interface.

The only thing that jumps out at me is the length of the command line.  Does anybody know if windows has a max length?

Traumflug, 2015-07-10:
I'd say this is a bug in ld.exe. No matter what garbage one tries to link, the linker should never crash (but report errors properly).

Possible workarounds:

- Go to Menu -> Edit -> Settings and remove one or both of the linker flags. Neither of them is neccessary, without them the binary gets just bigger.

- Try a different version of Arduino IDE. This will have a different version of the linker, which might not have this bug. Age of the IDE doesn't matter as long as it builds.

- Try to build with Arduino directly:
- - Rename the (top level) folder containing all the sources to "Teacup_Firmware".
- - Menu -> File -> Save board as..., then save not into config/, but directly into the main Teacup folder.
- - Same for the printer configuration.
- - Menu -> File -> Save config.h.
- - Then you can open Teacup_Firmware.pde directly in Arduino IDE. It should build and upload.

Traumflug, 2015-07-10:
I've updated Arduino IDE build instructions in the wiki: http://reprap.org/wiki/Teacup_Firmware#Installation_using_Arduino_IDE

Traumflug, 2015-07-10:
> The only thing that jumps out at me is the length of the command line. Does anybody know if windows has a max length?

It's apparently [32'768 characters](http://stackoverflow.com/questions/2381241/what-is-the-subprocess-popen-max-length-of-the-args-parameter) when using shell = False, which [is the default](https://docs.python.org/2/library/subprocess.html#popen-constructor). Also, Pythons popen() [should report a limit exhaustion](http://stackoverflow.com/questions/22385589/subprocess-popen-argument-list-is-too-long).

Executable and flags need some 400 characters. We currently have 29 object files, so each path can have about (32768 - 400) / 29 = 1116 characters. In the example above paths have about 80 characters with the -o in front of it, which should be fine.