Traumflug/Teacup_Firmware #172: Regressiontests

Traumflug, 2015-07-13:
Facing some quite fundamental changes to the code base during the generic ARM port, I just introduced some regression tests. Nothing super fancy so far, it just builds three standard configurations (Gen7, RAMPS, Teensy2) and the hostside simulator to make sure the situation isn't going bonkers. In the future there might be additional tests making sure the firmware actually works, G-code is parsed correctly and such stuff.

Invocation is simple:

    make regressiontests

Already on the *experimental* branch. Big thank you to the guy who introduced USER_CONFIG in the Makefiles, this came in quite handy here.

From now on, please run regressiontests before committing any changes to the firmware. Takes just some 20 seconds. Doing so nailed 4 bugs already :-)

triffid, 2015-07-13:
sounds like time for continuous integration!

phord, 2015-07-13:
Sweet!  Now you also need combinations of config files to ensure, for example, that my changes in #104 don't break when SDCARD is enabled, RAMPING is disabled, and an extra Heater is defined, for example.  :-)

Traumflug, 2015-07-13:
Well, create them. It's as simple as saving such a configuration in testcases/ and duplicating a line in Makefile-common.

Traumflug, 2015-07-30:
Just committed another Git tool, *git-regtest*.

The point of this tool is to run regression tests over a whole series of commits. Like if one wants declare a topic branch as 'done', ready for *experimental*, it's a good idea to do something like this:

    git checkout <topic branch>
    tools/git-regtest experimental

This will checkout experimental, run 'make regressiontests', then checkout the first commit of the branch, run 'make regressiontests' again, and so on, until it hits HEAD of the topic branch. It makes sure every single commit is fine, at least as far as automated testing can detect it.

Works for any commit older than the current one, for example

    tools/git-regtest HEAD~5

runs regression tests for the last 5 commits.

Another use it to find a commit which introduces a regression. Use it as before, the tool will stop at the commit introducing the regression.