Traumflug/Teacup_Firmware #174: Variable number of arguments and 16 vs. 32 bits

Traumflug, 2015-07-17:
In sersendf.c we have this code:

```C
#include	<stdarg.h>

#ifdef SIMULATOR
  #define GET_ARG(T) (va_arg(args, int))
#else
  #define GET_ARG(T) (va_arg(args, T))
#endif
[...]
    if (j == 4)
        serwrite_uint32(GET_ARG(uint32_t));
    else
        serwrite_uint16(GET_ARG(uint16_t));
```
On 32-bit ARM, this throws the following warning:

```
sersendf.c:122:52: warning: 'uint16_t' is promoted to 'int' when passed through '...'
             serwrite_uint16(GET_ARG(uint16_t));
                                                    ^
sersendf.c:122:52: note: (so you should pass 'int' not 'uint16_t' to 'va_arg')
sersendf.c:122:52: note: if this code is reached, the program will abort
```

va_arg man page comes with an example, which talks about "promoted types". Whatever that means:
```C
              case 'c':              /* char */
                   /* need a cast here since va_arg only
                      takes fully promoted types */
                   c = (char) va_arg(ap, int);
```
Now I'm obviously searching for a proper solution. Ideally one without #ifdef __ ARMEL__, #ifdef __ AVR__ and such kludges. Googling didn't bring up much. Also nothing about finding out what a "fully promoted type" is. Is it safe to always use an int, together with a cast? Like

```C
    serwrite_uint16((uint16_t)va_arg(args, int));
    serwrite_uint32((uint32_t)va_arg(args, int));
```
On 16-bit, int is 2 bytes, while fetching an uint32_t should pull 4 bytes from the argument list.

Or is there some way to test for the size of the promoted type? Like:

```C
#define GET_ARG(T) ((sizeof(int) < sizeof(T)) ? (T)va_arg(args, int) : (T)va_arg(args, T))
```
(probably won't compile warning-free, but I hope you get the idea.)

Thanks for the insights :-)

triffid, 2015-07-18:
I'd use a typedef in the platform-specific header, where we expect to see
ifdefs

Traumflug, 2015-07-18:
A *typedef*? Can't see right now where to put a typedef. Could you elaborate a bit?

One thing I found while crawling through the predefined macros and which should work reasonably as long as there are no 64 bit arguments:

```C
#if __SIZEOF_INT__ == 2
  #define GET_ARG(T) (va_arg(args, T))
#elif __SIZEOF_INT__ >= 4
  #define GET_ARG(T) ((T)va_arg(args, int))
#else
  #error Not prepared for this integer size. 
#endif
```

triffid, 2015-07-18:
in avr.h or whatever you're using, typedef promoted_uint16_t uint16_t, and
in arm.h typedef promoted_uint16_t uint32_t

then serwrite_uint16(GET_ARG(promoted_uint16_t)) as necessary

Traumflug, 2015-07-18:
OK, got it. Thanks for the help, @triffid.

closed 2015-07-18 by Traumflug