Traumflug/Teacup_Firmware #175: Configtool: wxPython installed, but not importable?

Traumflug, 2015-07-19:
See http://forums.reprap.org/read.php?1,516185,534248#msg-534248

He says he has Pronterface running, but Configtool refuses to import wx. Any ideas on how this can happen? I just verified that code for both is the same:

```Python
try:
  import wx
except:
  [...]
```
Is it possible wx is actually found, but buggy enough to raise an exception its self?

jbernardis, 2015-07-19:
If wxpython is properly installed,  it should be available to any program trying to import it. 

Does he perhaps have it installed in the printer face directory and not in the python library directory?

ntoff, 2015-07-19:
>I already had Pronterface but there seemed to be no python installed

That tells me he's using the precompiled pronterface. He's probably mixing 32bit and 64bit python/modules. To play it safe even though I'm on 64bit windows, I only install the 32bit version of python and its modules.