Traumflug/Teacup_Firmware #177: Configtool doesn't have place to enable internal pullup resistors

madmike8, 2015-07-25:
If it's there and I missed it... I'm sorry... I really really did search for it...

I'm setting up the UNO CNC shield V3, but the endstops didn't work. Looked in Configtool, and there was no place to enable internal pull up resistors. There was also no documentation in config.h as listed on the wiki. I googled and found the information I needed and manually added it to the config.h. Endstops now work. Even if the configtool doesn't offer a way to enable them, at least the documentation on them should be put back into the config.h

Traumflug, 2015-07-25:
It's a property of the printer, the best setting depends on the type of endstops connected. Printer tab -> Miscellaneous -> second from top in the left column

Wiki information might be a bit out of date, that's true. Can I invite you to update the wiki?

madmike8, 2015-07-25:
Doh! Oh well... I updated the wiki.

Thanks

Traumflug, 2015-07-25:
Hey, we have a issue "fixed" :-) Can we close it?

closed 2015-07-25 by madmike8