Traumflug/Teacup_Firmware #179: Gen3 support broken

Traumflug, 2015-07-27:
@phord wrote at https://github.com/Traumflug/Teacup_Firmware/commit/face7c0dc7ff7ba0050d359ee8475ef4cc8782bf#commitcomment-12381759:

I noticed two PIN definitions got left out of this file.  They are needed for the INTERCOM feature.  I suppose this feature is still supported, but I do not know if anyone is using it.

    #define TX_ENABLE_PIN            DIO12
    #define RX_ENABLE_PIN            DIO13

I have added them back in another commit (116238d) , but I have not looked to see how this affects the template code.

Traumflug, 2015-07-27:
See me surprised somebody noticed this :-)

I also think this commit is done well, but incomplete.

- New pins should be added to all board configs.
- More importantly, it should also be added to configtool/board.generic.h. Else Configtool will simply drop these pins when saving.

So I put this commit onto a new branch *issue-179* and restored the former *experimental*. I've more to say, but sending this now to inform you about the restored experimental branch.

phord, 2015-07-27:
I added the items to all the board configs and to board.generic.h.

I expect ConfigTool needs to learn to configure this feature, too, but I probably am not the one to decide how to do that.  Does it go on the "Communications" pane?

Traumflug, 2015-07-27:
> I added the items to all the board configs and to board.generic.h.

Excellent! Picked to *experimental*.

To be honest, I was close to drop Gen3 support recently. If one heard at all about printers using such a controller over the last two years, the first step their owners did was to replace the controller with a RAMPS and a stepper based extruder.

Is there any Gen3 user left on this planet? Especially the stuff in extruder/ didn't receive much attention for years, just updated by best guess when the code there showed up using "grep". Does Teacup for the extruder board build? Does it work? Any insight welcome :-)

phord, 2015-07-27:
I doubt anyone is using gen3 and current code. I only noticed this problem because I wanted to test breaking code paths while moving includes around. But if we drop gen3 support, we can also dump the intercom junk. 

Wouldn't bother me.

Traumflug, 2015-07-28:
Let's see:

http://forums.reprap.org/read.php?1,538423
http://3dprintboard.com/showthread.php?14775-Is-any-Generation-3-Electronics-user-out-there

Traumflug, 2015-07-30:
BTW., it looks like this is still broken, adding these two pins to board.generic.h wasn't sufficient. If I load a gen3 board in Configtool and just save it again, these two pins are lost. First guess is, because there's no GUI choice for them, so they're not in self.choices.keys() in getValues() in pinoutspage.py. Is this behaviour, dropping unknown pins, a bug?

Won't fix this now, in these two forum questions nobody using a Gen3 extruder board shows up. Looks much like support for this extruder board is going away anyways. Off the top of my head this means:

- The whole extruder/ directory going away and all references (in Makefile, etc.) to it.
- The definitions MOTHERBOARD and EXTRUDER. Code for MOTHERBOARD is kept, code for EXTRUDER deleted.
- There's also no longer a need for a DC motor driven extruder? Configtool takes great care to have a good user experience here.
- Same for TEMP_INTERCOM. There are extruder extension boards, e.g. for the Duet, but I'm pretty sure they work entirely different.

From this point on, a Gen3 is a pure stepper controller with SD card. I think so.

benj919, 2015-08-02:
Hi There
I'm right now in the process of bringing (or at least trying to) an old makerbot cupcake back into service. 
I do have a new printer at hand and want to use this old one to use up a few kg of 3mm filament, so no spending money on new boards/extruders...
If you're just dropping support for it, is there an older snapshot of this firmware supporting such an ancient config?
It has Gen3 electronics, a HBP, an MK5 Plastruder (DC Motor) and no endstops..

cheers

Traumflug, 2015-08-02:
Gen3 support wasn't removed, yet, so you can give it a try as-is. The build in the main directory for the main board (with or without Configtool); in *extruder/* for the extruder board (without Configtool).

... and don't be too disappointed if it doesn't work on the spot. You're probably the first since two years who tries.

benj919, 2015-08-03:
Thanks I'll try it, though it will probably take me a week or two

Traumflug, 2016-06-06:
Finally moved this patch, or more precisely: the remaining part of this patch, to *experimental*. Thank you very much for this contribution!

If there are issues remaining, please don't hesitate to open a new report.

closed 2016-06-06 by Traumflug