Traumflug/Teacup_Firmware #185: Feedrate can be set to zero

phord, 2015-09-04:
I noticed by chance that this command will crash the simulator (floating point exception; divide by zero):

    G1 X1 F

It's because target->F == 0, and it's often the denominator.  I actually noticed this when testing with parsertests.txt because it has the word "for" in it which has the same effect.  I suppose the ARM doesn't have the chance to segfault so it simply continues on with bad results.  This should be cleaned up somewhere, but I'm not sure where at the moment.  In the parser?  In the dda?

So I'm leaving it in this issue instead for now.  The problem existed for a long time now and no one has noticed.  But it should be cleaned up.

Traumflug, 2015-09-06:
Nice catch :-)

Two places for catching this come to mind: dda_create(), where maximum speeds are checked as well and process_gcode_command(), where machine limits are tested.

The question is: what to do with such an obvious syntax error? F0 doesn't make sense and so far Teacup does no syntax checking at all. Simply set it to the minimum, to F1?

Wurstnase, 2015-09-07:
I would simply ignore such values.