Traumflug/Teacup_Firmware #186: Faster distance algorithm

Traumflug, 2015-09-28:
Today distance algorithms faster than sqrt(x * x + y * y) crossed my way, so let me put this here to not forget it.

1. hypot(), precise and said to be part of standard libc: https://en.wikipedia.org/wiki/Hypot

2. Alpha max plus beta min algorithm, approximate and even faster: http://www.dspguru.com/dsp/tricks/magnitude-estimator

3. CORDIC: http://www.dspguru.com/dsp/faqs/cordic, https://en.wikipedia.org/wiki/CORDIC, https://www.mikrocontroller.net/articles/AVR_Arithmetik/Sinus_und_Cosinus_%28CORDIC%29 (german)

phord, 2015-09-28:
The 2nd item (alpha max + beta min) is a detuned version of the dist_estimator we currently use in approx_distance.  It is surprisingly accurate (in the tuned version).  I did some extensive tests on it a couple of years ago and even did some work to further improve the estimation here: 812b0b265e9d66cf8ea9f6f91ef66313c0216c51 

Our estimator works by choosing the best alpha/beta combo among four options based on the ratio of max:min.  I improved it and scaled it up to 8 options in that branch.

hypot() looks interesting, but I expect it uses sqrt internally and may be slow.

Cordic looks like a table lookup mechanism, but I haven't explored it deeply.

Traumflug, 2015-09-28:
Oh, we have this already? That's excellent :-)

Yes, CORDIC does kind of a table lookup. Not the usual lookup of nearby values for linear interpolation, but a table of 1/1 value, 1/10 value, 1/100 value and so on, then these table values are added up as needed. An implementation for exp() is on the *thermistortables* branch:

Base 10, used in the HP-35: https://github.com/Traumflug/Teacup_Firmware/commit/7830455e399a6c86960a0d18417d11ab200322b7

Base 2, faster on binary machines: https://github.com/Traumflug/Teacup_Firmware/commit/40b184f7a343b4953fc5a100007f8d6c8e436083

Traumflug, 2015-12-26:
Problem solved :-)

closed 2015-12-26 by Traumflug