Traumflug/Teacup_Firmware #187: Position Problem after Homing

tiago1080550, 2015-10-31:
Grettings,

I encountered an issue that i am unable to figure out. I have assembled a structure that uses the core xy kinematics(no z axis). After I send the GCODE to home the Y axis, he stops at the end stop and all it's good. The problem is sometimes, if I do a rapid move order (G0 Y10) after this homing movement, the motors will move in the oposite direction against the structure frame for no reason (i have defined the X_MIN and Y_MIN as 0 on config.h).
I'm i doing something wrong?

Thanks in advance

youssefaly97, 2015-10-31:
Currently G0 and G1 are the same in Teacup. Make sure all axes are correctly oriented. Does it do that every time ?

Shutdown the motors, bring the head with your hand roughly to the centre of the area of motion. Send G92 X0 and Y0 to set the current position as the origin. Then send G0/G1 commands to move a couple of centimeters in each direction to identify your axes' orientations. 

This is very harmful for the mechanics so while testing, make sure the motor currents are set at the minimum so any fault will stall it which is much better than ramming the carriages against the edge and having the belt slip.

tiago1080550, 2015-11-01:
Thanks for your repply. I'm using pronterface to help with the testing, and the axis orientation is as I intended. And no it does not do that every time, thats why i think it is strange. Sometimes even a slight movement in one of the axis directions after homing (G0 Y1) and the head will force the structure, others works as inteded.

Traumflug, 2015-11-01:
> Currently G0 and G1 are the same in Teacup.

Except for movement speed, so using a slow movement for testing is certainly a good idea.

And yes, a reproducible test case would be great. Likely it has to do something with the position where the controller was reset the last time.

Looking at the dda.c, this function looks suspect:

```C
/*! Distribute a new startpoint to DDA's internal structures without any movement.

  This is needed for example after homing or a G92. The new location must be in
  startpoint already.
*/
void dda_new_startpoint(void) {
  enum axis_e i;

  for (i = X; i < AXIS_COUNT; i++)
    startpoint_steps.axis[i] = um_to_steps(startpoint.axis[i], i);
}
```

Traumflug, 2015-11-01:
I mean, the above function assumes a 1:1 relationship between axis position and stepper position, which isn't true for the various 'core' mechanics types.

youssefaly97, 2015-11-01:
Traumflug:
Thanks for pointing out that detail. 

Tiago:
You do know you need to home with G28 Y right ?

tiago1080550, 2015-11-01:
Yes, of course. That's how i did the homing movement

Traumflug, 2016-05-26:
Similar issue: #216.

phord, 2016-05-31:
I think this is the same as #216.  In fact, 216 was a dup of this one.  I didn't find this one when I did my search for a prior report.  Maybe @tiago1080550 can test now to confirm it is fixed on experimental.

Traumflug, 2016-06-01:
Closing this as duplicate of #216. If there's something left, please write there.

closed 2016-06-01 by Traumflug