Traumflug/Teacup_Firmware #188: Professional PID tuning

youssefaly97, 2015-11-01:
I've reached out for help from electrical control engineers to tune the PID loop. They asked about what seems to be known as "The Transfer Function". Can someone give me the transfer function we're using for teacup ?

Traumflug, 2015-11-01:
The right question would be: "what transfer function has your heater?". It's a property of the heater, not of the controller.

A first step to improve tuning would be to find a procedure to measure this function experimentally. Another approach would be to find a procedure on how to improve the PID values by iteration (try, look at the response, readjust values, repeat, repeat, repeat, ...). A Python script doing this would be awesome, it could be part of Configtool.

Traumflug, 2016-04-14:
Looks like @Wurstnase takes care of this, let's continue with issue #206.

closed 2016-04-14 by Traumflug