Traumflug/Teacup_Firmware #189: STM32F411 port

Wurstnase, 2015-11-05:
Hi all,

I think it's better to make a new issue for that port. Currently I stopped with the Due-port.

Anyhow, the first progress is done. Serial works and I start to simplify it.

I delete the mbed-hal_tick_stm32 parts. But with that, the STM starts "start\nok\n" but after additional 4 or 5 chars it stops. I don't understand this part completely. It reconfigure the Systick? Any hints?

Wurstnase, 2015-11-05:
I found some ideas. I will test them and report.

Traumflug, 2015-11-05:
- Does delay_us() work? It's based on NOPs and more advanced CPUs might optimize this away internally, even if the compiler doesn't.

- Is the maximum string length about the same as the buffer size? If yes, characters might be written into the already filled buffer.

- Systick should be independent from serial workings. Especially with the baud rate being right.

Wurstnase, 2015-11-05:
delay.c isn't ported.
This one
```
// Reconfigure the HAL tick using a standard timer instead of systick.
HAL_StatusTypeDef HAL_InitTick(uint32_t TickPriority)
```
reconfigure it. That is called in the mbed-system_stm32.c.

After deleting the mbed-hal_tick_stm32.c the ```__weak HAL_StatusTypeDef HAL_InitTick(uint32_t TickPriority)``` is called.

The baud-rate works perfectly (the divisior of my last pull was wrong, but correct on my local version). But without the ticker I can only see the first 16? chars.

Traumflug, 2015-11-05:
16 bytes might well be the size of the UART send buffer. If you write more in one chunk, the remaining bytes are lost. You have to wait until some of the bytes are sent. serial_writechar() in serial-arm.c on master/experimental shows how it can be done. Well, either you find other tasks to spend time or you have to port delay.c before sending larger amounts.

delay-arm.c should work without changes. For testing delays without a scope simply use delay_ms(10000), send a character before and after, and if these two characters are sent with 10 seconds in between, you're fine for both, delay_ms() and delay_us(). Precision isn't crucial, stepper motors are timed by other mechanisms.

Wurstnase, 2015-11-06:
To precise it. I also add 
```
for (;;) {
  if (serial_rxchars() != 0) {
    serial_writechar(serial_popchar());
  }
}
```
But it doesn't show chars anymore after deleting the ticker. I hope I have more time this afternoon.

triffid, 2015-11-06:
Sounds like some important interrupt flags are set up by the ticker, serial
relies on the transmit empty interrupt to drain the ringbuffer.

Without interrupts working I would expect it to fill the output fifo once
and then get stuck

Traumflug, 2015-11-06:
Perhaps you simply expect a bit too much from the first few commits. If serial out works, fine, time to move on to the next commit.

Wurstnase, 2015-11-06:
Git is not my best friend... Or... Learning git the hard way :)

I start a new branch with cherrypicked commits and a clean rebase. I made some big mistakes at the beginning.

Wurstnase, 2015-11-06:
Or @Traumflug what is the best practise for rebase? Should I push after any rebase?

My first issue was to manipulate the corrupted files wrong. I've took 'ours' not 'theirs'. Or reverse?!?

Traumflug, 2015-11-07:
> I've took 'ours' not 'theirs'. Or reverse?!?

I sometimes wonder where people learn such stuff. "It doesn't work! It doesn't work!" ... and if one manages to catch such people before they rush away one hears about attempts which can't be found in any tutorial and only in a small corner of the manual. Happens not too rarely.

OK, this little story aside, one uses neither 'ours' nor 'theirs'. At least not until it's 100% sure it does the right things. Without them, conflicts can happen, that's the normal behaviour. Editing conflicts is no rocket science, one searches the conflicting file for ```====```, which will find two versions of the same code section. These two versions are joined to one by plain text editing. What to keep and what to delete depends on what the result should be. Along the way these three marker lines are removed as well. After editing a conflict one saves the file, then adds it with ```git add```.

All files done, one continues the rebase with ```git rebase --continue```. Until the rebase is completed. If it's unclear wether it's completed, simply try another --continue; it never hurts.

Unfortunately, Git is a bit unclear on which the conflicting files are. Lots of messages, sometimes with, sometimes without the file names. The easiest way to find the file(s) is to simply try to continue with ```git rebase --continue```, then it's simply lists the not yet solved ones (it uses the word 'merged', which is a bit misleading).

Wether to push or not to push is entirely independent from all this. Pushing after a rebase obviously makes most sense. You can push before, too, especially if the rebase doesn't work the way you like it.

Not to forget: one can always abort a rebase with ```git rebase --abort```. This will restore the repo to the point before the rebase attempt, without any clutter or half-done tasks dangling around.

Traumflug, 2015-11-07:
P.S.: for playing with another branch to not mess up the original (good idea!) I'd simply duplicate the existing one: ```git branch <oldname> <newname>``` Then you can rebase and rebase -i as much as you like without destroying anything.

triffid, 2015-11-07:
One thing to keep in mind with rebase is that if you rebase a pushed branch
it won't push again unless you force, and anyone else who wants to pull
will have to force pull and any branches they had based from yours will
have to be rebased too.

Generally I suggest that folks rebase local branches but merge branches
that exist elsewhere, Traumflug seems to have created his own workflow that
doesn't quite work like this though ;)

Wurstnase, 2015-11-07:
My way of learning is doing it. Anyhow, I rebase it exactly the way you discribed, the only thing I was wrong, I thought the rebase I made were more in the front than it is.

I start a rebase. Saw that something is wrong. Abort a rebase. Start again... 
However...

> One thing to keep in mind with rebase is that if you rebase a pushed branch it won't push again unless you force

Ok, a new point I've didn't know. Maybe this is also something which get me crashed?!?

Traumflug, 2015-11-07:
> Traumflug seems to have created his own workflow

This workflow certainly isn't an invention of me, it's well documented as "topic branches". Git comes with quite a number of tools to support this workflow, after all. Even the Git folks them selfs use this strategy.

> Maybe this is also something which get me crashed?!?

Again: no, can't happen, rebasing and pushing are two entirely intependent tasks.

> I start a rebase. Saw that something is wrong. Abort a rebase. Start again...
> However...

... however you don't tell us *what* went wrong.

Wurstnase, 2015-11-07:
It just something I made wrong in one of the first commits. I made a rebase, vote the wrong 'ours'/'theirs' and the problems began. I had a lot of broken commits. And it was getting worse and worse with each commit. Just something I have to learn. I think I was right, but now I know I wasn't.

Hopefully this new one works better. The first rebases works pretty fine and I think I'm on the right way now.

Wurstnase, 2015-11-07:
[image]
What I am doing wrong? I pull my last push. Rebase to experimental~73 (fastio). Repair, commit.

Traumflug, 2015-11-07:
> Repair, commit.

Repair, add, rebase --continue

Other than that, what's the problem? You have 6 commits locally on top of the remote branch. Looks perfectly fine.

Wurstnase, 2015-11-08:
Repair, add, rebase --continue until nothing is to do anymore. Right. 
I didn't rebase the experimental, sorry for confusion.

So what I was doing exactly.
- rebase experimental ~74
- add, rebase --continue
- git push
until this anything fine
- git pull
and here I need to repair my just pushed branch. I don't rebase anything.

Wurstnase, 2015-11-08:
Forgot what I've been written. I have a tool called Smartgit. But it's too smart.
It makes a
```git fetch --progress --prune origin```
when I normally want to pull.

Pure git works perfect.

Traumflug, 2015-11-08:
> rebase experimental~74

Looking at what's in the public repo, branch *arm-stm32f411-port* is based on *experimental~74* already, so it's no surprise another rebase attempt does nothing. To move up the ladder you'd have to rebase to *experimental~73*.

Wurstnase, 2015-11-08:
Anyhow, the main-issue was Smartgit.
Fastio works now :)

Wurstnase, 2015-11-08:
It couldn't be so complicated.
Git pull
Git rebase experimental~68 (or 67 or whatelse)
Repair, add, rebase --continue...
Whatever I made, an old commit is duplicated in front.

Traumflug, 2015-11-08:
> Whatever I made, an old commit is duplicated in front.

What is the content of this "duplicate"? If commits don't match 100%, the differing part is left in the rebased branch. Then you have two commits with the same description, but different content.

If this difference is just some whitespace or something neglibile, you can remove the commit with ```git rebase -i experimental~68```. An editor will open, listing all commits. Removing the line with the commit in question will also remove the commit from the branch.

If the difference is bigger you have to decide wether it needs to be kept or what else to do with it.

Wurstnase, 2015-11-09:
Ok, one hopefully last question about this.

```
git checkout arm-stm32f411-port
git pull
git rebase experimental~xx
git add xy
git rebase --continue
git add xz
git rebase --continue
```

Now I get a message like this:
```
git status
On branch arm-stm32f411-port
Your branch and 'origin/arm-stm32f411-port' have diverged,
and have 21 and 18 different commits each, respectively.
    (use "git pull" to merge the remote branch into yours)

nothing to commit, working directory clean
```

Everything is perfect, until I made a ```git pull``` where my problems start.

Traumflug, 2015-11-09:
> git checkout arm-stm32f411-port
> git pull

Again you mix rebasing with pushing/pulling. Please stop it! If you're the only contributor to this branch there's never a need to pull anything.

> Your branch and 'origin/arm-stm32f411-port' have diverged,

This is an informational message and expected after a rebase. As the diverge appeared because your local branch changed, not because the remote one did, this advice to pull is wrong. A ```git push -f origin arm-stm32f411-port``` will upload your local branch to the remote repo, which will also make them equal, so the message will disappear.

Wurstnase, 2015-11-09:
>> git checkout arm-stm32f411-port
>> git pull
> 
> 
> Again you mix rebasing with pushing/pulling. Please stop it! If you're the only contributor to this branch there's never a need to pull anything.

I'm on a different computer acutally and want to be sure, that I have the latest version.

> A git push -f origin arm-stm32f411-port will upload your local branch to the remote repo, which will also make them equal, so the message will disappear.

Ok, now I'm on the right way. That's the one I need. Thanks!

Traumflug, 2015-11-09:
> I'm on a different computer

In this case you're technically two users. Do the rebase on the computer with the newer development, after having this done, push it to the public repo, then go to the other computer and pull there.

That said, using "pull" can cause trouble. A "fetch" always works, because it doesn't update local branches. I'd do it like this:

```bash
git fetch
git branch -d arm-stm32f411-port            # delete the local branch
git checkout -b origin/arm-stm32f411-port   # create a new local branch from the remote one
```

This is a strategy which never causes conflicts.

Wurstnase, 2015-11-10:
Yeah, the most mystics are in my head now. :+1: 

I will made a reset to the last wrong ```git pull``` and will rebase/cherry-pick it as it should be.

I have some additional branches with minimized working serial. Btw. the Pin-Registers is a CortexM0 only thing?!? I don't find any similar at my M4. There are only port-register, but therefor a lot.

Traumflug, 2015-11-10:
> Btw. the Pin-Registers is a CortexM0 only thing?!?

Pin I/O is considered to be a "peripheral", so yes, it might be different on your MCU. Likely less a "M0-only" thing, more a "NXP-only" thing.

Wurstnase, 2015-11-11:
History is more clean now. ```git rebase -i HEAD~x``` is a nice way to repair some commits.

The SystemInit() needs a total rework. The RCC-part is hardly nested. I hope I can find a clean way.

Traumflug, 2015-11-11:
> git rebase -i HEAD~x is a nice way to repair some commits.

Glad to see you start to see the advantages of Git and topic branches!

Wurstnase, 2015-11-11:
So much really cool things in there!

When I use ```gitk --all&``` I have now a lot unused tracks. How could I clean this?
[image]

Traumflug, 2015-11-12:
Huh? A branch without tag or branch name? Not possible with Git's data structure. Looks like there's something wrong with gitk. Do you use Menu -> File -> Reload (Shift-F5) ? Menu item 'Update' (F5) isn't sufficient (and I'm wondering what the use of it is).

Wurstnase, 2015-11-12:
> Reload (Shift-F5)

:+1: I was hitting just F5.

phord, 2015-11-12:
Fwiw, I frequently have unnamed and untagged branches.  Submodules keep me "detached" so frequently that I just get used to it.  Try this in a project where you have rebased or otherwise abandoned changes:

  gitk $(git reflog --format=%h)

The reflog is useful for finding your way back to old locations (like the branch before you rebased, etc.).

Wurstnase, 2015-11-12:
Nice, thank you!
Just used something similar yesterday with ```git reset --h/s/m HEAD@{x}```

Wurstnase, 2015-11-19:
Eureka! Filesize is down to ~1300 bytes pre-delay.
The clock configure thing is really complex.

@Traumflug, you put a delay in the ```serial_writechar()```. Exactly ``` delay_us((1000000 / BAUD * 10) + 1);```. I need ```+7``` in the end. How does the compiler calculate this value? 
Because:
```
1000000 / BAUD = 8
8 * 10 + 1 = 81
```
or
```
1000000 / BAUD * 10 = 86
86 + 1 = 87
```
I guess it's more the former, and this is the resone I need a ```+7```?!?

Traumflug, 2015-11-19:
The 1000000 is to get microseconds. The 10 is to get transfered bytes from bit speed (8 data bits + 1 parity bit + some allowance). The 1 is to always round up. For testing you can simply send a longer text, like 200 bytes. If this gets sent completely you're fine. In doubt, make it a bit longer, doesn't hurt much.

Wurstnase, 2015-11-19:
Thanks for explanation.
I've tried it and only after adding additional 6µs the text was send completely.

Wurstnase, 2015-11-19:
Is it a problem to do this ```while (!serial_txchars()); ``` instead? Or what is the reason for using the delay?

```
/** Check wether characters can be written 
*/
uint8_t serial_txchars(void) {
  return USART2->SR &USART_SR_TXE;
}
```

Traumflug, 2015-11-19:
Yes, it can be a problem. If characters aren't written for some reason or another, such a loop would block forever. That's why I prefered the delay.

Wurstnase, 2015-11-19:
I reached the regression-test. I've installed the avr-stuff and made a thermistortable.h but I can't compile it.

[regression-test](https://gist.github.com/Wurstnase/d39d5a7870f22ee9abf4)
[thermistortable.h](https://gist.github.com/Wurstnase/fc34aeaf61c4fce18ef1)

Wurstnase, 2015-11-21:
@Traumflug How do you measure the max frequency? Just G1 X10000 F100000 and looks what happened on the scope?

Traumflug, 2015-11-21:
http://reprap.org/wiki/Step_rates#Measuring_step_rate

Wurstnase, 2015-11-22:
I'm struggling a little bit with the stepper timer.

I have a pulse time of the debug LED at 1.7µs. 

At 91.6kHz it starts to accelerate and decelerate twice. It starts and accelerate to 91.6kHz. After that it brakes instantly to 4kHz. When it starts with decelerating it jumps back to 91.6kHz and slows down.

Traumflug, 2015-11-22:
> At 91.6kHz it starts to accelerate and decelerate twice. It starts and accelerate to 91.6kHz. After that it restart with accelerating just to 4kHz.

A typical example for overload. Some variables obviously get overwritten. Acceleration calculations happen in the second half of dda_clock(), so if you can connect with a debugger, I'd try to find the variable with a watchpoint. Else it might help to add a plausibility test, like

```C
uint32_t move_step_no_verify;
[...]
    ATOMIC_START
      move_step_no = move_state.step_no;  // already there
      if (move_step_no < move_step_no_verify)
        serial_writechar('x');
      move_step_no_verify = move_step_no;
    ATOMIC_END
```

This is just an example for such a plausibility test, I can't say what the actual problem is. Only one 'x' per movement should appear on the serial line.

Another possible problem is that the 2 ms interrupt isn't protected against re-entry. There is some protection code for AVRs in timer-avr.c (```if ( ! busy) { ...```). LPC1114 does this on the hardware level. Perhaps the STM32 has no such re-entry protection in hardware.

> I have a pulse time of the debug LED at 1.7µs.

At 84 MHz, that's just 143 clock ticks. Excellent! LPC1114 needs some 290, AVRs need 320.

Wurstnase, 2015-11-22:
> At 84 MHz, that's just 143 clock ticks. Excellent! LPC1114 needs some 290, AVRs need 320.

At 96MHz, so it needs ~164 ticks. 
But it would be better with 290 and running with 130kHz ;)

```volatile uint32_t move_step_no_verify;```?

without volatile I don't see any 'x'. With I see ~10,000. 
Movelength 640,000 steps, max-speed should be 96kHz.

Traumflug, 2015-11-22:
Yes, volatile is a good idea :-)

Now the question is, how move_state.step_no gets smaller. Shouldn't happen, there's only a ```move_state.step_no++;``` in dda_step(), no decrement. Possible bugs include an overflow of move_state.steps[], see line 58 in dda.h.

Wurstnase, 2015-11-22:
Hmmm... I doesn't do anything wrong. I've taken the verify in the wrong position. Sorry.

Wurstnase, 2015-11-22:
> Only one 'x' per movement should appear on the serial line.

Should appear 'x' once per move or could appear?

Wurstnase, 2015-11-23:
I made yesterday some more debugging.
There are more than just one issue.
One is the multi-accelerating issue. With some more speed I've got 3 accelerations/decelersations.

But also the timer counter overtake the match register sometimes.

Wurstnase, 2015-11-23:
In dda_create() are some interessting comments. Maybe I run in an overflow while testing. I think, doesn't remember 100%, that I've tested moves longer than 2000mm.

My testing parameters are something like this.
640.000 steps/m
100 acc
96MHz clock speed.

Traumflug, 2015-11-23:
> I've tested moves longer than 2000mm.

Oh, didn't think of this. My review against overflows always assumed a max. travel distance of 1000 mm, so overflows on larger distances aren't unlikely.

Wurstnase, 2015-11-23:
Yes, testing a naked controller can show unexpected behaviours. No one, or 99.999% move longer than 1800mm. The minimum size of this printer must be >1.3m.

Traumflug, 2015-11-23:
> The minimum size of this printer must be >1.3m.

Looks like a code audit is required to squeeze the last few bits out of 32-bit signed integers. And/or to avoid signed ones at all. Signed integers are used only in the lookahead code, AFAIR.

Another solution could be to multiply the steps/mm value by 10 and change the G-code interpreter to divide distances by 10 (move the decimal point one to the left). That's about what floating point would do, just with less precision.

Third option is to use 64-bit integers, of course. Not a quick solution either, because some functions have the assumption of 32 bits hardcoded. For example inv_sqrt().

Wurstnase, 2015-11-23:
Or by simple testing long moves. Any move < 1800mm are ok. Any above could use something like:
```
#if defined (HUGE_PRINTER)
  if (distance > 1790)
    uint32_t move_duration = ((distance * 240) / dda->total_steps) * (F_CPU / 4000);
  else
    uint32_t move_duration = ((distance * 2400) / dda->total_steps) * (F_CPU / 40000);
#endif
```

Wurstnase, 2015-11-23:
@Traumflug I'm missing a commit from you. In the past I saw a commit for the 16bit buffer, which could do something wrong and could also happen on AVR. Maybe you remember it?

Traumflug, 2015-11-23:
> Or by simple testing long moves

There are occasions where the total distance from origin is calculated. For example, dda.c, line 190ff.

> it's unsigned already.

Yes, this part. Lookahead code not. dda->delta_um[] is signed.

A bit about the future of lookahead: currently start and stop ramps are stored in steps, which works perfectly for linear kinematics, but not so well for non-linear ones (delta, scara, ...). Storing these two points in microseconds would likely be an advantage and change quite a number of calculations in dda_lookahead.c.

> In the past I saw a commit for the 16bit buffer, which could do something wrong and could also happen on AVR.

'16 *byte* buffer'? Can't see what you mean. On ARM this buffer is in hardware, zero code fo it. There was a commit for the software buffer on AVRs: https://github.com/Traumflug/Teacup_Firmware/commit/47f0f0045c92d5b3b84962bb2f9c2c4b02d96379 It's on branch *experimental*, 4 or 5 commits from top.

Wurstnase, 2015-11-23:
> 47f0f00 It's on branch experimental, 4 or 5 commits from top.

Not this one...
I remember a commit "Can rarely happen on AVR" and maybe "often on ARM."

Traumflug, 2015-11-23:
Sounds like this one: https://github.com/Traumflug/Teacup_Firmware/commit/77630167e7983387d7e66142914a6e55fd2f5953

This is on *master*, some 40 commits back, before *arm-ports-start-here*.

[161 more replies]