Traumflug/Teacup_Firmware #190: I2C Display Support

Traumflug, 2015-11-10:
It escaped me for a while, somebody pushed code for I2C display support to the repo recently. Thank you for this!

Now, the unfortunate thing is, judging by the commit messages it apparently doesn't work, yet, still it went straight to branch *experimental*. So I did the following:

- Made a copy of *experimental* and named it *i2cdisplay*.
- Pushed this copy.
- Restored the former branch *experimental*.

With this done, it's a proper topic branch now. For those who fetched/pulled *experimental* recently, please do the following:

- If you made local commits on top of *experimental*, make patches from them with
```bash
git checkout experimental
git format-patch --keep-subject origin/experimental
```
- In any case, remove the local branch *experimental* then:
```bash
git branch -D experimental
```
- Then restore from the remote branch:
```bash
git fetch
git checkout -b origin/experimental
git checkout -b origin/i2cdisplay
```
- In case there were local branches, apply them on top of *i2cdisplay* with ```git am```.

After having this done, branch *i2cdisplay* is exactly what was *experimental* before.

This new branch has a number of, well, messy commits, but also quite a number of good ones. I'll try to sort this before too long. The general implementation strategy looks very good!

... now scratching my head where I could find a device supporting I2C to allow testing ... :-)

RaD, 2015-12-07:
Do you mean my module https://github.com/Traumflug/Teacup_Firmware/blob/ssd1306/i2c_bus.c ?

Traumflug, 2015-12-07:
This module, and some 50 additional commits. It's solved now.

Uhm. I still have no I2C device. Is it correct your display would work on a generic, ATmega based controller as long as it has an I2C connector? Uhm, could you afford to send me one?

RaD, 2015-12-07:
No problem, which kind of SSD display do you want? 5V or 3.3V?
I still have no time to finish with I2C queue, other stuff works pretty well, I have sent a photo in appropriate issue: [image]
Send me your postal address via rpopov/skype.

Traumflug, 2015-12-07:
Wohoow, a pixel based display! Interesting what you managed to squeeze into this small '328.

RaD, 2015-12-07:
It is smart display, check docs on SSD1306 128x32 I2C on Adafruit site.

RaD, 2016-01-21:
As @Traumflug said, the OLED screen is in his home. I hope the queue problem would be solved soon.

Wurstnase, 2016-02-23:
Great work!
[image]

RaD, 2016-02-23:
Still have no time to make right merge my I2C code with teacup queues.

Traumflug, 2016-04-02:
First, thank you for sending me a display. I got it working now to some extent. I2C appears to be a bit more complicated than serial or SPI, one has to stick to a protocol! :-)

Looking at the code I wonder wether this is a proven library fetched from somewhere or wether you wrote this yourself. If it's a library it likely comes with some sample code.

One issue I found is that the TWI interrupt never stops, causing buffers to overflow. Putting a stop there and suddenly a lot more works. Still I'd like to know how to properly stop I2C/TWI, so far I simply turn it off.

Wurstnase, 2016-04-02:
Finally I used u8glib at work. This works but needs more space.

Traumflug, 2016-04-03:
@RaD, I just pushed branch *i2c-rebuild*, so you can see what happens. Basically I picked the first commit from your branch and picked together a test case from your other commits. Then refined the stuff until it worked, all on the first commit. After that I added a number of simplifications. On how to enable the testcase see the top comment in i2c_test.c. Here it successfully writes a message to the screen.

Problem at this point is that only entire blocks can be sent. This makes e.g. clearing the display a wasteful operation, one had to create a buffer with 512 zeros. It also prohibits sending from program memory. As far as I can see this limitation is the problem you got stuck on.

Next planned step is to implement a flexible buffer, like serial uses it already and like you tried on. Not trivial, because I2C wants an explicite transmission start and transmission end. Simply blasting out bytes like with serial or SPI doesn't work. So far I think the best solution is to split the send handler into ...start, ...write and ...end. All this compatible with SPI.

When this works, too, all the other goodies you wrote can join. The implementation for Configtool looks just fine!

BTW., this display is kind of funny. Automatic scrolling supported, but not screen clearing. And it keeps its contents even without supply voltage. Turn the thing off and, hours later, on again and the last screen content is still there :-)

RaD, 2016-04-04:
Please, take a look on https://github.com/halfakop/Teacup_Firmware/commit/6343a22a4dfebaeec782028eb65a989e64fa4317

I have done some unfinished work there.

There is no problem with big block because of you can send many small block. For instance, to clear display it would be enough to send 64x of 8 bytes zero blocks, one by one.

My problem is how to make on C lang flexible buffer that will take fast messages from other firmware and slowly (i2c is not fast) send to display. Now I am busy little bit with other tasks but I hope I will return soon.

Traumflug, 2016-04-04:
> Please, take a look on halfakop@6343a22

Thanks. If I can't find a better solution, which is quite possible, we'll fall back to this. BTW., your block there is on the stack, which will go away before all bytes are sent. It should be declared 'static'.

One thing I couldn't find out so far is, how does the display distinguish between pixel data and commands? Unless I'm mistaken there are no data length fields and also no end-of-data markers. So far my assumption is that the distinction happens on the I2C level: each transmission start starts with a command and a transmission end is also a data end.

RaD, 2016-04-04:
First byte of packet: Co|Dc|0|0|0|0|0|0.

After sending I2C address of a device you may send command or data. If Co=0, then data bytes would be send. If Dc=0, next byte would be command. If Dc=1, next byte would be send to display video memory, pointer there will be incremented. 

Pay attention on data marker: https://github.com/halfakop/Teacup_Firmware/blob/ssd1306/display_ssd1306_i2c.c#L70

RaD, 2016-04-04:
At shortly:
0x00 command there
0x40 data bytes

Traumflug, 2016-04-04:
Yes, but how many data bytes? Once bytes are sent to display video, there's no apparent way to stop this for returning to the command level.

RaD, 2016-04-04:
Until I2С packet ends.

Traumflug, 2016-04-04:
Glad to see you came to the same conclusion :-)

This ringbuffer is now implemented and pushed. No reliable transmission end detection so far, but test code clears the screen just nicely with a simple loop.

RaD, 2016-04-05:
I think that display interface should give users the following function:
https://github.com/halfakop/Teacup_Firmware/blob/ssd1306/display_ssd1306_i2c.h#L39-L47

I mean that firmware doesn't do `print('t=230')` for hotend but just update DISPLAY_T structure and "display driver" itself prints all data in right display positions.

Traumflug, 2016-04-05:
> I think that display interface should give users the following function:
> https://github.com/halfakop/Teacup_Firmware/blob/ssd1306/display_ssd1306_i2c.h#L39-L47

Yes, absolutely. This is a higher level than what I'm currently on. Right now I work on the I2C part. Serving the display should be independent from this, because the same or similar displays can be connected via each of the protocols (I2C, SPI, 4-bit, 8-bit).

Next step will be, I think, a display connection broker. Like
```C
#if defined DISPLAYBUS_I2C
  #define displaybus_write(x) i2c_write(x)
#elif defined DISPLAYBUS_SPI
  #define displaybus_write(x) spi_rw(x)
#elif ...
```
Having this, dependencies of display code on the connection protocol is removed. *display_ssd1306_i2c.c* becomes *display_ssd1306.c* and uses *displaybus_write()* instead of *i2c_write()*. All without any binary size increase or additional runtime calculations.

No such golden solution so far for re-using *sersendf()* for writing to the display. Right now, *sersendf()* writes results straight to the serial line. It's output channels have to be switched somehow. Not sure wether this should be done by a global switching function, which has to be called before writing to a specific channel, or by carrying around the channel as a parameter, similar to fprintf(3).

Traumflug, 2016-04-05:
Good. I2C part done, bus working reliably in all test conditions I could set up. Ringbuffer working fine as well. @RaD, I'd be glad if you could run this code as well, it'd be not the first time code working perfectly here fails in some detail elsewhere.

[image]

BTW., this display is apparently light (opposite of dark) enough to make everything else around it dark on a photograph. I had to photshop lighting in the above picture substantially to get the surrounding visible again.

Controller is my trusty Gen7 1.1, these black/white ringed wires are part of my steampunk (and working!) SD card adaptor.

Traumflug, 2016-04-05:
Just a note: in Adafruit's library they put a 0x00 in front of each single command, even when sending several of them together. [Example](https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.cpp#L335-L342).

RaD, 2016-04-06:
About the note. This is allowable behaviour but unneeded.

Wurstnase, 2016-04-06:
I've pushed my u8g-stuff. Maybe you can use some lines of it. There are 2 buttons implemented. Not very pretty, but better than nothing.
https://github.com/Traumflug/Teacup_Firmware/tree/u8g_with_buttons

Traumflug, 2016-04-06:
> This is allowable behaviour but unneeded.

So you think it's pointless to support any display but yours. I better stay calm now. And thank you for testing the reviewed work.

Traumflug, 2016-04-06:
Let me try to explain why the code from halfakop didn't work. The bug, at least one, is [here](https://github.com/Traumflug/Teacup_Firmware/blob/0e1e6702736d91f7d436d1f121433cd3a9d44d87/i2c_bus.c#L98) This shouldn't be I2C_MODE_SARP, but I2C_MODE_SAWP.

Another bug is plain exhaustion of RAM, at least on the '328. While my test code isn't exactly the suggested one, it's close to it. And this https://github.com/Traumflug/Teacup_Firmware/commit/f32c23d422db8d39ac21f9645cb55a5cd5aba786 shows that all these buffers simply don't fit into these 2 kB RAM the '328 features. Even if it would just fit, there was zero space for stacks (memory allocated inside functions). The same commit also shows the solution: write directly from program memory.

Then I saw the implementation to queue up entire transmissions. Yes, excellent idea ... if there was enough memory to allow this. If one buffer doesn't fit into RAM, 8 of them fit much less.

That said, it's a bit questionable wether such a buffer queue makes sense. Memory has to be copied around, the queue has to be maintained, which costs CPU cycles, binary size, RAM. Sending a byte over I2C costs 9 I2C clocks, or about 360 CPU clocks at maximum I2C speed. This isn't much, one might end up spending more time maintaining the queue than saving elsewhere. That's why I don't plan to pick this part, I hope you understand.

Traumflug, 2016-04-06:
> I've pushed my u8g-stuff.

How much RAM does this cost?
How much Flash does this cost?
What can it do what the current code can't?
Does it work on small AVRs?

Wurstnase, 2016-04-06:
This code runs on an Arduino Nano. But I mainly pushed it for the buttons and the idea how to shift visual pages. When I correctly I'm at ~20k. Maybe a bit more. But I deactivate lookahead and some other things could be optimized out, while I don't use any stepper. (take a look at commit HEAD~2)

Traumflug, 2016-04-18:
I write and write and write ... just pushed a bunch of improvements / cleanups to Configtool on branch *i2c-rebuild*. 28 commits now and still only 50% of @RaD's work picked.

phord, 2016-04-19:
I just dropped a bunch more for you to consider.  :boom:

Traumflug, 2016-04-20:
Just pushed current state. The last few commits aren't the last word, but now the distance to *experimental* is a lot shorter.

Traumflug, 2016-04-20:
Also removed branch *ssd1306*. These two commits are on *experimental* now (and a few other branches).

Traumflug, 2016-04-26:
Another chunk on *experimental*.

- Infrastruture is mostly done, prepared for display bus and display type selection.
- sersendf_P() replaced by sendf_P(), which can write to both, the serial line and the display. A convenience macro sersendf_P() still exists, so no need to change code using it.
- I2C error handling is now successful. I2C turned out to be pretty unreliable when running at the "wrong" bitrate, at least on my test board. At 100'000 bits/s I experience about one transmission loss per second; no such odds when running at 200'000 bits/s. Looks like there's a need to make this user selectable.

Traumflug, 2016-04-28:
D'oh. Display hardware decided to say bye bye shortly before I was done. So I had to put the remaining plan into a large comment in *display_ssd1306.c*, see there. Latest achievements:

- I2C is now reliable. I misunderstood one of the bitfields, code was actually wrong.
- Display should now show a greeting message and X/Y/Z coordinates updating once a second.

Remaining parts shouldn't be too difficult to do. The queueing mechanism needs another tweak (see this comment), everything else is a matter of arranging things in a neat fashion.

All code on *experimental*, branch *i2c-rebuild* removed. I plan to remove the other related branches (*i2cdisplay-review*, *i2cdisplay*), too, after searching them for nuggets I missed so far.

I'm out, somebody please take over!

Traumflug, 2016-05-30:
Big spring cleaning today, branch *i2c-display* removed as well, after verifying that all its nuggets are on *master*.

Wurstnase, 2016-08-12:
Just tried this with the Adafruit Display. But can't get this to work. I changed the address to 0x3C and copied the u8g working init_sequence but unfortunately I don't see anything.

Any hints, how I could debug this?

Traumflug, 2016-08-12:
I2C is a tricky thing. Other than SPI it's an outright protocol.

Regarding the address, it's not an 8-bit address, but a 7-bit one. The last bit is a read/write flag. To make this a byte, one has to shift the address one to the left. 0x3C << 1 = 0x78. See http://www.i2c-bus.org/addressing/

If this still doesn't work it's time for some printf() style debugging to see wether bytes are sent at all. Wether the interrupt happens, with which status it happens, such stuff. During writing that code I used a few serial_writechar() with different characters to see what happens. I2C bus speed can be reduced to not overwhelm the serial line with debugging characters.

Wurstnase, 2016-08-12:
Oh dear!
I changed the address first and thought this was correct. Back to 0x78 and taking the new init sequence and everything works as expected. Thank @Traumflug

Traumflug, 2016-08-12:
This was quick :-) Excellent!

Wurstnase, 2016-08-16:
Or not :/
Friday anything looks perfect. Monday nothing works...
How should the debug should looks like?
This is what I get:
`.1.3.4.4.4.4.1.3.4.4.4.4.4.1.3.4.4.4.1.3.4.4.4.4.`

Wurstnase, 2016-08-16:
Some progress:
I flash the Nano with the adafruit example. (GFX etc.)
Display shows the flash screen etc.
Now I flash Teacup and it works.
I have connected the reset pin and I guess it's something with this?!

Wurstnase, 2016-08-16:
Sorry for the noise. It is the reset pin.

Flashed adafruit example. It works. 
Flashed Teacup. It works.
Unpower Nano
Power Nano
It doesn't work.

Wurstnase, 2016-08-16:
Really, really close to find the issue.
The problem could be the `displaybus_write(0x40, 0);` for the init sequence in display_tick().
Currently I'm trying some extra code. I have a second init_sequence but when I try to go through this I only get zeros. Any idea what I'm doing wrong?

```
static const uint8_t PROGMEM init_sequence_2[] = {
  0x00,
  0x21, /* column addr (0 =  reset) */
  0x00,
  0x7F, /* column end address (127 = reset) */
  0x22, /* page addr (0 0 reset) */
  0x00,
  0x03  /* page end addr*/
};
```

and later:

```
for (i = 0; i < sizeof(init_sequence_2); i++) {
  sendf_P(serial_writechar, PSTR("%su\n"), (uint8_t)init_sequence_2[i]);
}
```

Any init_sequence_2[i] is zero.

Traumflug, 2016-08-16:
You read out RAM, not Flash.
```
  sendf_P(serial_writechar, PSTR("%su\n"), (uint8_t)init_sequence_2[i]);
```
==>
```
  sendf_P(serial_writechar, PSTR("%su\n"), (uint8_t)pgm_read_byte(&init_sequence_2[i]));
```

phord, 2016-08-16:
Good catch, @Traumflug !

Also, you can drop the cast from pgm_read_byte since it already returns uint8_t.

    sendf_P(serial_writechar, PSTR("%su\n"), pgm_read_byte(&init_sequence_2[i]));

Wurstnase, 2016-08-16:
Thanks, but this code works?
https://github.com/Traumflug/Teacup_Firmware/blob/master/display_ssd1306.c#L111
`displaybus_write(init_sequence[i], (i == sizeof(init_sequence) - 1));`

Wurstnase, 2016-08-16:
YES! pgm_read_byte solves everything!

Traumflug, 2016-08-16:
> Thanks, but this code works?

Looks like you catched a bug over there. My fault.

Wurstnase, 2016-08-16:
https://github.com/Traumflug/Teacup_Firmware/tree/fix_ssd1306

waits for a rebase to experimental :+1: 
I will push tomorrow the init code for the Adafruit display. It's not much anymore what that display needs.

Wurstnase, 2016-08-17:
Added support for Adafruit SSD1306 and added new font 8x8.

Traumflug, 2016-08-17:
Nice!

TBH, I couldn't resist to look into additional fonts, too, and found a whole lot of them on retro-gaming sites like [this](https://damieng.com/blog/2011/02/20/typography-in-8-bits-system-fonts) or [this](http://kofler.dot.at/c64/font_01.html) (650+ 8-bit fonts!). Even a converter for creating C files from pictures exists: [Bitmap Converter](http://en.radzio.dxp.pl/bitmap_converter/).

Anyways. One thing I'm a bit unsure about is, how should we deal with Configtool? It's pretty obvious that people like you and me and @phord happily hack away on the firmware, but have or see little gain in adjusting Configtool, too. We put our extra #define into config.h somewhere and are done. On the other end of the spectrum we have users which barely manage to start a Python application. Those won't make Configtool more complete either. Configtool is a great thing, but it tends to fall behind.

Question: should we insist on new code to have an entry in Configtool before moving it to *experimental*/*master*? Or is it fine to have such technology inaccessible to generic users?

Wurstnase, 2016-08-17:
[Bitmap Converters](http://www.zilogic.com/blog/tutorial-glcd-bitmaps.html) are everywhere :)

I'm not very familiar with configtool. Last adaption costs me a lot of hours for just one additional drop down. The minimum requirement should be a TODO in the commits.

Wurstnase, 2016-08-17:
How does the SSD1306 change its pixel in vertical or horizontal mode?
```
---->
---->
---->
---->

---->
<----
---->
<----

^ ^ ^ ^
| | | |
| | | |
| | | |
```

phord, 2016-08-17:
> One thing I'm a bit unsure about is, how should we deal with Configtool? It's pretty obvious that people like you and me and @phord happily hack away on the firmware, but have or see little gain in [...]

I think it's a good policy to require developers to update ConfigTool to match their code changes. But in practice I think it's ok to make that a complaining policy.  Such as, _"Thanks for adding this cool new feature.  Please add it to configtool so we can accept it into the mainline."_   And then later, _"Ok, I understand ConfigTool is obtuse and you don't speak python.  I've added it for you here.  Please review and tell me if I misunderstood."_

But what I would really prefer is for ConfigTool to learn its options from the header files.  It already learns the help text there.  I agree it would be overkill to reinvent a GUI description language inside comments in the config headers.  Some things would never make sense, such as the adjustPinVisibility function, and we would always tweak them in python.  But the DISPLAY_TYPE options seem trivial to "learn" from board.generic.h.  It might help us remember to keep the Doxygen current, too, if Doxygen-blocks was where it learned it from.

Traumflug, 2016-08-17:
> But what I would really prefer is for ConfigTool to learn its options from the header files. It already learns the help text there. I agree it would be overkill to reinvent a GUI description language inside comments in the config headers.

I hope the idea of Configtool is more than just adding some GUI-sugar on top of headers. A tool with just 1:1 relationships could be done easier. Configtool can do much more than just flipping switches, it also has the power to do things graphically (think of endstop locations, build room size defined by dragging them in a drawing), to do calculations (think of steps/mm calculator, thermistor table creation) or to give step by step instructions during printer setup. IMHO, the number of checkboxes and choices would not increase, but reduce over time :-)

Is creating a one-click printer setup a solvable problem?

What I can do is to look up a few commits which add a choice or a checkbox and tag them. Configtool code is pretty repeating, so procedures can be repeated at another location for another #define.

Other than that I agree, making inclusion into Configtool mandatory is asking a lot. Probably too much. Having a Github issue here makes sure such things don't get forgotten.

Traumflug, 2016-08-17:
> Bitmap Converters are everywhere :)

D'oh. One simply has to save in XBM format. That's indeed easy :-)

Wurstnase, 2016-08-22:
While trying to make a bmp-parser I test my code.
Unfortunately I've got not that what I've expected.

This code in display-tick (of cause display_low_code is extended with low_code_logo)
```
      case low_code_logo:
        // Set horizontal adressing mode.
        displaybus_write(0x00, 0);
        displaybus_write(0x20, 0);
        displaybus_write(0x00, 1);

        // Write 512 zeros.
        displaybus_write(0x40, 0);
        for (i = 0; i < 512; i++) {
          displaybus_write(pgm_read_byte(&logo2[i]), (i == 511));
          if (pgm_read_byte(&logo2[i]) != 0xFF)
            debug++;
          if (i == 511)
            sersendf_P(PSTR("i = %su\ndebug: %u\n"), i, debug);
        }

        displaybus_write(0x00, 0);
        displaybus_write(0x20, 0);
        displaybus_write(0x02, 1);
        break;
```
prduces 384 non-0xFF-values while the complete array is 0xFF:
```
static const uint8_t PROGMEM logo2[] = {
0xFF, 0xFF...
};
```

What's going wrong here?

phord, 2016-08-22:
How big is logo2's initializer list?
What is the index of the FIRST non-0xFF byte?

I assume you checked that you initialized 'debug=0' somewhere, yes?

I'm not familiar with "%su", but I do see it in sendf.c.  Still, I would also be suspicious of it unless I used it a lot already.

Wurstnase, 2016-08-23:
logo2 has 512x 0xFF.
debug is set to 0. With my first test I print any byte where I saw this issue. I thought this could be the issue I changed the code to count only non-0xFF-values.

I changed a little bit the debug code and print every i again.
with PROGMEM:
```
0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xA6 0x2E 0xAF 0x01 0x01 0x01 0x01 
0x01 0x01 0x01 0x01 0x01 0x01 0x02 0x02 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 
0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 
0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 
0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x04 0x02 0x01 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x03 0x02 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x02 0x01 0x01 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x01 0x01 0x01 0x01 0x01 0x00 0x00 0x01 0x07 0x01 0x02 0x00 0x01 0x00 0x00 0x01 0x01 0x01 0x00 0x00 0x00 0x01 
0x01 0x01 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 
0x00 0x00 0x00 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x06 0x01 0x01 0x01 0x02 0x01 0x03 0x03 0x02 0x02 0x01 0x01 0x01 0x03 0x02 0x00 0x00 0x00 0x05 0x01 0x01 0x02 0x01 0x01 0x04 0x02 0x02 0x03 
0x00 0x03 0x01 0x00 0x01 0x01 0x00 0x00 0x03 0x01 0x01 0x01 0x02 0x03 0x02 0x03 0x03 0x00 0x05 0x02 0x02 0x01 0x07 0x03 0x00 0x01 0x00 0x06 0x00 0x00 0x04 0x01 
0x01 0x02 0x01 0x02 0x07 0x02 0x00 0x01 0x02 0x02 0x01 0x04 0x03 0x05 0x00 0x00 0x00 0x00 0x06 0x03 0x02 0x02 0x02 0x03 0x05 0x02 0x01 0x01 0x00 0x00 0x00 0x01 
0x01 0x02 0x02 0x01 0x02 0x03 0x01 0x01 0x01 0x02 0x03 0x05 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 

i = 255
debug: 384

```

without PROGMEM:
```
0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0x47 0xE2 0x74 0x07 0x41 0xE0 0x84 0x07 0x91 
0x05 0xA8 0xF4 0x20 0x51 0x37 0x42 0x30 0x93 0x89 0x00 0x20 0x93 0x88 0x00 0xDC 0x01 0xCB 0x01 0x80 0x5F 0x98 0x4D 0xAF 0x4F 0xBF 0x4F 0x80 0x93 0xEB 0x04 0x90 
0x93 0xEC 0x04 0xA0 0x93 0xED 0x04 0xB0 0x93 0xEE 0x04 0x04 0xC0 0x30 0x93 0x89 0x00 0x20 0x93 0x88 0x00 0x80 0x91 0x6F 0x00 0x82 0x60 0x80 0x93 0x6F 0x00 0x80 
0xE0 0x08 0x95 0x08 0x95 0x10 0x92 0x6F 0x00 0x08 0x95 0xAA 0x1B 0xBB 0x1B 0x51 0xE1 0x07 0xC0 0xAA 0x1F 0xBB 0x1F 0xA6 0x17 0xB7 0x07 0x10 0xF0 0xA6 0x1B 0xB7 
0x0B 0x88 0x1F 0x99 0x1F 0x5A 0x95 0xA9 0xF7 0x80 0x95 0x90 0x95 0xBC 0x01 0xCD 0x01 0x08 0x95 0xA1 0xE2 0x1A 0x2E 0xAA 0x1B 0xBB 0x1B 0xFD 0x01 0x0D 0xC0 0xAA 
0x1F 0xBB 0x1F 0xEE 0x1F 0xFF 0x1F 0xA2 0x17 0xB3 0x07 0xE4 0x07 0xF5 0x07 0x20 0xF0 0xA2 0x1B 0xB3 0x0B 0xE4 0x0B 0xF5 0x0B 0x66 0x1F 0x77 0x1F 0x88 0x1F 0x99 
0x1F 0x1A 0x94 0x69 0xF7 0x60 0x95 0x70 0x95 0x80 0x95 0x90 0x95 0x9B 0x01 0xAC 0x01 0xBD 0x01 0xCF 0x01 0x08 0x95 0x05 0x2E 0x97 0xFB 0x1E 0xF4 0x00 0x94 0x0E 
0x94 0x37 0x1C 0x57 0xFD 0x07 0xD0 0x0E 0x94 0xFE 0x1B 0x07 0xFC 0x03 0xD0 0x4E 0xF4 0x0C 0x94 0x37 0x1C 0x50 0x95 0x40 0x95 0x30 0x95 0x21 0x95 0x3F 0x4F 0x4F 
0x4F 0x5F 0x4F 0x08 0x95 0x90 0x95 0x80 0x95 0x70 0x95 0x61 0x95 0x7F 0x4F 0x8F 0x4F 0x9F 0x4F 0x08 0x95 0xEE 0x0F 0xFF 0x1F 0x05 0x90 0xF4 0x91 0xE0 0x2D 0x09 
0x94 0xA2 0x9F 0xB0 0x01 0xB3 0x9F 0xC0 0x01 0xA3 0x9F 0x70 0x0D 0x81 0x1D 0x11 0x24 0x91 0x1D 0xB2 0x9F 0x70 0x0D 0x81 0x1D 0x11 0x24 0x91 0x1D 0x08 0x95 0x0E 
0x94 0x45 0x1C 0xA5 0x9F 0x90 0x0D 0xB4 0x9F 0x90 0x0D 0xA4 0x9F 0x80 0x0D 0x91 0x1D 0x11 0x24 0x08 0x95 0xB7 0xFF 0x0C 0x94 0x54 0x1C 0x0E 0x94 0x54 0x1C 0x82 
0x1B 0x93 0x0B 0x08 0x95 0x2F 0x92 0x3F 0x92 0x4F 0x92 0x5F 0x92 0x6F 0x92 0x7F 0x92 0x8F 0x92 0x9F 0x92 0xAF 0x92 0xBF 0x92 0xCF 0x92 0xDF 0x92 0xEF 0x92 0xFF 
0x92 0x0F 0x93 0x1F 0x93 0xCF 0x93 0xDF 0x93 0xCD 0xB7 0xDE 0xB7 0xCA 0x1B 0xDB 0x0B 0x0F 0xB6 0xF8 0x94 0xDE 0xBF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 

i = 255
debug: 380
```

triffid, 2016-08-23:
perhaps try with &(logo2[i]) in case the compiler is interpreting it as
(&logo2)[i] - no type checking since pgm_read_ family is macros which
include cast to uint16 or uint32 depending on target chip.

Also, in the projects where I use PROGMEM it goes after the variable name
rather than before, ie const uint8_t logo2[] PROGMEM = { ... }

not sure if either of these is your problem, but I use PROGMEM all over the
place without issues

[25 more replies]