Traumflug/Teacup_Firmware #192: Implementation of M80 & M81 commands

RaD, 2015-12-08:
I guess it would fine to use PSU MOSFET Pin to control PSU via M80 & M81 commands.
Check m80m81 branch. What dou you think, fellows?

Direct link to the commit:
https://github.com/Traumflug/Teacup_Firmware/commit/ab85bef7bd905023b8fda05f017d997bcd33c370

Traumflug, 2015-12-08:
Hmm. Somehow it's apparently difficult to get the word out that M80 / M81 is simply obsolete. Firmware handles the PSU on its own, without any G-code. So the question is: what's the purpose of this pair?

That aside, as given in the commit, the PSU will still turn off on its own after a while without activity. Variable *psu_timeout* needs different handling to keep it on permanently.

Second thing is, there's indeed a use case where automatic power off can be harmful: printers with a belt driven Z axis. These need to keep power on to make the print head resist gravity after the print is done. A solution for these cases would be better. Like automatic ON, but OFF only on demand?

Immediate power off is handled by M0/M2 already, in alignment with LinuxCNC's implementation and the NIST standard.

RaD, 2015-12-09:
I do not familiar with CNC but accoring to Marlin Firmware
<https://github.com/MarlinFirmware/MarlinDev/blob/dev/Marlin_main.cpp#L127-L128>
the support M80/81 is continued. There is a farm of printers that should be
controlled remotely, they use Raspberry PI+NanoHeart/RAMPS hardware.
Firmware's user may assign pin to PSU POSFET Pin and control printers' PSU
via Gcode. The Z-drop-down printer users should just not use this option.
It seems all will be OK.

About the psu_timeout variable. No need exists to keep printer on between
tasks. When its task will finished the printer would be power down until
next task. Power down just turns off the power of NanoHeart/RAMPS part.

Also, I think I should wrap the following code with #ifdef PS_MOSFET_PIN:
https://github.com/halfakop/Teacup_Firmware/commit/ab85bef7bd905023b8fda05f017d997bcd33c370

Traumflug, 2015-12-09:
Marlin. Pfft. :-)

> There is a farm of printers that should be controlled remotely, they use Raspberry PI+NanoHeart/RAMPS hardware.

Teacup will handle this perfectly fine, without M80/81.

PS_MOSFET_PIN is really only for 2-stage PSUs. Those which turn on logic voltage for stepper drivers and motor voltage separately, with a delay in between. The only known controller doing so is the Sanguish with these TB6560 drivers. All others can use PS_ON_PIN.

RaD, 2015-12-09:
OK, in this case, can you show me an example of g-code that turns on the
PSU, make some moves and turns off the printer?

Traumflug, 2015-12-09:
G1 X100 F500

That's all. PSU will turn on, move the stepper and turn the PSU off a few seconds later.

RaD, 2015-12-09:
Ok, I guess you should talk about thie feature more :)
Will remove my m80m81 branch.

phord, 2015-12-09:
PS_ON_PIN is already managed well, so M80/M81 aren't needed usually.  But I sometimes would like manual control over the PS.  But it's a silly reason.

For some reason, my thermistors are inaccurate at low temperatures unless the PS is ON.  When they're very low (0C?) they report being very hot; I think this is to prevent continuous heating when the thermistor goes bad.

Here the printer is idle, the room temperature is actually about 18C, but my thermistors report crazy values.  When I move a motor, the PS comes on and the thermistors begin reporting sane values.  A few seconds of idle time later and the thermistors go crazy again.  
[image]

Because of this I cannot even turn the heat on for my printer if the motors are not moving.  When I set the BED temp to 100C, for example, it is not high enough to exceed the "bad" thermistor reading.  But if I set it to a high enough value, say 9999C, then it begins heating.  Of course, then the PS turns ON, too, and my thermistors begin reporting correct values.  So at least it doesn't go idle again and turn off the PS.

So, M80/M81 might be handy to override this behavior.  I think that would be kind of silly; but at the moment, it's not even an option.  :-)

A better solution, of course, is to fix my thermistors to run correctly on the 5v supply, I guess.

Traumflug, 2015-12-09:
M119 (endstop check) turns on the PSU, too.

Traumflug, 2015-12-09:
That said, it might be a worthwhile idea to make the heaters_all_zero(), used in clock.c, a temperatures_all_zero(), so the PSU is kept on as long as a heater is active. Cooling phases, where the heater is still active, but not actually heating for a longer time, are rare.

Traumflug, 2015-12-09:
Uhm. I forgot that not all heaters have a temperature sensor, so we'd need both.

RaD, 2015-12-10:
@phord, what hardware do you use? Your case seems the thermistors have pull-ups on PSU, not digital 5v supply.

phord, 2015-12-10:
I use Gen7 electronics. I'm pretty sure the pullups are on 5v, but I'm not
sure why they are affected by the PSU. For example, they continue to read
correctly when they are hot after the PSU turns off. I think I investigated
it once, but it's a minor inconvenience so I haven't revisited it.