Traumflug/Teacup_Firmware #193: LCD - CNC SHIELD

CoderSquirrel, 2015-12-09:
I saw there's branchs for lcd implementation but most of them are about reprap graphic card or i2c displays, i want to use a 16x2 display, i usualy implement it with liquid crystal at arduino, anyone had tried it?

Traumflug, 2015-12-09:
Bringing in Arduino libraries will also bring in a lot of bloat and incompatibilities. Just think of support for various ARM chips.

As far as I can see, pretty much everything needed to run a display is there, just unsorted and not yet reviewed. Feel free to do yet another 'best pick'. I hope I can chime in before too long.

The version entering *experimental*/*master* will most likely have this:

- One abstract 'class' ( = code separated out into a distinct file) for each transport medium, 4-bit, I2C, SPI. The slower ones with a queue similar to the one in the serial code to allow continueing with important stuff while not all bytes are sent to the display, yet.

- Display initialisation for each type of display (of course).

- Display sending code with adaption to display size (2 line vs. 4 line vs. pixel).

- A mechanism to send bytes out to the display at idle time.

- Most important for generic users: the fields needed to declare a display in Configtool. Even this is partially done already, thanks to @RaD.

As said, it's all there already, especially on branch *i2cdisplay*. If you want to do a review, like trying and confirming that it works in your setup, too, please go ahead. You can have write privileges to the repo any time, just say so. With two people confirming each other I'm positive to move it to the main branches.

CoderSquirrel, 2015-12-09:
Hi @Traumflug thank you very much for your quick answer, i download the lcd3 branch i'm making it work at my 16x2 lcd, and i'm willing to make a menu on it. I really can use and lcd with i2c or an newer lcd thats why i'm not going to look at i2cdisplay. 
The problema that i'm really having now is i'm trying to use the scl and sda port to recieve analog input form an joystick but it's not giving the values between the min and the max, just the min and the max.
Do i need to declare or read the analog input in a different way?

Traumflug, 2015-12-09:
I see. You want not only a display, but also a joystick.

Analog stuff is handled in analog.c. Usually used for temperature sensors. This occupies the Analog-Digital-Converter (ADC, there is only one) all the time, so you have to align your code with this. Something like adding the joystick pin to the other analog pins there. Readouts and storage for later use will happen automatically, then, so you always have the latest values.

CoderSquirrel, 2015-12-09:
Yeah, my first though was add it there, But once i'm using  SCL and SDA i guess it's already set up, isn't?

Traumflug, 2015-12-09:
Using analog stuff outside of analog.c causes a conflict. Two chunks of code messing with the same registers/hardware. Can't work.

CoderSquirrel, 2015-12-09:
Thanks @Traumflug  i'm not good at low level programming so i'm kinda stumbling a it.
So, i looked at the alanog.c and saw 
`AIO0_DDR &= ~analog_mask;
#ifdef	AIO8_DDR
AIO8_DDR &= ~(analog_mask >> 8);
#endif`
so the pins i want to use are, as i said, SCL and SDA, A4 and A5, as it are pins A as A8, i can declare them mask as that way, above?

Traumflug, 2015-12-09:
This #ifdef distinguishes between ATmegas with 8 and ones with 16 ADC channels.

Adding another pin is likely best done where the DEFINE_TEMP_SENSOR macro is used. On how this works, see https://github.com/Traumflug/Teacup_Firmware/issues/189#issuecomment-162013342

Traumflug, 2016-05-30:
Just in case you're still interested, code for HD44780-type displays was just pushed to *experimental*. It's official part of Teacup now. Configuration stuff can be found in Configtool -> board panel -> display panel. Should work with any controller providing the 7 pins required.