Traumflug/Teacup_Firmware #194: Ignored gcode lines

JoseAlcerreca, 2015-12-12:
Hi,

I'm using Teacup (tried master and experimental) with a custom board and an atmega328p on my CNC machine. I was using RAMPS1.4 and Sprinter before and the same gcode worked.

If I execute the following program some lines are ignored

G92 X200Y0Z0
G90
G1 Y70 F2000
G1 X150.000 F8000 (ignored)
G1 X-10.000 Y108.000 F8000 (ignored)
G1 Z26.000 F3000
G1 Z0

But if I execute the lines manually one by one, using a terminal or from OctoPrint, everything works as expected.

This is the debug output: 
http://pastebin.com/bi0cJQ4N
and board config
http://pastebin.com/8CFpCBEb

Another example:
G92 X0Y0Z0
G90
G1 Y40.000 F3000
G1 X10
G1 Y0 Z14.500 F3000
G1 Y0 Z27.000 F400
G1 Y0 Z14.500 F400
G4 P1000
G1 Y70 F2000 (ignored)
G1 X-260.000 Y108.000 F8000 (ignored)
G1 Z26.000 F3000
G1 Z0

Things I tried
 - LOOKAHEAD off and on
 - Changing the stepper drivers
 - Defining X, Y and Z on every line
 - Sending program from minicom and octoprint
 - experimental branch

Buffering issue? Some kind of limit protection? Am I missing something obvious?

Thanks!

Traumflug, 2015-12-12:
Interesting problem. It it works on a RAMPS it might be a RAM overflow. The '328 has much less RAM than a '1287. Reducing MOVEBUFFER_SIZE to 4 will free up some RAM.

RAM usage for static variables is reported after compilation, but there's also some space for the stack needed. Local variables go to the stack, so its size depends on program flow.

If this doesn't help it gets even more interesting. :-)

JoseAlcerreca, 2015-12-12:
On a RAMPS with Sprinter, not Teacup :), Arduino IDE shows plenty of free space after compilation. 

Would an overflow show errors in the console?

Would the simulator help?

Traumflug, 2015-12-12:
> Would an overflow show errors in the console?

A stack growing into the area of global variables can do anything, from barely noticeable to immediate reset. This is not to be confused with program memory (Flash) size. It's fine to fill Flash to 99%, there is no stack. RAM should be 80% or less.

Also, compiling with Configtool will generate more efficient code, even when using Arduino IDE's compiler.

JoseAlcerreca, 2015-12-12:
Tried reducing MOVEBUFFER_SIZE to 4 but still happening.

I'm doing a lot of tests, for example, line G1 X-260.000 F8000 is ignored here:

G92 X0Y0Z0
G90
G1 Y40.000 F3000
G1 X10
G1 Y0 Z14.500 F3000
G1 Y0 Z27.000 F400
G1 Y0 Z14.500 F400
G4 P1000
G1 Y70 F2000
G1 X-260.000 F8000
G1 Z26.000 F3000
G1 Z0

If I move it

G92 X0Y0Z0
G90
G1 Y40.000 F3000
G1 X10
G1 Y0 Z14.500 F3000
G1 Y0 Z27.000 F400
G1 X-260.000 F8000
G1 Y0 Z14.500 F400
G4 P1000
G1 Y70 F2000
G1 Z26.000 F3000
G1 Z0

The same X-260 line if ignored. If I move it to the beginning

G92 X0Y0Z0
G90
G1 X-260.000 F8000
G1 Y40.000 F3000
G1 X10
G1 Y0 Z14.500 F3000
G1 Y0 Z27.000 F400
G1 Y0 Z14.500 F400
G4 P1000
G1 Y70 F2000
G1 Z26.000 F3000
G1 Z0

It is executed! but not some of the lines after it. I feel like I'm taking crazy pills, I don't think how this could be hardware related. I also tried ACCELERATION_TEMPORAL, defining all the MIN and MAX values, and some other things, same behaviour. By the way I can reproduce 100% of the times, it fails very reliably :)

Can't program from configtool, the programmer is not in sync, not sure why. 

Maybe my Arduino Nano has a defect, but I doubt that is the problem since it's so deterministic. Feel free to close if you can't think of anything else.

Traumflug, 2015-12-12:
There are machine limits in the printer config. X_MIN, X_MAX, etc. These are used as soft endstops, so you can't move outside this envelope as long as they're defined.