Traumflug/Teacup_Firmware #195: rc servo control

ekaggrat, 2015-12-23:
is it possible to add a rc servo using the current arduino servo library? If a new mcode is added to turn the servo like in other firmwares would that work? or is there some other way of doing it?

Traumflug, 2015-12-23:
You can add a RC servo, but not really a part of the Arduino library. Teacup is entirely independent from this library.

AFAIK, these servos can be handled like a PWM heater, so the code section to look into would be heater_init() and heater_set() in heater-avr.c/heater-arm.c. You can define a heater without temperature sensor, then operate this heater with M106. The art would be to adjust the PWM timing via timer registers to fit the RC protocol. Source code of the Arduino library might give some hints on how to do this, but there are also independent RC code snippets: 

http://winavr.scienceprog.com/example-avr-projects/servo-motor-control-using-avr.html
http://www.electroons.com/electroons/servo_control.html
https://www.mikrocontroller.net/articles/Modellbauservo_Ansteuerung

A quick google found more code samples, but those using a delay won't work reliably, because the processor has many other things to do when operating a printer.

triffid, 2015-12-23:
Servos want a pulse with width of 750-2250us every ~20ms or so. The time
between pulses isn't critical but the width defines the target position. It
should be fairly easy to configure heater PWM to generate suitable pulses

ekaggrat, 2015-12-23:
I thought of that before and tried using the heater output pin to run the servo but it didnt work obiviously the timming was off. I will give it another try by changing the timing. Thankz