Traumflug/Teacup_Firmware #196: Code no longer compiles on newer Arduino IDE without help

phord, 2016-02-05:
From http://forums.reprap.org/read.php?147,33082,620340#msg-620340

We shouldn't have to delete the project directories to make the IDE happy.  We can change the code layout instead to isolate the Arduino code in its own subfolder.  This means all the IDE users will now need to launch the project from a new location.  And it will also make rebasing more complicated.  But c'est la vie.

Traumflug, 2016-02-05:
TBH, I'm a bit fed up with this Arduino mess. Again they tripped us up and worse, if using Arduino gets used, all the great Configtool is pointless. Then, people recognize Teacup as complicated to set up, totally escaping that it's actually easier than any other firmware, that calibration can be done from there, that helpers exist (I'd expect more helpers to appear, like ones to print test cubes).

This one user refuses to use Python. Can't imagine what weird politics forbids to use the about most popular scripting language.

Currently I tend more towards putting a file *arduinotrap.c* into *attic/* with about this contents:
```c
/*
  As the filename says, this is a trap. It works as long as the file exists, because
  Arduino IDE takes an recursive compile-everything apporach, so it tries to compile
  this file.

  For any other purposes, simply ignore this file.
*/

#error You're using Arduino IDE. Arduino compiles into bigger, less performant \
firmwares, makes configuration a rather complicated matter and misses all the \
goodies of Configtool. To build Teacup, just use 'python configtool.py' instead. \
If you still insist on using Arduino IDE, remove folders attic/, simulator/, \
extruder/, research/, and try again.
```
What do you think?

phord, 2016-02-05:
I understand your frustration and I've been bitten by Arduino's randomness, too.  But I don't think it's so much trouble to accommodate the Newbies who are uncomfortable with the command-line, gcc, "make-dot-exe" or the avr toolchain.  Some of these folks are very good with mechanics or electronics and it would be useful to see what they can do with Teacup at their side.

I'm actually impressed that Teacup is buildable from within the IDE.  I know it doesn't take much to set that up, but it is a bit mysterious because the IDE is stupid.  Once you have it working, it's relatively easy to continue supporting it.  Except for the occasional IDE brain-fart like this one, there is almost no work required to continue supporting it.  It's true it doesn't play nicely with other platforms' code.  But I think we can address those along the way.

Maybe it does make things a bit too ugly, though.  I mean... have a look at my commit on branch issue-196. d6b762ffeade4ff54280b61ed53aac9d486324c5  I found for example that my Arduino IDE 1.0.5 fails on any source file with a dash in the filename.  It fails to copy them to the "build" directory but it does try to compile them anyway.  wtf?  :8ball: So I could be be convinced to tell the IDE users to sod-off.

---

As for the attic, I'm not a fan of that in any case.  When I first found Teacup I spent some time trying to understand the extruder/ code before i realized it was obsolete noise that someone forgot to delete or something.  My own philosophy is that the "attic" code -- code that you put on a shelf for possible use or reference later -- is noise and should be left in-place in some branch named "shelf/accel_clock", "shelf/eeconfig", etc.  This way the code is kept in its original context and, if someone experiments with it, git will help them merge it with other branches using its knowledge of their different histories.  As it is now, one must guess about how to properly merge those old changes, what was different about them from the mainline, and what sequence of commits led to them being in the state they're in.  

My main objection to the attic, though, is that when I'm digging in the code I tend to use grep or similar features in my editor, and I am often led astray by duplicated or obsolete code which I then have to remember to ignore.

Traumflug, 2016-02-05:
I totally agree that Arduino IDE is kind of a de-facto standard and that users try this first. Actually it was the recommended way for non-developers to build Teacup until Configtool came along, so it did work.

Not using Arduino IDE doesn't mean one has to go to the command line. configtool.py is/should be double-clickable, could even come with a desktop icon. Configtool also builds for Gen7-AVR and Sanguinololus without requiring to install the Gen7 Arduino IDE Support package (or the older Sanguino extensions) into Arduino. It can use the compiler coming with the Arduino IDE, so no additional installations required.

Sources in *extruder/* aren't obsolete noise, they're the firmware for the Gen3 extruder board; a board seperate from the Gen3 main controller, running its own ATmega. As soon as I attempted to remove it, somebody came along and asked how to use it: https://github.com/Traumflug/Teacup_Firmware/issues/179 :-)

Stuff in *attic/* is an effort to reduce the number of old branches. Each branch shows up in the list of branches here on Github, so too many branches confuse people, people might be tempted to try such a branch, not recognizing how old it is.

Regarding solutions.

Gen3 extruder board support could be removed, along with the #defines needed for it (MAINBOARD, EXTRUDER, maybe more).

A simple way to get rid of grep results in *attic/* could be to simply tar/gzip these files. This makes them a bit less accessible, still they're there for the interested developer.

Can't think of a golden solution for Arduino IDE, though. Renaming dashes is one thing. We had to move all sources into a seperate directory to avoid compilation of sources in subdirectories, along with fixing all other build methods for this move. Arduino doesn't work for most ARM based controllers. Worst of all, people are mislead into thinking they have to edit files for configuration/calibration.

IMHO, ideally we'd place a command into the .ino file to launch Configtool :-) Not aware of such functionality, though.

phord, 2016-02-05:
I wasn't aware ConfigTool is also a make/build utility.  It has become its own IDE!  :-D

I do remember that about the extruder, so it needs to hang around.  But it won't build in the Arduino IDE as-is if directory recursing is turned on.  Blast.

I understand about too many old branches.  GitHub's interface really is a problem in many ways.  Another option is to keep one "attic" branch where the "attic" files are stored.  That doesn't solve the history problem, but it could keep the branch noise to a minimum.  

Yet another option is a workflow like this:
:ballot_box_with_check: Try some idea (accel_temporal)
:ballot_box_with_check: work it into existence
:ballot_box_with_check: decide that it's not going to work after all
:ballot_box_with_check: remove it again in a commit whose message says "This was a test, but it didn't work out in the end because {reasons}"
:ballot_box_with_check: "merge" that into experimental without  actually causing any changes to appear in experimental, but preserving history nonetheless
:white_check_mark:  remove the branch.  

Want find the work on ACCEL_TEMPORAL?     `git log --all -SACCEL_TEMPORAL`

Traumflug, 2016-02-06:
> I wasn't aware ConfigTool is also a make/build utility. It has become its own IDE! :-D

Perhaps you should give it a try to get a better opinion on wether it makes sense to give Arduino compatibility high priority. Instructions start here: http://reprap.org/wiki/Teacup_Firmware#Prerequisites

Traumflug, 2016-06-12:
I've tar'd up the stuff in *attic/*, which should help a bit on this topic.

mattgilbertnet, 2016-11-18:
FWIW, newer versions of Arduino handle the directories fine now. The only change that's needed for Teacup to load and compile in Arduino IDE is to change the file extension of TeacupFirmware.pde to .ino.

Wurstnase, 2016-11-18:
Yes, that's right. I made, a maybe hidden, hint into the wiki:
> If you're using Arduino IDE 1.6.x **to 1.6.9**

Since 1.6.10 it's working again.

phord, 2016-11-19:
I use a TeacupFirmware.ino symlink to TeacupFirmware.pde for testing with
the IDE.

Traumflug, 2016-11-19:
Fine :-) So I did this:

 - Moved branch *issue-196* to the attic, as it's no longer needed. These two commits were well done, still I think that's too intrusive.
 - Renamed *Teacup_Firmware.pde* to *Teacup_Firmware.ino*.
 - Added a hint to Configtool into this file.

Ready to close this issue?

Wurstnase, 2016-11-19:
Looks good for me.

commit 765a69c4 (2016-02-05): Remove dashes from filenames to appease Arduino IDE

Fix #196

Recently target-specific arm and avr variants of the source code were
added causing some files to be renamed with "-arm" or "-avr" appended
to their filenames.  The arduino IDE v1.0.5 (and probably others) has a
bug which causes it to exclude these files from its build directory even
as the build commands call on them specifically.

Rename the files to remove the dash from their name.  Replace the
dash with an underscore instead.

I considered renaming them to foo.avr.c instead of foo_avr.c, but
I don't want to risk confusing the s [... cut]
closed 2016-11-20 by phord