Traumflug/Teacup_Firmware #197: Travel Limits

filoumoteur, 2016-02-25:
Hi,

I'm trying to use travel limits feature in Teacup and either I don't understand how to use them or there's a bug.
On my printer, the head can go out of the bed by 30mm before the endstop is trigged (X axis).
So, I set the X_MIN to 30 and the X_MAX to 230, but the printer seems to only consider the X_MAX. More than that, is looks that it does X_MAX - X_MIN as X_MAX and X_MIN stays flush to the endstop switch. As example, I can set X_MAX to 400 and X_MIN to 200 and the head will never hit the frame.

Is that a software issue or I haven't understand how to use travels limits?

*here's my printer.h config file renamed as .txt because Guthub doesn't want .h files.
[printer.prusa_i3_diy.txt](https://github.com/Traumflug/Teacup_Firmware/files/147298/printer.prusa_i3_diy.txt)

triffid, 2016-02-26:
Fwiw it's really handy to be able to move the nozzle off the bed, all my
printers have a startup routine which purges the nozzle over the edge of
the bed and wipes the noodle off on the edge of the bed before starting
each print.

To ensure I print in the centre, I either set the endstop location to a
negative value or simply tell my slicer where the bed centre is

Wurstnase, 2016-02-26:
I think you understand this wrong. If you want to set the 0 position of X to the edge of your bed and you can go 30mm ahead of it, you should set x_min to **-**30.

The is the way Marlin do it. But maybe Teacup works a bit different?!?