Traumflug/Teacup_Firmware #198: Hello @rollingdice!

Traumflug, 2016-03-06:
Just added @rollingdice as a collaborator.

Just a simple rule: keep fingers off of the master and experimental branch for now. Instead, open as many new branches as you feel need for. Branches are good for testing, trying, showing, communicating, whatever. Good stuff from there will be cherry-picked over to experimental and after an additional testing period to master. See also issue #81.

Feel welcome!

rollingdice, 2016-03-07:
Thanks @Traumflug. I'll try to complete delta support on Teacup firmware. Currently, I'm looking to creating a new config tab on Configtool for delta printer. Another thing I'll try to implement is individual tower angle correction.

dcousens, 2016-06-30:
Probably could close this issue?

Traumflug, 2016-06-30:
Uhm, this support for deltas in Configtool is still missing.